realsenseai/librealsense #1950: Regarding confidence maps of point clouds

sandeand, 2018-06-25:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 5.09.02.00 | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    | 4.15.0-23-generic  | 
| Platform                                 | PC  |
| SDK Version                            |  2.12  | 
| Language                            |  C   | 
| Segment			|  Robot  | 

### Issue Description
Hello, I have a question about confidence maps for point clouds for computer vision applications.

I found [this question](https://github.com/IntelRealSense/librealsense/issues/1619), as well as [this](https://github.com/IntelRealSense/librealsense/issues/448) and I am a bit confused: Can I get confidence maps for point clouds out of librealsense or the SDK? If not, what other options do I have to find confidence values for the point cloud?

If this is not a possibility, I am aware that I can opt for using the 'High Accuracy' preset described [here](https://github.com/IntelRealSense/librealsense/wiki/D400-Series-Visual-Presets) to get high confidence points. This would be a good plan B to be fair, but if it would be possible to get the confidence of each point of the high density preset's point cloud, then that would be ideal.

dorodnic, 2018-06-25:
Hi @sandeand 
I believe that the D400 series hardware can't provide a confidence map.
The SR300 has a dedicated confidence channel (but it was never implemented in librealsense).
It is somewhat more complicated to estimate confidence in stereo matching solution (D400) than structured-light (SR300).

sandeand, 2018-06-26:
Okay, that makes a lot of sense. 

Then that brings me back to the question of a plan C - is there any way apart from the presets and camera settings to control or get some read on the accuracy of the points measured?

dorodnic, 2018-06-26:
There is a lot of settings under **Advanced Mode** in the viewer, and it is all exposed via the `advanced_mode` API. 
I'm not aware of additional APIs that can be used for this.

sandeand, 2018-06-26:
Okay, I will look into Advanced Mode as well as some of the filters built into the SDK. Thank you for your help, I appreciate it!

closed 2018-06-26 by sandeand