realsenseai/librealsense #1951: Brightness of IR sensor.

shiyamon, 2018-06-26:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.09.11.00 | 
| Operating System & Version |   {Win 10 / MacOS  | 
| Platform                                 | PC |
| SDK Version                            | 2.13.0 }                          | 
| Language                            |  C++  | 
| Segment			|  others                         | 

Hi, I am using RealSense camera as IR camera for sensing IR light. I often used R200 but switched to D435 since R200 is no longer available.
I found that IR frame of D435 is much brighter than R200. I tried to tweak exposure but couldn't get similar result with R200.
I know the module is different but does librealsense have any options that controls exposure or IR filter?
The attachment is comparison between D435(left) and R200(right) shooting the same room.

[image]

agrunnet, 2018-06-26:
Yes you can change the exposure manually as well as the gain. You can also change the autoexposure setpoint if you prefer. It is under the advanced settings.

RealSense-Customer-Engineering, 2018-07-16:
[Realsense Customer Engineering Team Comment]
let me close this ticket here, please feel free to open new ticket if any question

closed 2018-07-16 by RealSense-Customer-Engineering