realsenseai/librealsense #1952: depth_sensor start, Segmentation Error

dmitryshendryk, 2018-06-26:
|---------------------------------|------------------------------------------- |
| Camera Model                      D415
| Firmware Version                   (Open RealSense Viewer --> Click info) 
| Operating System & Version  Ubuntu 16.04
| Kernel Version (Linux Only)     4.14.13                                    |
| Platform                                 PC
| SDK Version                                2 
| Language                            C++
| Segment			others

I have a problem with depth_sensor, when i call start function, i got Segmentation fault (core dumped),
if i comment it, no error. startFrames() function call from main(), any ideas why color_sensor works and depth_sensor doesn't?

realsense_int.hpp

`
class RealSense {
    public:
        RealSense(int width= 640, int height=480, int fps=30);
        ~RealSense();
         void startFrames();
        
    private:
        int _width;
        int _height;
        int _fps; 
        rs2::context ctx;   
        rs2::config cfg;
        rs2::sensor color_sensor;
        rs2::depth_sensor *depth_sensor;
        rs2::device dev;
        
};
`

realsense_int.cpp
`
RealSense::RealSense(int width, int height, int fps)
    : _width(width), _height(height), _fps(fps)
    {
       cout << "RealSense" << "\n";
        if (ctx.query_devices().size() == 0)
        {
            cout << "RealSense device not detected" << "\n";
            exit(0);
        }
    cfg.enable_stream(RS2_STREAM_DEPTH, width, height, RS2_FORMAT_Z16, fps);
    cfg.enable_stream(RS2_STREAM_COLOR, width, height, RS2_FORMAT_RGB8, fps);

    dev = ctx.query_devices().front();
    rs2::depth_sensor temp_depth_sensor = dev.query_sensors()[0].as<rs2::depth_sensor>();
    color_sensor = dev.query_sensors()[1].as<rs2::sensor>();

    depth_sensor = &(temp_depth_sensor);

    rs2::stream_profile depth_stream_profile = depth_sensor->get_stream_profiles()[208];
    rs2::stream_profile color_stream_profile = color_sensor.get_stream_profiles()[23];

    depth_sensor->open(depth_stream_profile);
    color_sensor.open(color_stream_profile);

 
    };

 void RealSense::startFrames()
 {
        // This line gives segmentation error
       depth_sensor->start([&](rs2::frame f) { cout << 12 << '\n'; });
       color_sensor.start([&] (rs2::frame f) { cout << 13 << '\n' ;});
 }

 RealSense::~RealSense()
    {
         depth_sensor->stop();
        color_sensor.stop();
    };
`

dorodnic, 2018-06-26:
This line:
`depth_sensor = &(temp_depth_sensor);`
Seems very suspicious. Regardless of RealSense, you are taking a pointer to a local variable, which is undefined behaviour in C++. 
Please try changing `rs2::depth_sensor *depth_sensor;` to `rs2::sensor depth_sensor;` and then downcast (`depth_sensor.as<rs2::depth_sensor>()`) when you use it.

dmitryshendryk, 2018-06-27:
Already tried this, in that way I will have
no matching function for call to `‘rs2::depth_sensor::depth_sensor()’`
Because depth_sensor class doesn't have default constructor, it requires sensor.
And i don't want to create it, i get it from `dev.query_sensors()`

Correct me if i'm wrong

dorodnic, 2018-06-27:
Yes, but you could save it as just `sensor` and then downcast only when you actually use it.

dmitryshendryk, 2018-06-27:
@dorodnic 
Thanks for help. Solved!

closed 2018-06-27 by dmitryshendryk