realsenseai/librealsense #1953: Bad performance with RealSense rendering PointClouds in Unity

nachoreimat, 2018-06-26:
| Required Info                         |                          |
|---------------------------------|-----------------|
| Camera Model                       |      D415        | 
| Firmware Version                   |    05.09.11.0.0   | 
| Operating System & Version |       {Win 10      | 
| Platform                                 | PC, i7, gtx1070 |
| SDK Version                           |        2.12.0        | 
| Language                               |      C#/unity     | 

### Issue Description
Bad performance with RealSense rendering PointClouds in Unity. Lag is very high and frames per second falls when increasing resolution.

Dear all, I have been doing some tests of rendering PointClouds in Unity using the Unity wrapper code that comes with the RealSense SDK. My purpose is to do a high quality photo-realistic and real-time rendering for a VR application, so the lag should be close to10ms and a high frame-rate.

Those tests gave me very bad performance in terms of Lag and Fps. Let me show you:

Resolution          |  FPS  | LAG(ms)       |
|------------------|------|---------------|
640x480              |  19   |  241              |
848x480              |  14   |  326              |
1280x720            |   4    |  545              |
1920x1080(RGB) |    2   |  983              |

I don't know where is the bottle neck. Is it in the align step? maybe in the Particle System used to render it?

Anyone achieved highest quality with good performance? I would appreciate any hint/help.

Thanks in advance

nachoreimat, 2018-06-26:
Good, 
I am also trying to render it with another technique. I will report my findings
Please keep me updated.

Thanks

RealSense-Customer-Engineering, 2018-07-09:
[Realsense Customer Engineering Team Comment]
Hi @nachoreimat,

I am interested if you have any findings so far?

jaredgaul, 2018-07-10:
I had a similar issue a while back using the Kinect and rendering depth information. I had much better performance by using the depth texture as an input to a vertex shader and manipulating a mesh that way. Essentially a dynamic height map. Went from 7fps in the kinect sample to 80fps. You could do the same thing with this. I had custom built the height map mesh in script so there was one vert per pixel.

matkatz1, 2018-07-24:
A mesh-based solution is available on release 2.14.0

RealSense-Customer-Engineering, 2018-07-25:
[Realsense Customer Engineering Team Comment]
Hi @nachoreimat,

As @matkatz1 mentioned, please check librealsense 2.14.0 to see how it improves the performance.

RealSense-Customer-Engineering, 2018-08-02:
[Realsense Customer Engineering Team Comment]
Hi @nachoreimat,

any update?

RealSense-Customer-Engineering, 2018-08-06:
[Realsense Customer Engineering Team Comment]
Since the performance should be improved, will close this. If any further question, you still can re-open or create another ticket.

closed 2018-08-06 by RealSense-Customer-Engineering

nachoreimat, 2018-08-06:
Hi all, 
Sorry for the delayed answer, I've been busy and then some holidays.

Yes, we implemented a solution based on the geometry shader and the performance increased.
Our current solution is very close to the mesh based solution you uploaded on the version 2.14.0 so we can say we both are in the good direction.
Your new mesh based solution works really good and it keeps constant at 60fps.
Good job!
Thanks