realsenseai/librealsense #1954: librealsense ROS Wrapper kernel patch failed on IntelAero

baumanta, 2018-06-26:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415, D435 | 
| Firmware Version                   | 05.09.11.00 | 
| Operating System & Version |   Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  4.4.76-aero-1.2                                         | 
| Platform                                 | Intel Aero  |
| SDK Version                            |  { master / 2.0.3 }                          | 
| Language                            |  C++                      | 
| Segment			|  Drone                          | 

### Issue Description
I have experienced a similar issue as described in #1320 . I followed the instructions under [ROS wrapper](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md) to install the ROS wrapper. But I got an error when executing the kernel patch script with
`./scripts/patch-realsense-ubuntu-lts.sh`. This is probably because the script is for an Ubuntu 16.04 LTS kernel and I have the already patched Aero kernel. I can still launch the Realsense Camera and get all the data. But the incorrectly (or not at all) patched kernel causes the realsense node to throw the following warning:

 26/06 15:45:18,160 WARNING [139849831933824] (backend-v4l2.cpp:486) Failed to read busnum/devnum. Device Path: /sys/dev/char/81:11/device/../../../../../../../../../../
 26/06 15:45:18,160 WARNING [139849831933824] (backend-v4l2.cpp:486) Failed to read busnum/devnum. Device Path: /sys/dev/char/81:12/device/../../../../../../../../../../
 26/06 15:45:18,160 WARNING [139849831933824] (backend-v4l2.cpp:486) Failed to read busnum/devnum. Device Path: /sys/dev/char/81:13/device/../../../../../../../../../../
 26/06 15:45:18,161 WARNING [139849831933824] (backend-v4l2.cpp:486) Failed to read busnum/devnum. Device Path: /sys/dev/char/81:14/device/../../../../../../../../../../
 26/06 15:45:18,161 WARNING [139849831933824] (backend-v4l2.cpp:486) Failed to read busnum/devnum. Device Path: /sys/dev/char/81:15/device/../../../../../../../../../../
 26/06 15:45:18,161 WARNING [139849831933824] (backend-v4l2.cpp:486) Failed to read busnum/devnum. Device Path: /sys/dev/char/81:16/device/../../../../../../../../../../
 26/06 15:45:18,162 WARNING [139849831933824] (backend-v4l2.cpp:486) Failed to read busnum/devnum. Device Path: /sys/dev/char/81:17/device/../../../../../../../../../../
 26/06 15:45:18,162 WARNING [139849831933824] (backend-v4l2.cpp:486) Failed to read busnum/devnum. Device Path: /sys/dev/char/81:18/device/../../../../../../../../../../
 26/06 15:45:18,162 WARNING [139849831933824] (backend-v4l2.cpp:486) Failed to read busnum/devnum. Device Path: /sys/dev/char/81:19/device/../../../../../../../../../../
 26/06 15:45:18,163 WARNING [139849831933824] (backend-v4l2.cpp:486) Failed to read busnum/devnum. Device Path: /sys/dev/char/81:0/device/../../../../../../../../../../
 26/06 15:45:18,163 WARNING [139849831933824] (backend-v4l2.cpp:486) Failed to read busnum/devnum. Device Path: /sys/dev/char/81:1/device/../../../../../../../../../../
 26/06 15:45:18,163 WARNING [139849831933824] (backend-v4l2.cpp:486) Failed to read busnum/devnum. Device Path: /sys/dev/char/81:2/device/../../../../../../../../../../
 26/06 15:45:18,163 WARNING [139849831933824] (backend-v4l2.cpp:486) Failed to read busnum/devnum. Device Path: /sys/dev/char/81:3/device/../../../../../../../../../../
 26/06 15:45:18,164 WARNING [139849831933824] (backend-v4l2.cpp:486) Failed to read busnum/devnum. Device Path: /sys/dev/char/81:4/device/../../../../../../../../../../
 26/06 15:45:18,164 WARNING [139849831933824] (backend-v4l2.cpp:486) Failed to read busnum/devnum. Device Path: /sys/dev/char/81:5/device/../../../../../../../../../../
 26/06 15:45:18,164 WARNING [139849831933824] (backend-v4l2.cpp:486) Failed to read busnum/devnum. Device Path: /sys/dev/char/81:6/device/../../../../../../../../../../
 26/06 15:45:18,165 WARNING [139849831933824] (backend-v4l2.cpp:486) Failed to read busnum/devnum. Device Path: /sys/dev/char/81:7/device/../../../../../../../../../../
 26/06 15:45:18,165 WARNING [139849831933824] (backend-v4l2.cpp:486) Failed to read busnum/devnum. Device Path: /sys/dev/char/81:8/device/../../../../../../../../../../
 26/06 15:45:18,165 WARNING [139849831933824] (backend-v4l2.cpp:486) Failed to read busnum/devnum. Device Path: /sys/dev/char/81:9/device/../../../../../../../../../../
 26/06 15:45:18,166 WARNING [139849831933824] (backend-v4l2.cpp:486) Failed to read busnum/devnum. Device Path: /sys/dev/char/81:10/device/../../../../../../../../../../

This warning is somehow logged in the system log of the OS causing the log to build up and fill all the memory (in a matter of about 5min). Also the logging needs about 70% of the CPU power of the Intel Atom board.

Is there a possibility to provide a kernel patching script for the Intel Aero kernel (4.4.76-aero-1.2), to correctly install the ROS wrapper?

RealSense-Customer-Engineering, 2018-06-29:
[Realsense Customer Engineering Team Comment]
Hi @baumanta,

Please check if you install Intel Realsense permission scripts located in librealsense source directory:
`sudo cp config/99-realsense-libusb.rules /etc/udev/rules.d/`
`sudo udevadm control --reload-rules && udevadm trigger`
If not, please install that. Then, reboot the system and try again.

baumanta, 2018-06-29:
I followed the instructions step by step. I did install the permission scripts under step 3 before trying to patch the kernel

RealSense-Customer-Engineering, 2018-07-05:
[Realsense Customer Engineering Team Comment]
HI @baumanta,

If you think below warning message "Failed to read busnum/devnum ..." affects your system performance. You can try to comment out the debug log in "backend-v4l2.cpp", and rebuild librealsense.
// LOG_WARNING("Failed to read busnum/devnum. Device Path: " << path);

Let me know how this works.

RealSense-Customer-Engineering, 2018-07-16:
[Realsense Customer Engineering Team Comment]
Hi @baumanta,

Any update? Still need any support for this topic?

baumanta, 2018-07-16:
We decided to not further use Intel Aero with the D415. Therefore, the problem is not relevant for me anymore. But if the Intel Aero should be compatible with the new Realsense generation, I would suggest providing a file to properly patch the Aero kernel.

RealSense-Customer-Engineering, 2018-07-17:
[Realsense Customer Engineering Team Comment]
Ok, @baumanta, Thanks for the suggestion. We will try to support more platforms.

RealSense-Customer-Engineering, 2018-09-10:
[Realsense Customer Engineering Team Comment]
D415/D435 camera + librealsense 2.15.0 + FW 5.10.3 can stream well as a USB2 device on Aero board + Ubuntu 16.04 + Kernel 4.13.0-36.
Don't need to apply the patch manually. Just install the DKMS pre-built package.
$ sudo apt install librealsense2-dkms

closed 2018-09-10 by RealSense-Customer-Engineering