realsenseai/librealsense #1955: Android ''fail to initialize  libuvc Last Error: Permission denied''

fotsing365, 2018-06-26:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  {D400}  | 
| Firmware Version                   |{}  | 
| Operating System & Version |   {Android 6.0.1} | 
| Kernel Version (Linux Only) |   |  )                                         | 
| Platform                                 | Rooted Lenovo YT3-850F |
| SDK Version                            |   2.12.0                          | 
| Language                            |  C++                        | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
< After rooting my android tablet I just followed from [instructions](https://github.com/IntelRealSense/librealsense/blob/master/doc/android/AndroidJavaApp.md)  but after running  "realsense_app" > I got following error: ''fail to initialize  libuvc Last Error: Permission denied''. Every single step of process went well. Please somebody know how to fix it?

dorodnic, 2018-06-26:
Hi @fotsing365 
Did you run step 16?

> Open the Terminal Emulator application and type below lines in order to move to Super User mode and change the USB permissions.

```
su
lsusb
chmod 0777 /dev/bus/usb/<Bus number>/<Dev Number>
```

Please note you need `<Bus number>/<Deb number>` for the device with PID 8086.

fotsing365, 2018-06-26:
Hi @dorodnic  I did it

fotsing365, 2018-06-26:
respectively screen shot of  "lsusb", "realsense_app launched" and "Super S
[image]
[image]
[image]

u"

dorodnic, 2018-06-26:
I believe this should be sufficient, could you please also post the `chmod` command you are issuing?

fotsing365, 2018-06-26:
Chmod screen
[image]
shot

fotsing365, 2018-06-26:
It seems like realsense_app don't ask root access

fotsing365, 2018-06-27:
I am not an android developer. But what I think it is necessary to do is to modify the application so that it can ask the access to the root. Take  a look on this [link](https://www.sitepoint.com/developers-guide-risky-android-app-permissions/) perhaps you could find solution.

fotsing365, 2018-06-28:
Hi everyone 
After trying debugging using Logcat messages I notice that, the execution of native function start **startStreaming()** stops on  instruction auto  **list = ctx.query_devices();** (it is to detect RealSense camera)

fotsing365, 2018-06-28:
Hi everyone 
After investigating  I notice that during the realsense_app installation, android studio displays following in the LogCat console:   

        06-28 14:57:40.073 13978-13978/com.example.realsense_app I/e.realsense_app: type=1400 audit(0.0:17331): avc: denied { read } for name="usb" dev="tmpfs" ino=68209 scontext=u:r:untrusted_app:s0:c512,c768 tcontext=u:object_r:usb_device:s0 tclass=dir permissive=1
    type=1400 audit(0.0:17332): avc: denied { open } for path="/dev/bus/usb" dev="tmpfs" ino=68209 scontext=u:r:untrusted_app:s0:c512,c768 tcontext=u:object_r:usb_device:s0 tclass=dir permissive=1
    type=1400 audit(0.0:17333): avc: denied { create } for scontext=u:r:untrusted_app:s0:c512,c768 tcontext=u:r:untrusted_app:s0:c512,c768 tclass=netlink_kobject_uevent_socket permissive=1
    type=1400 audit(0.0:17334): avc: denied { bind } for scontext=u:r:untrusted_app:s0:c512,c768 tcontext=u:r:untrusted_app:s0:c512,c768 tclass=netlink_kobject_uevent_socket permissive=1
    type=1400 audit(0.0:17335): avc: denied { setopt } for scontext=u:r:untrusted_app:s0:c512,c768 tcontext=u:r:untrusted_app:s0:c512,c768 tclass=netlink_kobject_uevent_socket permissive=1

aarun2211, 2018-06-29:
@fotsing365 Were you able to make any progress with the issue? I seem to be having exactly the same issue. I have been able to run the headless tools and examples, but I have not been able to use the app.

fotsing365, 2018-07-02:
Hi everyone 
 I think the solution to this problem is just to set the  access right  to the camera. I did it just by add theses two lines in the file *AndroidManifest.xml* 

<uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE" />
 <uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE" />

Adding this allow me to see the warning *No RealSense devices where found!* when there is no camera connect on my tablet. But application crashes when I connect the camera
for more informations about how to fix this issue take a look on this [link](https://stackoverflow.com/questions/33162152/storage-permission-error-in-marshmallow/41221852#41221852). 
I am really disappointed my Master project despite all my expenses to get me these cameras is now blocked.

RealSense-Customer-Engineering, 2018-07-03:
[Realsense Customer Engineering Team Comment]
Please check the permission access the device node

chmod -R 777 /dev/bus/usb/

chmod 666 /dev/video* /*it seems not be required when using libuvc*/

And disable SELinux (SEAndroid)

setenforce 0

fotsing365, 2018-07-03:
Hi @RealSense-Customer-Engineering. Thanks. I have already fix camera acces the problem now is about application crashing when I connect my RealSense D435.

RealSense-Customer-Engineering, 2018-07-05:
[Realsense Customer Engineering Team Comment]
@fotsing365 Could you please create another ticket to track the crash error and capture the logcat and dmesg logs for the crash? Thanks!

fotsing365, 2018-07-05:
Hi @RealSense-Customer-Engineering  
 I am going to do so

RealSense-Customer-Engineering, 2018-07-09:
[Realsense Customer Engineering Team Comment]
@fotsing365 That's fine. Considering the issue in this ticket resolved, close it accordingly. Thanks!

closed 2018-07-09 by RealSense-Customer-Engineering

alex1122a, 2018-09-13:
Dear @RealSense-Customer-Engineering ,
I am facing the same issue. None of abovementioned instructions resolved the issue.
- Camera Model	: D435
- Operating System & Version:	Android 5.0.1
- Platform:	Rooted Galaxy S4 GT-I9500
- SDK Version:	2.16.0
Could you kindly provide further instructions?