realsenseai/librealsense #1956: Multiple camera weird behavior and losing frames

nachoreimat, 2018-06-26:
| Required Info                         |                          |
|---------------------------------|-----------------|
| Camera Model                       |      D415        | 
| Firmware Version                   |    05.09.11.0.0   | 
| Operating System & Version |       Win 10      | 
| Platform                                 | PC, i9, 2xgtx1080 |
| SDK Version                           |        2.13.0        | 
| Language                               |      C#/unity     | 
### Issue Description
Multiple camera support. 

I think this issue can be solved with a quick question: How many cameras can work at highest resolution on a single machine?

Anyway I explain my real problem.

I would like to use  at least 4 cameras for 3D reconstruction. By doing a quick check with the Intel Realsense Viewer with last version of SDK and Firmware, I have been able to have 3 cameras working with Depth and Color streams at max. resolution. But when I connect the 4th camera all of them start loosing frames and freezing. They are connected in USB 3.0 ports directly on the motherboard.

Sometimes same behavior even happens when connecting the 2nd or 3rd camera. So I have not been able to find a pattern on this behavior. 
I am starting to thing my machine has a problem or something but it is a powerful one, so it should not be the problem. 

Do I have not enough power for too many cameras? it can be a problem of USB bandwith? Any hints?

Thanks in advance

dorodnic, 2018-06-26:
Hello @nachoreimat 
There should be no problem to use more than one camera, but you should be aware of USB3 bandwidth limitations. Ideally, you would want to connect several USB controllers to PCI-E slots on your motherboard and connect 2 cameras to each. 
When two cameras share a USB3 hub (and sometimes even the USB controller) they will inevitably start loosing frames in somewhat unpredictable way. 
We had very good experience with [Renesas usb 3.0 controllers](https://www.renesas.com/en-us/solutions/key-technology/connectivity-wired/usb.html), but other brands can also work well.

agrunnet, 2018-06-26:
Did you have a chance to real the multicamera white paper on the RealSense website? It outlines some of the constraints which include power and bandwidth and many more. There are tables for number of cameras.

dorodnic, 2018-06-26:
Link [here](https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/RealSense_Multiple_Camera_WhitePaper.pdf)

nachoreimat, 2018-06-27:
Thank you for the comments and the paper. I found it very interesting, also answered a lot of doubts about synchronization and other problems we have been found.

So It seems that 4 cameras will only work in the same machine at lowest resolution, unless we design a system taking into account all the bandwidth, power and computation restrictions. Am I right?

agrunnet, 2018-06-27:
Yes you can use 4 cameras through a hub if you use 640x360 resolution. You will need to make sure the hub is powered.

closed 2018-07-06 by RealSense-Customer-Engineering