realsenseai/librealsense #1958: Color Image becomes dark when out of while loop

ljc19800331, 2018-06-27:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { SR300} | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Linux (Ubuntu 16.04)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {python}                          | 
| Segment			|  { Point Cloud }                          | 

### Issue Description

Hello, I want to capture the color image and save as png file with opencv. However, when I run the python example (opencv_viewer_example.py) I can see a good real-time image with bright color. This is written in the while loop:
try:
    while True:
......................................................CODE...............................................................
        cv2.namedWindow('RealSense', cv2.WINDOW_AUTOSIZE)
        cv2.imshow('RealSense', images)
        cv2.waitKey(1)
finally:
pipeline.stop()
.....................................................................................................................
This shows good realtime image. However, when I want to save one image at one time, and I delete the "while loop" and only keep the remaining code, the image becomes much darker then the previous one.

I am wondering is there anyway we can change the brightness of the color image in python? Or do I need to change the code? Thank you for the answer.

dorodnic, 2018-06-27:
Hi @ljc19800331 
Yes, you can control exposure, gain, brightness and contrast via set_option API. 
That said, I suspect the problem is related to auto-exposure (stream starts dark and then changes exposure to adjust), did you try skipping ~5 first frames?

ljc19800331, 2018-06-29:
Great thanks. It solve my problem.

closed 2018-06-29 by ljc19800331