realsenseai/librealsense #1959: D400 Inter Camera HW Sync

ankarako, 2018-06-27:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   |5.9.13 | 
| Operating System & Version |   Win10| 
| Kernel Version (Linux Only)    |  - | 
| Platform                                 | PC|
| SDK Version                            |  2.13.0                          | 
| Language                            |  C/C++| 
| Segment			| FVV}                          | 

### Issue Description
Hello, 

I'd like to ask if the Inter Camera HW Sync is really available. There is an _rs2_option_ added in the latest SDK (v2.13.0: https://github.com/IntelRealSense/librealsense/releases). However, the recommended FW version according to this release is 5.9.14. When you go to the available FWs (https://realsense.intel.com/intel-realsense-downloads/#firmware) you can see that the latest FW is 5.9.11. When you click on it (https://downloadcenter.intel.com/download/27792/Latest-Firmware-for-Intel-RealSense-D400-Product-Family), you can see that the latest available FW is 5.9.13. So I updated one my devices FW to 5.9.13.

When I run the following code:
```
#include "stdafx.h"
#include <librealsense2/rs.h>
#include <librealsense2/rs_advanced_mode.hpp>
#include <iostream>

int main()
{
	rs2::context cxt;
	rs2::device dev = cxt.query_devices().back();
	auto sensors = dev.as<rs400::advanced_mode>().query_sensors();
	
	for (auto sensor : sensors) {
		std::cout << sensor.get_info(rs2_camera_info::RS2_CAMERA_INFO_RECOMMENDED_FIRMWARE_VERSION) << std::endl;
		std::cout << sensor.get_info(rs2_camera_info::RS2_CAMERA_INFO_FIRMWARE_VERSION) << std::endl;
		std::cout << sensor.supports(RS2_OPTION_INTER_CAM_SYNC_MODE) << std::endl;
	}
	

    return std::getchar();
}
```
I get:
```
05.09.09.02
05.09.13.00
0
05.09.09.02
05.09.13.00
0
``` 
From which I conclude that the FW does not support Inter Camera HW Sync. Any help or information is really appreciated.

agrunnet, 2018-06-27:
Correct. We are currently working on validating and releasing new FW that includes this multicamera sync.

ankarako, 2018-06-27:
@agrunnet Thank for the reply.

closed 2018-06-27 by ankarako