realsenseai/librealsense #1960: Wrong/missing timestamps on depth when resolution is different from IR

sogartar, 2018-06-27:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.09.11.00 | 
| Operating System & Version |  Linux (Ubuntu 16 | 
| Kernel Version (Linux Only)    |  4.10.0                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.13.0                          | 
| Language                            |  C++                          | 
| Segment			|  others                          | 

### Issue Description
I am trying to run color+depth+IR, but I want the depth to have different resolution from the infrared.

The setup is as follows.
```c++
rs2::config config;
config.enable_stream(RS2_STREAM_DEPTH, 0, 640, 360, RS2_FORMAT_Z16, 30);
config.enable_stream(RS2_STREAM_INFRARED, 1, 1280, 720, RS2_FORMAT_Y8, 30);
config.enable_stream(RS2_STREAM_COLOR, -1, 1920, 1080, RS2_FORMAT_BGR8, 30);
```
Then I get these results.

| Field | Depth | IR |
|--|--|--|
| number |                                             64 | 64 |
| timestamp |                       **70714.423** | 10263.186 |
| timestampDomain |  **Hardware Clock** | Hardware Clock |
| Actual Fps |                                      **1** | 30 |
| Backend Timestamp |   1530026872867 | 1530026872825 |
| Frame Timestamp |           **70714423** | 10263186 |
| Time Of Arrival |            1530026872873 | 1530026872839    |
| Sensor Timestamp |               **missing** | 10261365 |
| Frame Counter |                     **missing** | 64 |

From the above table we can see that the temstamp, timestamp domain and fps are wrong. Also the sensor timestamp and frame counter from the metadata are missing.

If the resolution of the infrared matches that of the depth I don't get these problems.

Where does decimation of the depth occur? It seems that `librealsense::decimation_filter::decimate_depth` is not used for that. Does it happen on the device? Is IR first decimated and then depth is computed or first depth is computed and then decimated?

sogartar, 2018-06-27:
This may be related to #1773 and #1727.

dorodnic, 2018-06-27:
Something funny is going on, since I would expect this to not work. The device is not capable of producing depth and IR at different resolution and decimate is not used for such request, so we will have to reproduce and check whats going on.

RealSense-Customer-Engineering, 2018-07-09:
[Realsense Customer Engineering Team Comment]
@sogartar
I use C++ example to configure the setting as yours but found
it will throw a error message when run  pipe.start(config) after configure camera with same configuration as yours
and issue gone after set IR and depth same resolution
 The device is not capable of producing depth and IR at different resolution and decimate is not used for such request.

dorodnic, 2018-07-18:
@RealSense-Customer-Engineering this might be related to the specific software version, as we did make a lot of changes in that area of code. Did you try with 2.13?

dorodnic, 2018-09-06:
Hopefully it was a false-alarm. @sogartar you are always welcomed to open a new issue if you need more support.

closed 2018-09-06 by dorodnic