realsenseai/librealsense #1962: strange values when using rs2_deproject_pixel_to_point

omerbrandis, 2018-06-27:
hello, 

i'm using rs2_deproject_pixel_to_point with a d415 camera, librealsense version 2.12.0.

for pixels 0,0 at a distance of 0.3 ( which i guess means 30cm ) 
I receive point[0] = -0 and point[1] = -0
while i should be getting something closer to -15,-11  ( assuming no optical skew , calculated using simply trigonometry).

Can someone please explain these results?
( i do get "reasonable results when using larger values for parm depth)

Thanks,
Omer.

RealSense-Customer-Engineering, 2018-06-29:
[Realsense Customer Engineering Team Comment]
@omerbrandis
Can you share the code? I don't really understand what do you mean about pixels 0,0 with distance 0.3 meter. Since the depth generated from stereo camera use left camera as reference camera, the left sight of image contains t strip of black area which doesn't contain any depth information. pixel 0.0 must within this area, so I don't really understand why this pixel has the depth value of 0.3. you can refer to the image attached below.

carypai, 2018-06-29:
[image]

omerbrandis, 2018-07-01:
Dear Realsense Customer Engineering Team,

I'm trying to calculate an object's physical position ( in relation to the camera ) from its pixel location.

according to https://github.com/IntelRealSense/librealsense/wiki/Projection-in-RealSense-SDK-2.0
 this can be done using the rs2_deproject_pixel_to_point function.

+ a very similar strange value (0,0) is returned from rs2_deproject_pixel_to_point for the opposite point 639,479 ( i'm using 640X480 resolution) , while it should return something nearing 15,11.

please advise,
Omer.

RealSense-Customer-Engineering, 2018-07-02:
[Realsense Customer Engineering Team Comment]
@omerbrandis
I still don't get your point, the 3D space center {0,0,0} referring to the center of the physical imager (Left IR). So the 3D point deprojected from upper left corner of 2D image pixel, you should get back the negative x, negative y, and positive z value. You can reference to the description below.

https://github.com/IntelRealSense/librealsense/wiki/Projection-in-RealSense-SDK-2.0#point-coordinates
**"Each stream of images provided by this SDK is also associated with a separate 3D coordinate space, specified in meters, with the coordinate `[0,0,0]` referring to the center of the physical imager"**

you can refer tot the description of rsutil.h function "rs2_deproject_pixel_to_point", what you should provide are 1)the pixel you want to deproject to 3D point, 2)depth value within the pixel, and 3)intrinsic of specific imager(left image) then you can get back the 3D point relative to the 2D pixel.
function description is "**Given pixel coordinates and depth in an image with no distortion or inverse distortion coefficients, compute the corresponding point in 3D space relative to the same camera**"

omerbrandis, 2018-07-02:
Dear Realsense Customer Engineering Team ,

you wrote:
"...what you should provide are"

"1)the pixel you want to deproject to 3D point," 
-> I have done this. for example pixel position x=639 y=479.

"2)depth value within the pixel"
-> i have done this. i've passed the value 0.3.

"3)intrinsic of specific imager"
-> i have also done this. 
used the following code  
` auto Color_stream   = MyPipelineProfile.get_stream(RS2_STREAM_COLOR).as<rs2::video_stream_profile>();
    auto CamIntrinsicts = Color_stream.get_intrinsics();`

the first two values I receive in the first output parameter (called point)  of function rs2_deproject_pixel_to_point are 0 and 0 .

values i expect to get should be near the values: 15,11.

Omer.

RealSense-Customer-Engineering, 2018-07-03:
[Realsense Customer Engineering Team Comment]
@omerbrandis
There are two issue found form your input.
1) the depth value, you need to also multiply the depth scale unit, you can refer to the depth frame "get_distance" method also github issue #1890
2) the intrinsic you use is the color_stream intrinsic, which is not right, you should use the Left IR camera intrinsic, which can be retrieved from depth sensor.

omerbrandis, 2018-07-04:
100%,
Thanks.
Omer.

omerbrandis, 2018-07-10:
dear Realsense Customer Engineering Team,

I have performed additional tests,
using same code as mentioned above,
but i've changed camera to d435 ( firmware 5.9.11).

i have placed an object at pixel position 522. ( distance is 59cm , ( depth scale is 0.001 so input the the function is 590)
the deprojection function returns relative position of 30.3704,
but the object is only 18 cm to the right of the center of the camera.

please advise.
Omer.

omerbrandis, 2018-07-11:
here is the code 
`// get & store camera's depth scale
    rs400::advanced_mode MyDevice=MyPipelineProfile.get_device();
    rs2::depth_sensor DepthSensor = MyDevice.first<rs2::depth_sensor>();
    DepthUnits = DepthSensor.get_option(RS2_OPTION_DEPTH_UNITS);

    // get & store depth intrinsics
    auto DepthStream = MyPipelineProfile.get_stream(RS2_STREAM_DEPTH).as<rs2::video_stream_profile>();
    DepthIntrinsics = DepthStream.get_intrinsics();

// convert DistanceToTarget  from "Cm"  to camera's depth scale
    float DistanceToTargetInDepthScale = (float)DistanceToTargetCm / 100 / DepthUnits;

std::cout << " !!!  depthunits = " << DepthUnits << "  DistanceToTargetInDepthScale  =  " << DistanceToTargetInDepthScale  << std::endl;

    // using librealsense's built in function for this
    float ResultVector[3];
    float InputPixelAsFloat[2];
    InputPixelAsFloat[0]=XPixelNumber;
    InputPixelAsFloat[1]=YPixelNumber;

rs2_deproject_pixel_to_point(ResultVector, &DepthIntrinsics, InputPixelAsFloat, DistanceToTargetInDepthScale);
`

when 
rgb&depth stream resolution are 640X480
InputPixelAsFloat[0] = 522
DistanceToTargetInDepthScale = 590 ( i.e. 59cm since depth_scale = 0.001)

i get a value of 303.704 in ResultVector[0] ( which means 30.3704 cm since my depth_scale is 0.001).
( i should be getting some closer to 18cm)

Please advise.
Omer.

omerbrandis, 2018-07-12:
I have reason to believe that the cause for this error is bad factory calibration settings.
when i change value for &DepthIntrinsics.fx from the factory set value of 383.806 to  a value  of 610.231 (determined by opencl camera calibration ) the results improve significantly.

this opens up a new set of questions 
1. could this be a simple "bug" in the data stored for intrinsics of d435 , firmware 05.09.11.00 , for resolution 640X480 ?

2. is the intrinsics data used by RS for things like depth detection or rgb to depth stream alignment 
( i.e. i should update the intrinsics in the camera itself with the new information ...)

thanks,
Omer.

RealSense-Customer-Engineering, 2018-07-12:
[Realsense Customer Engineering Team Comment]
@omerbrandis
I don't think that's the calibration issue, since from my own 435, I got the same fx as you got. I think the problem comes from your distance, you divide depth_unit instead of multiply the depth_unit. I suggest you to use the "get_distance(int x, int y)" to get the depth value to feed into rs2_deproject_pixel_to_point.

omerbrandis, 2018-07-12:
Hello,

I'm not 100% sure I fully understand your last post,

but if you wrote that you have managed to reproduce (=received the same results for the same input) my results using your code and your camera
then it seems to me that you have proved that the problem is not in my code and not in my camera ( or its calibration ).

I repeat the relative world position of an object at 59cm away at pixel x=522 is not 30cm....

Please advise,
Omer.

RealSense-Customer-Engineering, 2018-07-13:
[Realsense Customer Engineering Team Comment]
@omerbrandis
I mean the code you wrote "float DistanceToTargetInDepthScale = (float)DistanceToTargetCm / 100 / DepthUnits;" divides the depth scale. But the last parameter you feed into "rs2_deproject_pixel_to_point" should be pixel data in specific row/column to multiply the depth unit. And there is API inside the "depth_frame" class help to retrieve the right data which is called "get_distance", you only need to feed in the row and column then the function returns back the right data you need to feed into "rs2_deproject_pixel_to_point".

omerbrandis, 2018-07-13:
Dear Realsense Customer Engineering Team

a. I am familiar with the get_distance function.

b. I perform 
"float DistanceToTargetInDepthScale = (float)DistanceToTargetCm / 100 / DepthUnits;"
since this is what i was instructed to do by the Realsense Customer Engineering Team , 
please review the full thread.

c. but to get things going , i have created a new version of my code that does what you have instructed:

```
rs2::depth_frame TempDepthFrame = AlignedLastCapturedFrames.get_depth_frame();
    float TempDistance = TempDepthFrame.get_distance(XPixelNumber,YPixelNumber);
    std::cout << " distance fed to deproject " << TempDistance << std::endl;
    rs2_deproject_pixel_to_point(ResultVector, &DepthIntrinsics, InputPixelAsFloat,TempDistance);

    std::cout << "deprojection float : relativex = " << ResultVector[0] << "  relativey= " << ResultVector[1] << std::endl;

```

here is the output:
 distance fed to deproject 0.59
deprojection float : relativex = 0.303704  relativey= -0.120317

**which is still incorrect.**

please advise.
Omer.

RealSense-Customer-Engineering, 2018-07-16:
[Realsense Customer Engineering Team Comment]
@omerbrandis
I've tried similar scenario as yours with the code attached, the result is right without any problem. Please try to test with the code and let me know if you still face the same error result for x and y point. I've also attached the test result and environment from me in below response.

`​#include <librealsense2/rs.hpp> // Include RealSense Cross Platform API`
`#include <librealsense2/rs_advanced_mode.hpp>`
`#include <librealsense2/rsutil.h>`
`#include "../example.hpp" // Include short list of convenience functions for rendering`

`#include <algorithm> // std::min, std::max`
`#include <string>`
`#include <iostream>`

`#include <time.h>`
`//#include <sys/time.h>`
`#include <stdlib.h>`
`#include <stdio.h>`

`using namespace rs2;`
`using namespace std;`
`// Helper functions`
`void register_glfw_callbacks(window& app, glfw_state& app_state);`

`int main(int argc, char * argv[]) try`
`{`
` // Create a simple OpenGL window for rendering:`
` window app(1280, 720, "RealSense Pointcloud Example");`
` // Construct an object to manage view state`
` glfw_state app_state;`
` // register callbacks to allow manipulation of the pointcloud`
` register_glfw_callbacks(app, app_state);`

` pipeline pp;`
` config cfg;`
` float ResultVector[3];`
` float InputPixelAsFloat[2];`
` InputPixelAsFloat[0] = 416;`
` InputPixelAsFloat[1] = 316;`

` cfg.enable_stream(RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Z16 , 30);`
` cfg.enable_stream(RS2_STREAM_COLOR, 640, 480, RS2_FORMAT_RGB8, 30);`
` auto MyPipelineProfile = pp.start(cfg);`

` rs400::advanced_mode MyDevice = MyPipelineProfile.get_device();`
` rs2::depth_sensor DepthSensor = MyDevice.first<rs2::depth_sensor>();`
` float DepthUnits = DepthSensor.get_option(RS2_OPTION_DEPTH_UNITS);`

` // get & store depth intrinsics`
` auto DepthStream = MyPipelineProfile.get_stream(RS2_STREAM_DEPTH).as<rs2::video_stream_profile>();`
` rs2_intrinsics DepthIntrinsics = DepthStream.get_intrinsics();`
` // convert DistanceToTarget from "Cm" to camera's depth scale`

` for (int i = 0; i<10; i++)//to skip first few frames when device just initiated`
`  auto frames = pp.wait_for_frames();`

` auto frames = pp.wait_for_frames();`

` rs2::depth_frame depthFrame = frames.get_depth_frame();`
` float distance = depthFrame.get_distance(416, 316);`

` std::cout << " !!! depthunits = " << DepthUnits << " DistanceToTargetInDepthScale = " << distance << std::endl;`

` // using librealsense's built in function for this`

` rs2_deproject_pixel_to_point(ResultVector, &DepthIntrinsics, InputPixelAsFloat, distance);`

` std::cout << "x = " << ResultVector[0] << ", y = " << ResultVector[1] << ", z = " << ResultVector[2] << std::endl;`

` std::cout << "end" << std::endl;`

` return EXIT_SUCCESS;`
`}`
`catch (const rs2::error & e)`
`{`
` std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n " << e.what() << std::endl;`
` return EXIT_FAILURE;`
`}`
`catch (const std::exception & e)`
`{`
` std::cerr << e.what() << std::endl;`
` return EXIT_FAILURE;`
`}`
``

carypai, 2018-07-16:
I select pixel 416, 316 in the depth image which indicate the point on my cup has the distance about 39cm.
[image]
[image]

omerbrandis, 2018-07-16:
Dear Realsense Customer Engineering Team

when I test my code using pixel 416, 316  and depth= 0.387
i get very similar results to the ones you received , my results are x = 0.0923275   y = 0.0753535. 

but these values are incorrect.

I have recreated a scene in which an object projects at x=416,y=316,z=0.387 .

lets assume for the moment that rs2_deproject_pixel_to_point accepts depth data in meters, and returns ResultVector in meters, 
If so, the result x=0.097 means that the object in question is 9 cm to the right of the center of the camera,
but its not, in my scene it is only about 1cm to the right of the center of the camera. 

 
moreover, before you conclude that the results should be multiplied by 10  
please retest with additional targets, like the example i've provided before in which the object is at around 59cm away and about 18cm to the right of the camera ( it should project at around pixel 522). 
i expect you to receive a result of about 30cm 

Thanks,
Omer.

ev-mp, 2018-07-16:
@omerbrandis  hello,
It is not clear what do you mean by "values are incorrect" without supplying the reference data.
Please specify expected vs actual results along with rosbag recording, to present the case for in-depth review.

Additionally, in the code segment you provide:
>rs2::depth_frame TempDepthFrame = AlignedLastCapturedFrames.get_depth_frame();
    float TempDistance = TempDepthFrame.get_distance(XPixelNumber,YPixelNumber);
    std::cout << " distance fed to deproject " << TempDistance << std::endl;
    rs2_deproject_pixel_to_point(ResultVector, &DepthIntrinsics, InputPixelAsFloat,TempDistance);

The `AlignedLastCapturedFrames` implies that the depth images are not the original depth, but rather depth aligned to color. Therefore it is importation to verify that `&DepthIntrinsics` in the above code refers to the intrinsic of aligned depth.   
Also note that in the new aligned depth the location of the objects in the scene is different from original data and corresponds to the point-of-view of the color.

omerbrandis, 2018-07-16:
Dear Realsense Customer Engineering Team

1. i have done my best to explain what i mean by "in correct".
i have provided multiple examples of expected vs actual results.
and i can provide more if you wish.

(
 I have explained at the very start that a returned value of 30cm for an object that is only about 18 to the right of the camera is not correct.
i have also written in regard to you previous post, that a returned value of 9cm for an object that is only 1 cm to the right of the camera is incorrect.
)
I'm afraid I don't know how i can be more clear about this.

you mentioned " reference data" , can you please explain what you mean by this?
 
2. i do not understand why you refer to rosbag, ros has nothing to do with my application.

3. regarding stream alignment:

3.a. to the best of my knowledge, the results of the deproject method have nothing to do with the actual stream data, and/or alignment, 
only the data that is actually fed to it at run-time.

( i have tested multiple versions of the code, including the one that explicitly gets the distance to the object, all return incorrect values).

3.b. regardless, in my code the intrinsics are extracted before alignment is performed.

Thanks,
Omer.

ev-mp, 2018-07-16:
Hello @omerbrandis ,
The data presented is partial and misses some critical elements, such as actual intrinsic values used in transformations and the source of depth data (raw depth/depth aligned to color).

Please record a attach short record (<5sec) with 'realsense-viewer' with depth and color sensor streaming and mark the measured position of the points of interest. (Rosbag is the format 'realsense-viewer' uses internally). The video will be used to verify and reproduce the results.

Deprojection of 2D pixels into 3D space is based on intrinsic of the stream , so by mentioning
> in my code the intrinsics are extracted before alignment is performed.

it could be that the intrinsic passed to transformation was taken from the original depth, while the (i,j) pixel coordinates were obtained from the aligned depth, which can explain the offsets.

omerbrandis, 2018-07-17:
Dear Realsense Customer Engineering Team

[As we already have a solution to the problem] Management has instructed me to discontinue working on this thread.

Thank you for all your time and effort.
Omer.

closed 2018-07-17 by omerbrandis