realsenseai/librealsense #1963: write_to_ply does not work when loading from .bag file.

oscarzhaoyb, 2018-06-27:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | 05.08.15.00 | 
| Operating System & Version |   Ubuntu 16  | 
| Kernel Version (Linux Only)    |  4.13.0-45                                         | 
| Platform                                 | PC |
| SDK Version                            |  { 2.11.1 }                          | 
| Language                            |  {C++ }                          | 
| Segment			|  {others }                          | 

### Issue Description
I used the pointcloud.cpp as an template.
when I add the following line in the code, the write_to_ply does not work. The ply file only contains headers without any point cloud data or RGB data.
    // 
    rs2::config cfg;
    string str0 = "a.bag";
    cfg.enable_device_from_file(str0.c_str());   
    
    pipe->start(cfg);
...
   points.write_to_ply(filename,color);
if I just use pipe->start();, the write_to_ply file will fill point cloud data in the ply file.

Could you please tell me where I did incorrectly to save the ply file from a .bag file?

dorodnic, 2018-06-27:
Could you share the full code please? Perhaps you need to skip couple of frames to make sure both depth and RGB are available?

oscarzhaoyb, 2018-06-29:
// License: Apache 2.0. See LICENSE file in root directory.
// Copyright(c) 2015-2017 Intel Corporation. All Rights Reserved.

#include <librealsense2/rs.hpp> // Include RealSense Cross Platform API
#include "../example.hpp"          // Include short list of convenience functions for rendering

#include <algorithm>            // std::min, std::max
#include <string>
#include <iostream>

#include <time.h>
#include <sys/time.h>
#include <stdlib.h>
#include <stdio.h>

using namespace std;
// Helper functions
void register_glfw_callbacks(window& app, glfw_state& app_state);

int main(int argc, char * argv[]) try
{
    // Create a simple OpenGL window for rendering:
    window app(1280, 720, "RealSense Pointcloud Example");
    // Construct an object to manage view state
    glfw_state app_state;
    // register callbacks to allow manipulation of the pointcloud
    register_glfw_callbacks(app, app_state);
    
     

    // Declare pointcloud object, for calculating pointclouds and texture mappings
    rs2::pointcloud pc;
    // We want the points object to be persistent so we can display the last cloud when a frame drops
    rs2::points points;

    // Declare RealSense pipeline, encapsulating the actual device and sensors
    //rs2::pipeline pipe = std::make_shared<rs::pipeline>();
    auto pipe = std::make_shared<rs2::pipeline>();

    rs2::config cfg;
    //cfg.enable_device_from_file("a.bag");
    string str0 = "a.bag";
    string str1 = "a.bag";
    //cfg.enable_device_from_file(str0.c_str());  
    cfg.enable_record_to_file(str1.c_str());
    pipe->start(cfg); //File will be opened at this point 

    // Start streaming with default recommended configuration
    //pipe->start();
    rs2::device device = pipe->get_active_profile().get_device();

    string filename;
    int n = 0;

    struct timeval start;
    int frame_cnt = 0;
    string filelogname = "stream.log";
    FILE* pfFilelog = fopen(filelogname.c_str(),"w");
    cout << "loop starts" << endl; 
    while (app) // Application still alive?
    {
        //filename = "/var/voters/bulk_c/MA_Office/206/realsensedata/pcd"+to_string(n);
	filename = to_string(n)+".ply";        
	// Wait for the next set of frames from the camera
	cout << "get the time now"<< endl;	
	gettimeofday(&start, NULL);        
	cout << "time obtained : " << start.tv_sec << " sec, " << start.tv_usec << " usec" <<endl;
	auto frames = pipe->wait_for_frames();

        auto depth = frames.get_depth_frame();

        // Generate the pointcloud and texture mappings
        points = pc.calculate(depth);

        auto color = frames.get_color_frame();

        // Tell pointcloud object to map to this color frame
        pc.map_to(color);
	
	points.export_to_ply(filename, color);
	cout << "point cloud file saved." << endl;
        fprintf(pfFilelog,"%d,%d,%d\n",start.tv_sec,start.tv_usec,frame_cnt);
	fflush(pfFilelog);
	cout << "write to log at frame: " << frame_cnt << endl; 
	frame_cnt++;

        // Upload the color frame to OpenGL
        app_state.tex.upload(color);

        // Draw the pointcloud
        draw_pointcloud(app.width(), app.height(), app_state, points);
 	n++;
    }

    fclose(pfFilelog);
    return EXIT_SUCCESS;
}
catch (const rs2::error & e)
{
    std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;
    return EXIT_FAILURE;
}
catch (const std::exception & e)
{
    std::cerr << e.what() << std::endl;
    return EXIT_FAILURE;
}

oscarzhaoyb, 2018-06-29:
when I convert .ply from .bag file, using the following code,
cfg.enable_device_from_file(str0.c_str());  
//cfg.enable_record_to_file(str1.c_str());
pipe->start(cfg); //File will be opened at this point 

// Start streaming with default recommended configuration
//pipe->start();

when I convert .ply from streaming data, using the following code,
//cfg.enable_device_from_file(str0.c_str());  
//cfg.enable_record_to_file(str1.c_str());
//pipe->start(cfg); //File will be opened at this point 

// Start streaming with default recommended configuration
pipe->start();

RealSense-Customer-Engineering, 2018-07-12:
[Realsense Customer Engineering Team Comment]
@oscarzhaoyb
I've tried your code with slightly changes (since I run on Windows), the code works without problem. Please see the below attached image also the code :
`int main(int argc, char * argv[]) try`
`{`
` // Create a simple OpenGL window for rendering:`
` window app(1280, 720, "RealSense Pointcloud Example");`
` // Construct an object to manage view state`
` glfw_state app_state;`
` // register callbacks to allow manipulation of the pointcloud`
` register_glfw_callbacks(app, app_state);`

` // Declare pointcloud object, for calculating pointclouds and texture mappings`
` rs2::pointcloud pc;`
` // We want the points object to be persistent so we can display the last cloud when a frame drops`
` rs2::points points;`

` // Declare RealSense pipeline, encapsulating the actual device and sensors`
` auto pipe = std::make_shared<rs2::pipeline>();`

` rs2::config cfg;`
` string str0 = "test.bag";`
` cfg.enable_device_from_file(str0.c_str());  `
` pipe->start(cfg);`
` rs2::device device = pipe->get_active_profile().get_device();`

` string filename;`
` int n = 0;`

` int frame_cnt = 0;`
` cout << "loop starts" << endl;`
` while (app) // Application still alive?`
` {`
`  filename = to_string(n) + ".ply";`
`  cout << "get the time now" << endl;`
`  auto frames = pipe->wait_for_frames();`

`  auto depth = frames.get_depth_frame();`

`  // Generate the pointcloud and texture mappings`
`  points = pc.calculate(depth);`

`  auto color = frames.get_color_frame();`

`  // Tell pointcloud object to map to this color frame`
`  pc.map_to(color);`

`  points.export_to_ply(filename, color);`
`  cout << "point cloud file saved." << endl;`

`  // Upload the color frame to OpenGL`
`  app_state.tex.upload(color);`

`  // Draw the pointcloud`
`  draw_pointcloud(app.width(), app.height(), app_state, points);`
`  n++;`
` }`

` return EXIT_SUCCESS;`
`}`
`catch (const rs2::error & e)`
`{`
` std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n " << e.what() << std::endl;`
` return EXIT_FAILURE;`
`}`
`catch (const std::exception & e)`
`{`
` std::cerr << e.what() << std::endl;`
` return EXIT_FAILURE;`
`}`
`​`

carypai, 2018-07-12:
Images : 
[image]
[image]

oscarzhaoyb, 2018-07-17:
Hello, carypai,

Thanks for the support. Mine works in Linux finally after I reconfigure the code. Could you please let me know how you configure your code in windows? ( I guess you use vc++).

What ".h" and ".lib" do you set in "Include", "linkers", etc? Mine shows a lot of errors on GLFW. I don't see them in the github code.

Thanks a lot.

closed 2018-07-30 by RealSense-Customer-Engineering