realsenseai/librealsense #1965: RealSense crashes in playback_sensor.cpp

Chimdi511, 2018-06-27:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Operating System & Version |   Windows 10  |              | 
| Platform                                 | PC |
| SDK Version                           |   2.12.0                          | 
| Language                            |  C++                       | 

For some reason my code sometimes crashes in playback_sensor.cpp, but in the part of my code that has nothing to do with this. If this is relevant, I import data from bag files using these methods: 

`       west_cfg.enable_device_from_file("C:\\Users\\calagor\\Documents\\west.bag");
	west_pipe.start(west_cfg); // Load from file

	east_cfg.enable_device_from_file("C:\\Users\\calagor\\Documents\\east.bag");
	east_pipe.start(east_cfg); // Load from file

	north_cfg.enable_device_from_file("C:\\Users\\calagor\\Documents\\south.bag");
	north_pipe.start(north_cfg); // Load from file

	south_cfg.enable_device_from_file("C:\\Users\\calagor\\Documents\\south.bag");
	south_pipe.start(south_cfg); // Load from file`

However, my program crashes much farther down in my code, where I'm purely dealing with the PCL library. Thank you in advance !

Chimdi511, 2018-06-27:
To be more specific,, this is where the program crashes in RealSense. 

`void playback_sensor::handle_frame(frame_holder frame, bool is_real_time)
{
    if(frame == nullptr)
    {
        throw invalid_value_exception("null frame passed to handle_frame");
    }
    if(m_is_started)
    {
        frame->get_owner()->set_sensor(shared_from_this());
        auto type = frame->get_stream()->get_stream_type();
        auto index = static_cast<uint32_t>(frame->get_stream()->get_stream_index());
        frame->set_stream(m_streams[std::make_pair(type, index)]);
        frame->set_sensor(shared_from_this());
        auto stream_id = frame.frame->get_stream()->get_unique_id();
        //TODO: Ziv, remove usage of shared_ptr when frame_holder is cpoyable
        auto pf = std::make_shared<frame_holder>(std::move(frame));
        m_dispatchers.at(stream_id)->invoke([this, pf](dispatcher::cancellable_timer t)
        {
            frame_interface* pframe = nullptr;
            std::swap((*pf).frame, pframe);
            **m_user_callback->on_frame((rs2_frame*)pframe);**
        });
        if(is_real_time)
        {
            m_dispatchers.at(stream_id)->flush();
        }
    }
}`

The bolded line is where the program crashes.

Chimdi511, 2018-06-28:
To give more information, this is the error I get.

> Unhandled exception thrown: read access violation.
this->_Mypair._Myval2.__this->m_user_callback._Ptr was nullptr.

RealSense-Customer-Engineering, 2018-06-29:
[Realsense Customer Engineering Team Comment]
@Chimdi511
Can you attach your full code if possible, since I've done a experiment to import 4 bag files and render them for few minutes, no issue happened. I attach the code I revised from rs-record-playback project, please have a test with four bag files along with the binary called "0.bag", "1.bag", "2.bag", "3.bag"(please record the bag file before launch the application). attached below

carypai, 2018-06-29:
[rs-record-playback.zip](https://github.com/IntelRealSense/librealsense/files/2148365/rs-record-playback.zip)
[image]

Chimdi511, 2018-06-29:
I changed my code to include pipe.stop() and now it doesn't crash anymore.

RealSense-Customer-Engineering, 2018-07-02:
[Realsense Customer Engineering Team Comment]
@Chimdi511
Glad to see your problem fixed, close this ticket.

closed 2018-07-02 by RealSense-Customer-Engineering