realsenseai/librealsense #1966: Android, "failed to initialize libuvc"

aarun2211, 2018-06-27:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D415  | 
| Firmware Version                   | 05.08.15.00| 
| Operating System & Version |  Linux (Ubuntu16.04) | 
| Kernel Version (Linux Only)    |  4.13.0                                        | 
| Platform                                 | Android, Oreo 8.1.0  |
| SDK Version                            |  2                        | 
| Language                            |  C                        | 
| Segment			|  VR                         | 

### Issue Description
I am trying to connect the D415 to a Rooted Nexus 6P to the Realsense D415. I used the Easy OTG Checker to make sure the camera is discovered by the phone. I used chmod to give read, write and execute access to the USB devices (as mentioned in the Readme). I also changed the owner from root to a non-super user. Despite this, when I open the realsense_app, it mentions "failed to initialize libuvc Last Error: Permission denied". Any help would be appreciated.
Thanks.

RealSense-Customer-Engineering, 2018-07-03:
[Realsense Customer Engineering Team Comment]
Please try disable SELinux(SEAndroid)

setenforce 0

And take permission for device

chmod -R 777 /dev/bus/usb/

aarun2211, 2018-07-03:
Hello, 
Thank you for the response. I tried this, and the camera seems to turn on as I can see a faint red glow on the camera's IR LED array, but the application seems to be crashing when I open the app. I'm having a similar issue as #1955. I would appreciate if this issue could be looked into. 
Thanks!

RealSense-Customer-Engineering, 2018-07-05:
[Realsense Customer Engineering Team Comment]
@aarun2211 The failure of this ticket should be resolved. Could you please open another ticket to track the crash issue and provide the crash logs(logcat and dmesg) for the crash in the new ticket? Thanks!