realsenseai/librealsense #1967: Please fix asymmetry in rs::context notifications about devices being added / removed

BrendanDrew, 2018-06-27:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 series (D400, D415, D435) | 
| Firmware Version                   | N/A | 
| Operating System & Version |   All  | 
| Kernel Version (Linux Only)    |  All                                         | 
| Platform                                 | All  |
| SDK Version                            |  Master as of 719b0b95cdfd6a5c9a3a07e34bea48fce5f35d2a                          | 
| Language                            |  C++, Python                         | 
| Segment			|  All                          | 

### Issue Description
The C++ (and, consequently, python API which is what I actually care about) for notifications when devices are added / removed is asymmetric. The events expose a list of newly added devices but force you to query for the removal of specific devices. I would like these to be made symmetric in the following ways:

1. Add a method like `bool was_added(const rs2::device& dev) const` to see if a specific device has been re-added (e.g. after issuing a software reset to the device). The signature may need to change (e.g. instead of assuming you have a live device handle, support doing this by serial number maybe since the handles must have potentially been invalidated by the reset event)

2. Add a method like `device_list get_removed_devices() const`

3. Update all the bindings to reflect these changes please :-)

RealSense-Customer-Engineering, 2018-06-29:
[Realsense Customer Engineering Team Comment]
hi @BrendanDrew,
thanks for your input, may I know your need for this?
it's nice to have or necessary for what usage?
thanks.

dorodnic, 2018-07-11:
The PR was merged, @BrendanDrew your contribution is appreciated. If you decide to share some kind of Gtk "template" for multi-camera usage it would also make a good optional example / tutorial.

closed 2018-07-11 by dorodnic