realsenseai/librealsense #1969: start(S on_frame)  upcasting callback

dmitryshendryk, 2018-06-28:
|---------------------------------|------------------------------------------- |
| Camera Model D415
| Firmware Version (Open RealSense Viewer --> Click info)
| Operating System & Version Ubuntu 16.04
| Kernel Version (Linux Only) 4.14.13 |
| Platform PC
| SDK Version 2
| Language C++
| Segment	others

Hello. Have the issue with callback, i declare base struct and operator inside and want to overload it, but the start() function for some reason upcasting it to base struct. Can anyone explain why?

**realsense.hpp**
`RealSense::RealSense(int width, int height, int fps, int depth_profile, int color_profile)
    : _width(width), _height(height), _fps(fps)
    {
        if (ctx.query_devices().size() == 0)
        {
            cout << "RealSense device not detected" << "\n";
            exit(0);
        }
    cfg.enable_stream(RS2_STREAM_DEPTH, width, height, RS2_FORMAT_Z16, fps);
    cfg.enable_stream(RS2_STREAM_COLOR, width, height, RS2_FORMAT_RGB8, fps);

    dev = ctx.query_devices().front();
    depth_sensor = dev.query_sensors()[0].as<rs2::depth_sensor>();
    color_sensor = dev.query_sensors()[1].as<rs2::sensor>();

    rs2::stream_profile depth_stream_profile = depth_sensor.get_stream_profiles()[depth_profile];
    rs2::stream_profile color_stream_profile = color_sensor.get_stream_profiles()[color_profile];

    depth_sensor.open(depth_stream_profile);
    color_sensor.open(color_stream_profile);

 
    };

 void RealSense::startFrames(RGBCallBack *rgb, DepthCallBack *depth)
 {
    // This should be called Custom RGB and Custom Depth
   // But it gives base RGBCallBack and DepthCallBack ----- shouldn't be like thos 
      depth_sensor.start(*rgb);
      color_sensor.start(*depth);
 }

 RealSense::~RealSense()
    {
        depth_sensor.stop();
        color_sensor.stop();
    };`

**realsense.hpp**

`  struct RGBCallBack
    {
        public:
            RGBCallBack() {};

    virtual void operator()(rs2::video_frame frame)
        {
            cout << "RGBCallBack" << "\n";
        }    
    };

    struct DepthCallBack
    {
        public:
            DepthCallBack() {};

    virtual  void operator()(rs2::video_frame frame) 
        {
            cout << "DepthCallBack" << "\n";
        } 

    }; 

class RealSense {

    
    public:
        RealSense(int width= 640, int height=480, int fps=30,
         int depth_profile=208, int color_profile=23);
        ~RealSense();

    void startFrames(RGBCallBack *rgb, DepthCallBack *depth);   
    
    private:
        int _width;
        int _height;
        int _fps; 
        rs2::context ctx;   
        rs2::config cfg;
        rs2::sensor color_sensor;
        rs2::sensor depth_sensor;
        rs2::device dev;

};
`

**main**

`
struct CustomRGBCallBack : RGBCallBack   
{
  public:
  CustomRGBCallBack():RGBCallBack()  {};
 
   void operator()(rs2::video_frame frame)
  {
      cout << "Custom RGB" << "\n";
  }
};

struct CustomDepthCallBack :DepthCallBack
{
  public:
    CustomDepthCallBack():DepthCallBack(){};
  
   void operator()(rs2::video_frame frame) 
  {
    cout << "Custom Depth" << "\n";
  }
};

int main(int argc, char * argv[]) 
{ 
   RealSense camera(640, 480, 30);

   WindowGL app(600, 400, "Test OpenGL");
    
   rs2::colorizer color_map;

  RGBCallBack  *rgb = new CustomRGBCallBack();
  DepthCallBack *depth =new  CustomDepthCallBack();
  
  camera.startFrames(rgb, depth);

     while(app) {

     }
    return 0;
}`

dorodnic, 2018-07-01:
To the best of my understanding, we are not assuming `operator()` has to be virtual, and hence virtual call is not generated here. 
That said, the generic syntax is mainly in-place to enable using C++11 lambdas. 
If lambdas are supported by your compiler, you could change:
`depth_sensor.start(*rgb);`
to something like:
`depth_sensor.start([&](){ rgb->operator(); });
If lambdas are unavailable, you could create a wrapper class that takes your base class by reference and issues a virtual call in its `operator()`.

dmitryshendryk, 2018-07-02:
Here the results
Using  `color_sensor.start([&](rs2::frame f){ rgb->operator()(); });`  i still can't get call of my custom struct, only base operator get called.

Now i've defined virtual function 
```
virtual  void rgbTest(rs2::frame f)
     {
             cout << "Base RGB" << "\n";
     }

virtual  void depthTest(rs2::frame f)
     {
             cout << "Base Depth" << "\n";
     }
```

But there start some problems,
If i use only one sensor and one callback, it' ok
` color_sensor.start([&](rs2::frame f){ rgb->rgbTest(f); });
` 
If i use two sensors and two callbacks then i get Segmentation error
```
color_sensor.start([&](rs2::frame f){ rgb->rgbTest(f); });
   depth_sensor.start([&](rs2::frame f){ depth->depthTest(f); });
```
Even like this get Segmentation error 

 ```
color_sensor.start([&](rs2::frame f){ rgb->rgbTest(f); });
   depth_sensor.start([&](rs2::frame f){ cout << f << "\n"; });
```
But without callbacks its ok

```
color_sensor.start([&](rs2::frame f){  cout << f << "\n"; });
depth_sensor.start([&](rs2::frame f){ cout << f << "\n"; });

```

Any idea why ?

RealSense-Customer-Engineering, 2018-07-03:
[Realsense Customer Engineering Team Comment]
@dmitryshendryk
I made a sample for you, please refer to the attachment below.

carypai, 2018-07-03:
[test.zip](https://github.com/IntelRealSense/librealsense/files/2158164/test.zip)

dmitryshendryk, 2018-07-04:
Just one line of the code, helped a lot, thanks :) . I have last question regarding rendering color and depth. I've used this code for rendering https://github.com/IntelRealSense/librealsense/blob/master/examples/example.hpp 

In my callback 
```
virtual void rgbTest(rs2::frame f)
	{
        color_image.render(f, { 1280 / 2, 0, 1280 / 2, 720 });
		cout << "Base RGB" << "\n";
	}
```

But for some reason the  glGenTextures  returns 0, i read there may be the reason, that it's not called in a thread where the opneGL context was created. I haven't found any examples how to do rendering asynchronous. Any idea?

dorodnic, 2018-07-04:
Hi @dmitryshendryk 
With OpenGL you can't do rendering asynchronously, all OpenGL calls must be on the same thread. 
That said, you can work with the camera asynchronously, but then do rendering on the main thread.
For this purpose we provide `frame_queue` primitive - in your callback do `q.enqueue(f);` and on the main OpenGL thread you can either wait or poll frames from that queue and render them.

dmitryshendryk, 2018-07-07:
@dorodnic 
Thanks a lot. Helped!

closed 2018-07-07 by dmitryshendryk