realsenseai/librealsense #1972: ROS2 debian support?

testkit, 2018-06-28:
We have [ROS2 realsense ](https://github.com/intel/ros2_intel_realsense)enabled based on librealsense 2.9.1, and we'd like to make ROS2 realsense debian package. Is there any plan to provide ROS2 librealsense package recently?

dorodnic, 2018-07-02:
Hi @testkit 
Yes, we plan to provide it as soon as possible. We had some setbacks in ROS support lately, but Debian for ROS is planned and will come within one-two months.

testkit, 2018-07-03:
@dorodnic, thanks for your reply! I'd like to double check you would provide both ROS and ROS2 debian packages, right? The ROS2 packaging guideline is at https://github.com/ros2/ros2/wiki/Releasing-a-ROS-2-package-with-bloom.

RealSense-Customer-Engineering, 2018-08-24:
[Realsense Customer Engineering Team Comment]
Hi @testkit,

For ROS2, please create a ROS2 GitHub issue to track the request: https://github.com/intel/ros2_intel_realsense/issues

closed 2018-08-24 by RealSense-Customer-Engineering

testkit, 2018-08-27:
@RealSense-Customer-Engineering, I'm talking about ROS2 debian package for librealsense but not ROS2 wrapper for librealsense. Similar to ROS1 debian package for librealsense this project did before, see branch [rosdebian](https://github.com/IntelRealSense/librealsense/tree/rosdebian).