realsenseai/librealsense #1977: Inquries regarding scanning technology found in SR300, R200

wjdghks0721, 2018-06-29:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | ALL SDK2.0  | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

 
To whomever it may concern,

Good Day. My name is Junghwan Hyun, writing from Korea regarding your products. We're currently doing R&D to use Intel's RealSense cameras to use in hospitals to help patients rehabilitate. We're planning to use the 3D scanning technology to measure patients' body types.

I was wondering if the 3D scanning technology found in Intel RealSense Cameras SR300 and R200 models will be supported in Intel SDK2.0 in the new D400 series, and if there are plans for additional development.

I'd appreciate your answer. Thanks.

Cordially,
Junghwan Hyun

RealSense-Customer-Engineering, 2018-07-05:
[Realsense Customer Engineering Team Comment]
hi wjdghks0721,

The RealSense SDK 2.0 is focused on providing depth across multiple operating systems and wrapper. All open source. We provide a few code samples which we hope the community will add to. We also provide tools like the viewer. What we are hoping to do is expand into middleware like person tracking or scanning via third-party partners. Please watch our site as we bring these partners on board.

if you are NDA customer to Intel, would suggest you to check with our sales channel for further discussion.

RealSense-Customer-Engineering, 2018-07-13:
[Realsense Customer Engineering Team Comment]
let me close this ticket here, please feel free to open new ticket if any question.

closed 2018-07-13 by RealSense-Customer-Engineering