realsenseai/librealsense #1979: Patch for Odroid

ipa-fog, 2018-06-29:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   | SDK2.0, ROS | 
| Operating System & Version |   Ubuntu16.04 | 
| Kernel Version (Linux Only)    |  4.14.47-132                                       | 
| Platform                                 | Odroid-XU4|
| SDK Version                            |  { legacy / 2.13.0 }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			| Robot                          | 

### Issue Description
I tried to install the SDK on an Odroid-XU4. When I execute the patch  `./scripts/patch-realsense-ubuntu-odroid.sh ` no errors occurred but it finished incomplete. 
* In Line 36 I don t get into the if-statement because my hostname (uname -n) is not equal. I changed it in my local File but thats not nice.
* In Line 107 and 108 link to an not existing path. In path `usr/src`  is only a different Kernel folder ` linux-headers-4.4.0-128` without `.config` and `/Module.symvers` Files. Therefore the Patch stops here. 

Could somebody help me please?

Without the Patch I get really strange depth images due to wrong format encodings.

Thanks!

RealSense-Customer-Engineering, 2018-07-04:
[Realsense Customer Engineering Team Comment]
Hi @ipa-fog,

It looks the patch script has some problems there. However, although no patch applied, you should be able to get the normal depth stream. Can you try "rs-capture" first to see if the depth stream is ok?

RealSense-Customer-Engineering, 2018-07-05:
[Realsense Customer Engineering Team Comment]
Hi @ipa-fog,

As for the patch script, there are some modifications required. you need to get the kernel from below source. Ultimately, you should be able to build 'uvcvideo.ko' to patch some new camera formats. Also, suggest to upgrade to Linux odroid 4.14.52-145 which is the latest. For the file 'Module.symvers', a quick way is to get it from /lib/modules/4.14.xxx/build**.**

kernel_name="odroid_bionic"
ubuntu_codename="bionic"
#[! -d ${kernel_name}] && git clone -b $kernel_branch git://#kernel.ubuntu.com/ubuntu/ubuntu-${ubuntu_codename}.git --depth #1 ./${kernel_name}
[! -d ${kernel_name}] && git clone --depth 1 https://github.com/hardkernel/linux -b odroidxu4-4.14.y ./${kernel_name}

However, still let me know what you mean about "strange depth images due to wrong format encodings". Any error log showing if you run 'rs-capture' or 'realsense-viewer'?

ipa-fog, 2018-07-05:
Hi @RealSense-Customer-Engineering 

rs-capture is not working because I am using SSH and got this error:

```
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
Segmentation fault
```
But the `rs-depth` and `rs-distance` tools are working fine.

I visualize the Depth image with RVIZ (ROS):
https://github.com/ipa-fog/cam_issue/blob/master/Screenshot%20from%202018-07-05%2014-24-58.png

Thanks for your code modification. But now the script break at

ipa-fog, 2018-07-05:
dmesg output when I plugin the camera: 

```
[ 4046.035095] usb 4-1.2: new SuperSpeed USB device number 7 using xhci-hcd
[ 4046.056042] usb 4-1.2: New USB device found, idVendor=8086, idProduct=0b07
[ 4046.056050] usb 4-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 4046.056056] usb 4-1.2: Product: Intel(R) RealSense(TM) 435
[ 4046.056061] usb 4-1.2: Manufacturer: Intel(R) RealSense(TM) 435
[ 4046.056066] usb 4-1.2: SerialNumber: 813513021477
[ 4046.059731] uvcvideo: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[ 4046.059745] uvcvideo: Unknown video format 00000032-0000-0010-8000-00aa00389b71
[ 4046.059962] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) 435 (8086:0b07)
[ 4046.060811] uvcvideo: Unable to create debugfs 4-7 directory.
[ 4046.060927] uvcvideo 4-1.2:1.0: Entity type for entity Intel(R) RealSense(TM) 435 with was not initialized!
[ 4046.060934] uvcvideo 4-1.2:1.0: Entity type for entity Processing 2 was not initialized!
[ 4046.060943] uvcvideo 4-1.2:1.0: Entity type for entity Intel(R) RealSense(TM) 435 with was not initialized!
[ 4046.060949] uvcvideo 4-1.2:1.0: Entity type for entity Camera 1 was not initialized!
[ 4046.061175] input: Intel(R) RealSense(TM) 435: Int as /devices/platform/soc/soc:usb3-0/12000000.dwc3/xhci-hcd.3.auto/usb4/4-1/4-1.2/4-1.2:1.0/input/input5
[ 4046.062274] uvcvideo: Unknown video format 36315752-1a66-a242-9065-d01814a8ef8a
[ 4046.062285] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) 435 (8086:0b07)
[ 4046.062657] uvcvideo: Unable to create debugfs 4-7 directory.
[ 4046.062819] uvcvideo 4-1.2:1.3: Entity type for entity Processing 7 was not initialized!
[ 4046.062828] uvcvideo 4-1.2:1.3: Entity type for entity Extension 8 was not initialized!
[ 4046.062836] uvcvideo 4-1.2:1.3: Entity type for entity Camera 6 was not initialized!

```

dmesg when the Camera is running with ROS
```

[ 4128.110902] ------------[ cut here ]------------
[ 4128.110933] WARNING: CPU: 1 PID: 20529 at drivers/media/v4l2-core/v4l2-ioctl.c:1288 v4l_enum_fmt+0x1034/0x1780
[ 4128.110938] Unknown pixelformat 0x00000000
[ 4128.110941] Modules linked in: fuse tun uvcvideo videobuf2_vmalloc cfg80211 rfkill spidev binfmt_misc evdev spi_s3c64xx uio_pdrv_genirq uio gpio_keys extcon_usb_gpio exynos_gpiomem ipv6
[ 4128.110991] CPU: 1 PID: 20529 Comm: nodelet Tainted: G        W       4.14.40-131 #1
[ 4128.110994] Hardware name: ODROID-XU4
[ 4128.111016] [<c0110b14>] (unwind_backtrace) from [<c010ce98>] (show_stack+0x10/0x14)
[ 4128.111030] [<c010ce98>] (show_stack) from [<c08a7ff4>] (dump_stack+0x84/0x98)
[ 4128.111046] [<c08a7ff4>] (dump_stack) from [<c01242cc>] (__warn+0xec/0x104)
[ 4128.111056] [<c01242cc>] (__warn) from [<c012431c>] (warn_slowpath_fmt+0x38/0x48)
[ 4128.111066] [<c012431c>] (warn_slowpath_fmt) from [<c069bff0>] (v4l_enum_fmt+0x1034/0x1780)
[ 4128.111077] [<c069bff0>] (v4l_enum_fmt) from [<c0699aac>] (__video_do_ioctl+0x2a0/0x2fc)
[ 4128.111086] [<c0699aac>] (__video_do_ioctl) from [<c06993d8>] (video_usercopy+0x264/0x68c)
[ 4128.111097] [<c06993d8>] (video_usercopy) from [<c0297590>] (do_vfs_ioctl+0x9c/0x8e0)
[ 4128.111106] [<c0297590>] (do_vfs_ioctl) from [<c0297e08>] (SyS_ioctl+0x34/0x5c)
[ 4128.111117] [<c0297e08>] (SyS_ioctl) from [<c0108780>] (ret_fast_syscall+0x0/0x54)
[ 4128.111122] ---[ end trace 48b4c1154651863d ]---
[ 4128.113272] ------------[ cut here ]------------
[ 4128.113296] WARNING: CPU: 1 PID: 20529 at drivers/media/v4l2-core/v4l2-ioctl.c:1288 v4l_enum_fmt+0x1034/0x1780
[ 4128.113301] Unknown pixelformat 0x00000000
[ 4128.113304] Modules linked in: fuse tun uvcvideo videobuf2_vmalloc cfg80211 rfkill spidev binfmt_misc evdev spi_s3c64xx uio_pdrv_genirq uio gpio_keys extcon_usb_gpio exynos_gpiomem ipv6
[ 4128.113350] CPU: 1 PID: 20529 Comm: nodelet Tainted: G        W       4.14.40-131 #1
[ 4128.113354] Hardware name: ODROID-XU4
[ 4128.113372] [<c0110b14>] (unwind_backtrace) from [<c010ce98>] (show_stack+0x10/0x14)
[ 4128.113385] [<c010ce98>] (show_stack) from [<c08a7ff4>] (dump_stack+0x84/0x98)
[ 4128.113399] [<c08a7ff4>] (dump_stack) from [<c01242cc>] (__warn+0xec/0x104)
[ 4128.113409] [<c01242cc>] (__warn) from [<c012431c>] (warn_slowpath_fmt+0x38/0x48)
[ 4128.113419] [<c012431c>] (warn_slowpath_fmt) from [<c069bff0>] (v4l_enum_fmt+0x1034/0x1780)
[ 4128.113430] [<c069bff0>] (v4l_enum_fmt) from [<c0699aac>] (__video_do_ioctl+0x2a0/0x2fc)
[ 4128.113438] [<c0699aac>] (__video_do_ioctl) from [<c06993d8>] (video_usercopy+0x264/0x68c)
[ 4128.113448] [<c06993d8>] (video_usercopy) from [<c0297590>] (do_vfs_ioctl+0x9c/0x8e0)
[ 4128.113456] [<c0297590>] (do_vfs_ioctl) from [<c0297e08>] (SyS_ioctl+0x34/0x5c)
[ 4128.113467] [<c0297e08>] (SyS_ioctl) from [<c0108780>] (ret_fast_syscall+0x0/0x54)
[ 4128.113471] ---[ end trace 48b4c1154651863e ]---
[ 4128.140931] ------------[ cut here ]------------
[ 4128.140962] WARNING: CPU: 1 PID: 20529 at drivers/media/v4l2-core/v4l2-ioctl.c:1288 v4l_enum_fmt+0x1034/0x1780
[ 4128.140967] Unknown pixelformat 0x00000000
[ 4128.140971] Modules linked in: fuse tun uvcvideo videobuf2_vmalloc cfg80211 rfkill spidev binfmt_misc evdev spi_s3c64xx uio_pdrv_genirq uio gpio_keys extcon_usb_gpio exynos_gpiomem ipv6
[ 4128.141021] CPU: 1 PID: 20529 Comm: nodelet Tainted: G        W       4.14.40-131 #1
[ 4128.141024] Hardware name: ODROID-XU4
[... 97 lines cut ...]
[ 4128.262583] WARNING: CPU: 4 PID: 20529 at drivers/media/v4l2-core/v4l2-ioctl.c:1288 v4l_enum_fmt+0x1034/0x1780
[... cut ...]
[ 4128.344387] uvcvideo: Failed to query (SET_CUR) UVC control 11 on unit 3: -32 (exp. 1).
[ 4128.401725] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 4128.458845] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 4128.522703] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 4128.581045] uvcvideo: Failed to query (SET_CUR) UVC control 3 on unit 3: -32 (exp. 4).
[ 4128.637809] uvcvideo: Failed to query (SET_CUR) UVC control 3 on unit 3: -32 (exp. 4).

```

RealSense-Customer-Engineering, 2018-07-05:
[Realsense Customer Engineering Team Comment]
Hi @ipa-fog,

Per my quick test of kernel 4.14.37-135 and 4.14.52-145, 'rs-capture' tool can work normally. Need to further check your issue.

I will also create a modified patch script for reference. Stay tuned.

RealSense-Customer-Engineering, 2018-07-08:
[Realsense Customer Engineering Team Comment]
Hi @ipa-fog,

Please try below script for Odroid XU4. Note: This is verified based on kernel 4.14.52-145.
https://github.com/freemanlo/librealsense/blob/odroid_build/scripts/patch-realsense-ubuntu-odroid-xu4-4.14.sh

RealSense-Customer-Engineering, 2018-07-10:
[Realsense Customer Engineering Team Comment]
Hi @ipa-fog,

How does it work for you?

ipa-fog, 2018-07-10:
Hi @RealSense-Customer-Engineering,
I tried your patch on a fresh Ubuntu installation. The script is running fine. I only had to modify Line 107 and 108 because in `usr/src ` was again no folder of the current kernel . I set the link as you proposed to:

```
# Copy configuration
sudo cp /lib/modules/$(uname -r)/build/.config .
sudo cp /lib/modules/$(uname -r)/build/Module.symvers .
```

Finally when I plug in the camera and run dmesg I get no more "unknown format errors"

```

[ 3194.498384] usb 4-1.2: new SuperSpeed USB device number 4 using xhci-hcd
[ 3194.519088] usb 4-1.2: New USB device found, idVendor=8086, idProduct=0b07
[ 3194.519096] usb 4-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 3194.519101] usb 4-1.2: Product: Intel(R) RealSense(TM) 430
[ 3194.519106] usb 4-1.2: Manufacturer: Intel(R) RealSense(TM) 430
[ 3194.519110] usb 4-1.2: SerialNumber: 735413020665
[ 3194.529072] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) 430 (8086:0b07)
[ 3194.529880] uvcvideo: Unable to create debugfs 4-4 directory.
[ 3194.529978] uvcvideo 4-1.2:1.0: Entity type for entity Intel(R) RealSense(TM) 430 with was not initialized!
[ 3194.529985] uvcvideo 4-1.2:1.0: Entity type for entity Processing 2 was not initialized!
[ 3194.529991] uvcvideo 4-1.2:1.0: Entity type for entity Intel(R) RealSense(TM) 430 with was not initialized!
[ 3194.529997] uvcvideo 4-1.2:1.0: Entity type for entity Camera 1 was not initialized!
[ 3194.530184] input: Intel(R) RealSense(TM) 430: Int as /devices/platform/soc/soc:usb3-0/12000000.dwc3/xhci-hcd.3.auto/usb4/4-1/4-1.2/4-1.2:1.0/input/input2
[ 3194.531168] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) 430 (8086:0b07)
[ 3194.531618] uvcvideo: Unable to create debugfs 4-4 directory.
[ 3194.531748] uvcvideo 4-1.2:1.3: Entity type for entity Processing 7 was not initialized!
[ 3194.531756] uvcvideo 4-1.2:1.3: Entity type for entity Extension 8 was not initialized!
[ 3194.531763] uvcvideo 4-1.2:1.3: Entity type for entity Camera 6 was not initialized!
```

closed 2018-07-10 by ipa-fog

amerkay, 2018-09-13:
Thanks @RealSense-Customer-Engineering for the updated patch file. I tried several times to go through the compilation, I even built kernel 4.14.52-145 as you tested the installation on it, and I keep hitting the dead end "uvcvideo: Unable to create debugfs 3-3 directory". I believe that's the same issue as @ipa-fog.

Running `dmesg`:
```
[  125.039039] usb 4-1.2: USB disconnect, device number 3
[  169.135360] usb 3-1.2: new high-speed USB device number 3 using xhci-hcd
[  169.236109] usb 3-1.2: New USB device found, idVendor=8086, idProduct=0b07
[  169.236116] usb 3-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  169.236121] usb 3-1.2: Product: Intel(R) RealSense(TM) 435
[  169.236127] usb 3-1.2: Manufacturer: Intel(R) RealSense(TM) 435
[  169.251469] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) 435 (8086:0b07)
[  169.252232] uvcvideo: Unable to create debugfs 3-3 directory.
[  169.252334] uvcvideo 3-1.2:1.0: Entity type for entity Intel(R) RealSense(TM) 435 with was not initialized!
[  169.252341] uvcvideo 3-1.2:1.0: Entity type for entity Processing 2 was not initialized!
[  169.252347] uvcvideo 3-1.2:1.0: Entity type for entity Intel(R) RealSense(TM) 435 with was not initialized!
[  169.252353] uvcvideo 3-1.2:1.0: Entity type for entity Camera 1 was not initialized!
[  169.252574] input: Intel(R) RealSense(TM) 435: Int as /devices/platform/soc/soc:usb3-0/12000000.dwc3/xhci-hcd.3.auto/usb3/3-1/3-1.2/3-1.2:1.0/input/input2
[  169.253479] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) 435 (8086:0b07)
[  169.253843] uvcvideo: Unable to create debugfs 3-3 directory.
[  169.253948] uvcvideo 3-1.2:1.3: Entity type for entity Processing 7 was not initialized!
[  169.253955] uvcvideo 3-1.2:1.3: Entity type for entity Extension 8 was not initialized!
[  169.253961] uvcvideo 3-1.2:1.3: Entity type for entity Camera 6 was not initialized!
```

Any help would be appreciated.

freemanlo, 2018-09-21:
Hi @amerkay,

I have modified the patch script according to your comment. Do you successfully apply the patch and D435 can stream normally?

amerkay, 2018-09-27:
Hey @freemanlo,

Yes. It worked fine. Note it did not work on the 20171213 image minimal, as the /lib/modules/4.14.69-135/build was not available, that's the minimal ODROID image.

abylikhsanov, 2019-03-01:
Hi,

Did anyone tried on 4.14.94-155? I am able to compile but when I run ROS, the camera cannot be detected.