realsenseai/librealsense #1980: how to build realsense-viewer and library version 2.13.0

impet14, 2018-06-29:
I have built the package using cmake and make it. I could not get the binary file of realsense-viewer. 
Could anyone give me a suggestion?

dorodnic, 2018-07-01:
If you build inside `~/librealsense/build/` the viewer should be under `~/librealsense/build/tools/realsense-viewer/`.

RealSense-Customer-Engineering, 2018-07-02:
[Realsense Customer Engineering Team Comment]
hi @impet14,

would like to know which OS you are using and your question is you can't find the binary or you see some error message?

impet14, 2018-07-02:
Thanks, everyone. Now I could build the realsense-viewer by following the tutorial on how to build from source.

RealSense-Customer-Engineering, 2018-07-05:
[Realsense Customer Engineering Team Comment]
hi @impet14,

that's great, so do you need more support on this topic or ok for me to close this ticket?

dorodnic, 2018-07-11:
@impet14 feel free to open new issue if you need more help

closed 2018-07-11 by dorodnic