realsenseai/librealsense #1982: PointCloud misalignment

ljc19800331, 2018-06-30:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  SR300 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Linux (Ubuntu 16.04)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {python }                          | 
| Segment			|  {Robot }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
Hello, 
I am currently facing a problem when mapping the color image to the point cloud and I found small misalignment between the image and the point cloud. My goal is to choose a rectangular region in the image and output the corresponding 3D point cloud in this region. However, the textured point cloud of the rectangular region is misaligned with the region in the image. I am writing this script in python. This is main part of my script: 

###################################################################
align_to = rs.stream.color
align = rs.align(align_to)
aligned_frames = align.process(frames)
aligned_depth_frame = aligned_frames.get_depth_frame()         
color_frame = aligned_frames.get_color_frame()
color_image = np.asanyarray(color_frame.get_data())

# The rectangular region
rows = list(range(y, y + H))
cols = list(range(x, x + W))

for i in rows:
for j in cols:
depth = depth_frame.get_distance(j, i)
depth_point = rs.rs2_deproject_pixel_to_point(depth_intrin, [j, i], depth)
color_point = rs.rs2_transform_point_to_point(depth_to_color_extrin, depth_point)
pixel_point = rs.rs2_project_point_to_pixel(color_intrin, color_point)
j_idx = pixel_point[0]
i_idx = pixel_point[1]
if (np.isnan(i_idx) or np.isnan(j_idx) or j_idx > 640 or i_idx > 480):
                i_idx = 0
                j_idx = 0
colorvec = color_image[np.int(i_idx), np.int(j_idx), :]
Pc[count, :] = depth_point
Color_Vec[count, :] = colorvec
count += 1
###################################################################

question 1: Is the color frame from the stereo camera different from the middle camera (The camera that takes the standard image) 
question 2: Is there a good way to choose a random region in the image and return the corresponding 3D point sets in python? 
question 3: When I use realsense-viewer in ubuntu, I still find the misalignment with the point cloud. Is this a problem with the camera, as shown in the following figure 3, the left edge of the white paper is misalignment with the point cloud? 

Thank you for the answer.

[image]
[image]
[image]

ev-mp, 2018-08-21:
Hello @ljc19800331,
There seem to be several questions entangled:
>question 1: Is the color frame from the stereo camera different from the middle camera (The camera that takes the standard image)

SR300 device has two sensors: Depth/IR and RGB. The third "eye" in the front panel is the structured light pattern  projector. So the meaning of `...color frame from the stereo` and `...middle camera` is not clear. Please elaborate

>question 2: Is there a good way to choose a random region in the image and return the corresponding 3D point sets in python?

In case you refer to choosing a random region in RGB image, then using the `align` object will allow to obtain the corresponding depth region in a straight-forwarded manner. Note that generating pointcloud from "depth aligned to color" will produce slightly different results when compared with pointcloud from original depth frame. 
This is due to the fact that "aligned depth" pixels must reside on the grid of the reference (color) frame.
And rounding-up/down depth coordinates during texture mapping will manifest in certain misalignment when converting "Aligned depth" 2D image to 3D pointcloud.

>question 3: When I use realsense-viewer in ubuntu, I still find the misalignment with the point cloud. Is this a problem with the camera, as shown in the following figure 3, the left edge of the white paper is misalignment with the point cloud?

I can see some pixels off-plane in image 2 (pointcloud), but it is a depth map artifact that also could be due post-processing filters
[image]
Can you mark specifically what you refer as misalignment in image no 3?

Referring the code,
>colorvec = color_image[np.int(i_idx), np.int(j_idx), :]

The coordinates are trimmed rather than rounded up/down, i.e. [6.99]->[7~~6~~]  . This also produce some errors

ljc19800331, 2018-08-24:
@ev-mp Thank you so much for the answer. I still have some questions for this problem.

1. Actually the image 3 misalignment is similar to the image 2 which shows the depth map artifacts. I am not sure whether this artifact can be fixed by the software or some post-processing filter. Or this is a normal artifact due to the limitation of the hardware (I mean it is difficult to generate depth image without any problems). Is there any solution we can reduce the effect of this artifact? Since I am doing some medical application, a small artifact will make a huge influence (Maybe I need to choose another camera).

2. For the depth-to-image alignment, I am using the textures and vertices output from python. I first choose the texture coordinates for a specific region, then convert these coordinates to the pixel coordinates. The corresponding index are used for the vertices and thus I can generate the colorized point cloud. I am not sure whether this is a correct to reduce the artifact effect shown in image 2 and 3. I am still looking for the solution.

3. Yes thanks for the advice. I was originally thinking this product is simply an integration system of three cameras -- RGB, left and right color imager (where it can generate the stereo images for 3D reconstruction. I finally realized this is a IR red camera and projector. 

Thank you so much for the answer.

ev-mp, 2018-09-02:
Hello @ljc19800331  and sorry for the delay,
It is still not clear what is the misalignment you describe. 
It looks as if the screenshots are from different scenes, or at least I cannot correlate across images:
 - The numbers scale appears in image 1 and 3 but not in 2.
 - The grid appear in image 2 only.
 - The red frame appears in image 3 but is absent is 1 and 2.

Please take a snapshot both in 2d and 3d with `realsense-viewer` of a single coherent scene and mark the artifacts you're trying to investigate so we can isolate and zoom on it. You may also use the viewer's embedded recorder and attach a bag file for reproduction.

ljc19800331, 2018-09-03:
@ev-mp Thanks for your reply. I have labelled the artifact region shown in the following figures. When I extract the texture and vertex coordinates and try to reconstruct the colorized point cloud, it shows the same result as it does in realsense-viewer (i.e. the artifact is showing). Thanks again for your reply.
 
[image]
[image]
[image]

dorodnic, 2018-10-11:
We added some fixes in this area of code in latest release (2.16.1), but I think this is just the limit of specific camera calibration. Neither the depth nor the depth-to-color alignment are perfect, and it can vary on unit to unit basis.

closed 2018-12-12 by RealSense-Customer-Engineering