realsenseai/librealsense #1983: The filters in the managed SDK leak memory

smirkingman, 2018-07-01:
Camera Model Intel Realsense D435
Firmware Version 05.09.11.0 
Operating System  Windows 10
Version 10.0.17134 Build 17134
Platform PC i7-6770HQ 
Environment Visual Studio 2017 15.7.4
SDK Version 2.13.0
Language VB.Net 4.6.1
 Segment Robot                     

### Issue Description
The filters in the managed SDK leak memory, approximately 11Mb/minute at 30FPS with all the filters enabled:
[image]
This is the complete console app, un-comment as suggested to see memory slowly leaking (run for **at least** 30 seconds to observe the issue):

`
Imports Intel.RealSense
Module Module1
    Sub Main()
        Dim pipeline As New Pipeline()
        Dim coloriser As New Colorizer()
        Dim cfg As New Config()
        cfg.EnableStream(Stream.Depth, 848, 480, Format.Z16, 30)
        cfg.EnableStream(Stream.Color, 848, 480, Format.Rgb8, 6)
        Dim pipelineprofile As PipelineProfile = pipeline.Start(cfg)
        Dim profiles As StreamProfileList = pipelineprofile.Streams
        Dim colourstream As Stream = pipelineprofile.Streams.Where(Function(x) x.Format = Format.Rgb8).Single.Stream
        Dim depthstream As Stream = pipelineprofile.Streams.Where(Function(x) x.Format = Format.Z16).Single.Stream
        Do
            Using frames As FrameSet = pipeline.WaitForFrames()
                ' Get aligned colour frame
                Dim align As New Align(colourstream)
                Dim aligned As FrameSet = align.Process(frames)
                align.Dispose()
                Dim filtered As VideoFrame = aligned.DepthFrame
                ' do something with aligned.ColorFrame
                aligned.Dispose()
                ' Decimation doesn't leak
                Dim decimationfilter As New DecimationFilter()
                filtered = decimationfilter.ApplyFilter(filtered)
                decimationfilter.Dispose()

                ' Uncomment any of the filters below and run for a few minutes to see memory leaking

                ' transform the scene into disparity domain 
                ' (uncomment disparitytodepth as well)
                'Dim depthtodisparity As New DisparityTransform(True)
                'filtered = depthtodisparity.ApplyFilter(filtered)
                'depthtodisparity.Dispose()
                ' Spatial
                'Dim spatialfilter As New SpatialFilter
                'filtered = spatialfilter.ApplyFilter(filtered)
                'spatialfilter.Dispose()
                ' Temporal
                'Dim temporalfilter As New TemporalFilter
                'filtered = temporalfilter.ApplyFilter(filtered)
                'temporalfilter.Dispose()
                '' Hole filling, best done in the disparity domain, says https://github.com/IntelRealSense/librealsense/issues/1591
                'Dim holefillingfilter As New HoleFillingFilter
                'filtered = holefillingfilter.ApplyFilter(filtered)
                'holefillingfilter.Dispose()
                ' Revert back to depth domain
                'Dim disparitytodepth As New DisparityTransform(False)
                'filtered = disparitytodepth.ApplyFilter(filtered)
                'disparitytodepth.Dispose()
                'Dim coloriseddepth As VideoFrame = coloriser.Colorize(filtered)
                ' do something with coloriseddepth
                'coloriseddepth.Dispose()

                Dim data(848 * 480) As UShort
                filtered.CopyTo(data)
                ' do something with data
                filtered.Dispose()
            End Using
        Loop
    End Sub
End Module

`
Sorry about the tacky code formatting, the code tags seem not to like VB.

dorodnic, 2018-07-02:
Hi @smirkingman 
It seems somewhat extremely inefficient to create and dispose the filter object. Is this just to demonstrate the leak? In addition, every time you assign to `filtered` it is strongly recommended to `dispose` the previous value. Are you sure this is a leak and not just native objects piling up in the finalizer queue, waiting to be GC-ed?

RealSense-Customer-Engineering, 2018-07-12:
[Realsense Customer Engineering Team Comment]
@smirkingman
Does the suggestion from dorodnic work for you?

smirkingman, 2018-07-22:
Sorry for the delay in replying.
Thank you Dorodnic for investigating, you were on the right track.
Of course, allocating and disposing the filters is clumsy, it was indeed for demonstration.
However, continuously allocating and disposing filters or videoframes in the loop causes a leak.
This leaks, albeit slowly, if you uncomment the Dispose just before the GC:

`
Imports Intel.RealSense
Module Module1
    Sub Main()
        Dim pipeline As New Pipeline()
        Dim coloriser As New Colorizer()
        Dim cfg As New Config()
        cfg.EnableStream(Stream.Depth, 848, 480, Format.Z16, 30)
        cfg.EnableStream(Stream.Color, 848, 480, Format.Rgb8, 6)
        Dim pipelineprofile As PipelineProfile = pipeline.Start(cfg)
        Dim profiles As StreamProfileList = pipelineprofile.Streams
        Dim colourstream As Stream = pipelineprofile.Streams.Where(Function(x) x.Format = Format.Rgb8).Single.Stream
        Dim depthstream As Stream = pipelineprofile.Streams.Where(Function(x) x.Format = Format.Z16).Single.Stream
        Dim align As New Align(colourstream)
        Dim aligned As FrameSet
        Dim decimationfilter As New DecimationFilter()
        Dim depthtodisparity As New DisparityTransform(True)
        Dim spatialfilter As New SpatialFilter
        Dim temporalfilter As New TemporalFilter
        Dim holefillingfilter As New HoleFillingFilter
        Dim disparitytodepth As New DisparityTransform(False)
        Dim filtered As VideoFrame
        Do
            Using frames As FrameSet = pipeline.WaitForFrames()
                aligned = align.Process(frames)
                filtered = aligned.DepthFrame
                filtered = decimationfilter.ApplyFilter(filtered)
                filtered = depthtodisparity.ApplyFilter(filtered)
                filtered = spatialfilter.ApplyFilter(filtered)
                filtered = temporalfilter.ApplyFilter(filtered)
                filtered = holefillingfilter.ApplyFilter(filtered)
                filtered = disparitytodepth.ApplyFilter(filtered)
                Dim data(848 * 480 - 1) As UShort
                filtered.CopyTo(data)
                'filtered.Dispose() ' *** Uncomment this and we leak memory ***
            End Using
            GC.Collect(3)
        Loop
    End Sub
End Module

`

Conclusion: There seems to be an issue with memory allocation when creating and destroying filters/video frames. The issue can be circumvented by pre-allocating everything up front.

closed 2018-07-22 by smirkingman