realsenseai/librealsense #1985: Kernel oops when plugging in D435

mjog, 2018-07-02:
I just tried building librealsense from source following the [prereq and build instructions](https://github.com/IntelRealSense/librealsense/blob/development/doc/installation.md) and am getting the following kernel oops on plugging in a D435.

The system is an up-to-date Ubuntu 18.04 install: `Linux payens 4.15.0-23-generic #25-Ubuntu SMP Wed May 23 18:02:16 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux`.

The module rebuild seemed to work fine accourding to dmesg:

```
[95071.490275] usbcore: deregistering interface driver uvcvideo
[95071.682686] media: Linux media interface: v0.10
[95071.686272] Linux video capture interface: v2.00
[95071.708179] usbcore: registered new interface driver uvcvideo
[95071.708180] USB Video Class driver (1.1.1)
[95071.728645] usbcore: deregistering interface driver uvcvideo
[95071.924426] media: Linux media interface: v0.10
[95071.929488] Linux video capture interface: v2.00
[95071.935082] usbcore: registered new interface driver uvcvideo
[95071.935083] USB Video Class driver (1.1.1)
```

But on plugging in the camera, I get the following:

```
[95191.546748] usb 2-2: new SuperSpeed USB device number 7 using xhci_hcd
[95191.571398] usb 2-2: New USB device found, idVendor=8086, idProduct=0b07
[95191.571404] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[95191.571407] usb 2-2: Product: Intel(R) RealSense(TM) 430
[95191.571409] usb 2-2: Manufacturer: Intel(R) RealSense(TM) 430
[95191.571410] usb 2-2: SerialNumber: 816513021255
[95191.572557] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) 430 (8086:0b07)
[95191.574250] uvcvideo: Unable to create debugfs 2-7 directory.
[95191.574340] uvcvideo 2-2:1.0: Entity type for entity Intel(R) RealSense(TM) 430 with was not initialized!
[95191.574342] uvcvideo 2-2:1.0: Entity type for entity Processing 2 was not initialized!
[95191.574344] uvcvideo 2-2:1.0: Entity type for entity Intel(R) RealSense(TM) 430 with was not initialized!
[95191.574345] uvcvideo 2-2:1.0: Entity type for entity Camera 1 was not initialized!
[95191.574364] ------------[ cut here ]------------
[95191.574365] sysfs: cannot create duplicate filename '/dev/char/240:0'
[95191.574375] WARNING: CPU: 3 PID: 20050 at /build/linux-uT8zSN/linux-4.15.0/fs/sysfs/dir.c:31 sysfs_warn_dup+0x56/0x70
[95191.574376] Modules linked in: hid_sensor_gyro_3d(OE) hid_sensor_accel_3d(OE) uvcvideo(OE) videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_core videodev(OE) media btrfs zstd_compress xor raid6_pq ufs qnx4 hfsplus hfs minix ntfs msdos jfs xfs libcrc32c hid_sensor_trigger industrialio_triggered_buffer hid_sensor_iio_common hid_sensor_hub brcmfmac brcmutil cfg80211 cmac cdc_ether usbnet r8152 mii rfcomm bnep binfmt_misc nls_iso8859_1 snd_hda_codec_cirrus snd_hda_codec_generic snd_hda_codec_hdmi snd_hda_intel joydev snd_hda_codec snd_hda_core snd_hwdep snd_pcm applesmc input_polldev snd_seq_midi snd_seq_midi_event snd_rawmidi intel_rapl x86_pkg_temp_thermal coretemp kvm_intel kvm snd_seq irqbypass intel_cstate intel_rapl_perf btusb snd_seq_device btrtl snd_timer btbcm btintel input_leds bluetooth
[95191.574417]  intel_pch_thermal lpc_ich ecdh_generic bcm5974 mei_me bdc_pci snd mei shpchp spi_pxa2xx_pci dw_dmac_pci soundcore dw_dmac_core apple_bl sbs acpi_als kfifo_buf sbshc spi_pxa2xx_platform industrialio mac_hid sch_fq_codel parport_pc ppdev lp parport ip_tables x_tables autofs4 algif_skcipher af_alg dm_crypt hid_apple hid_generic usbhid hid crct10dif_pclmul crc32_pclmul i915 ghash_clmulni_intel pcbc aesni_intel aes_x86_64 crypto_simd glue_helper i2c_algo_bit cryptd drm_kms_helper ahci syscopyarea sysfillrect libahci sysimgblt fb_sys_fops drm thunderbolt video [last unloaded: hid_sensor_gyro_3d]
[95191.574444] CPU: 3 PID: 20050 Comm: kworker/3:2 Tainted: G        W  OE    4.15.0-23-generic #25-Ubuntu
[95191.574445] Hardware name: Apple Inc. MacBookPro12,1/Mac-E43C1C25D4880AD6, BIOS MBP121.88Z.0167.B18.1610241407 10/24/2016
[95191.574447] Workqueue: usb_hub_wq hub_event
[95191.574450] RIP: 0010:sysfs_warn_dup+0x56/0x70
[95191.574451] RSP: 0018:ffffc17501ee3680 EFLAGS: 00010282
[95191.574452] RAX: 0000000000000000 RBX: ffff9f4bd3961000 RCX: 0000000000000006
[95191.574453] RDX: 0000000000000007 RSI: 0000000000000086 RDI: ffff9f4f2ed96490
[95191.574453] RBP: ffffc17501ee3698 R08: 0000000000000001 R09: 00000000000012b7
[95191.574454] R10: 0000000000000001 R11: 0000000000000000 R12: ffffc17501ee36f9
[95191.574455] R13: ffff9f4f1c17a5d8 R14: 0000000000000001 R15: ffffffffffffffef
[95191.574456] FS:  0000000000000000(0000) GS:ffff9f4f2ed80000(0000) knlGS:0000000000000000
[95191.574457] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[95191.574457] CR2: 00007fb860dc5004 CR3: 000000022420a003 CR4: 00000000003606e0
[95191.574458] Call Trace:
[95191.574462]  sysfs_do_create_link_sd.isra.2+0x9f/0xb0
[95191.574464]  sysfs_create_link+0x25/0x40
[95191.574466]  device_add+0x5b2/0x680
[95191.574468]  cdev_device_add+0x4e/0x80
[95191.574470]  ? cdev_init+0x55/0x60
[95191.574473]  media_devnode_register+0xe7/0x180 [media]
[95191.574475]  __media_device_register+0x77/0x110 [media]
[95191.574477]  uvc_probe+0x226e/0x2a20 [uvcvideo]
[95191.574480]  usb_probe_interface+0x129/0x300
[95191.574481]  ? uvc_alloc_entity+0x130/0x130 [uvcvideo]
[95191.574482]  ? usb_probe_interface+0x129/0x300
[95191.574484]  driver_probe_device+0x31e/0x490
[95191.574486]  __device_attach_driver+0x86/0x100
[95191.574487]  ? __driver_attach+0xf0/0xf0
[95191.574488]  bus_for_each_drv+0x6b/0xb0
[95191.574490]  __device_attach+0xdd/0x160
[95191.574491]  device_initial_probe+0x13/0x20
[... 83 lines cut ...]
[95191.578160]  usb_new_device+0x269/0x490
[95191.578161]  hub_port_connect+0x70f/0xa90
[95191.578163]  hub_event+0x84e/0xb10
[95191.578165]  process_one_work+0x1de/0x410
[95191.578166]  worker_thread+0x32/0x410
[95191.578168]  kthread+0x121/0x140
[95191.578169]  ? process_one_work+0x410/0x410
[95191.578171]  ? kthread_create_worker_on_cpu+0x70/0x70
[95191.578173]  ? do_syscall_64+0x115/0x130
[95191.578175]  ret_from_fork+0x35/0x40
[95191.578176] Code: 85 c0 48 89 c3 74 12 b9 00 10 00 00 48 89 c2 31 f6 4c 89 ef e8 ac c6 ff ff 4c 89 e2 48 89 de 48 c7 c7 70 54 4f a7 e8 9a 8e d8 ff <0f> 0b 48 89 df e8 60 12 f4 ff 5b 41 5c 41 5d 5d c3 66 0f 1f 84 
[95191.578206] ---[ end trace 5aec56c19219beec ]---
[95191.578223] media: media_devnode_register: cdev_device_add failed
```

lsusb shows the device as being present:

```
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
    |__ Port 2: Dev 7, If 0, Class=Video, Driver=, 5000M
    |__ Port 2: Dev 7, If 1, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 2: Dev 7, If 2, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 2: Dev 7, If 3, Class=Video, Driver=, 5000M
    |__ Port 2: Dev 7, If 4, Class=Video, Driver=uvcvideo, 5000M
```

mjog, 2018-07-02:
Hmm, after a reboot it seems to be semi-working - the realsense viewer picks up the camera but is having a lot of trouble streaming data from it.

mjog, 2018-07-02:
And now, after running `./scripts/patch-realsense-ubuntu-lts.sh` once more, I get the same oops as above again.

ev-mp, 2018-07-02:
Hello @mjog ,
I'm quite confused as I read MacBook12 is kernel logs - what is the HW setup (CPU) you're using ?

mjog, 2018-07-02:
It's a MacBookPro 13" running Ubuntu Bionic.

pedrombmachado, 2018-07-02:
Same happens with the new kernel version 4.15.0-24 on Ubuntu 16.04.4 LTS x86_64
```
$ cat /var/crash/librealsense2-dkms.0.crash
ProblemType: Package
DKMSBuildLog:
 DKMS make.log for librealsense2-dkms-1.2.0 for kernel 4.15.0-24-generic (x86_64)
 Mon  2 Jul 15:11:28 BST 2018
 make: Entering directory '/usr/src/linux-headers-4.15.0-24-generic'
 /bin/sh: 1: cannot create /var/lib/dkms/librealsense2-dkms/1.2.0/build/4.15.0/drivers/media/usb/uvc/.cache.mk: Directory nonexistent
 /bin/sh: 1: cannot create /var/lib/dkms/librealsense2-dkms/1.2.0/build/4.15.0/drivers/media/usb/uvc/.cache.mk: Directory nonexistent
 /bin/sh: 1: cannot create /var/lib/dkms/librealsense2-dkms/1.2.0/build/4.15.0/drivers/media/usb/uvc/.cache.mk: Directory nonexistent
 /bin/sh: 1: cannot create /var/lib/dkms/librealsense2-dkms/1.2.0/build/4.15.0/drivers/media/usb/uvc/.cache.mk: Directory nonexistent
 /bin/sh: 1: cannot create /var/lib/dkms/librealsense2-dkms/1.2.0/build/4.15.0/drivers/media/usb/uvc/.cache.mk: Directory nonexistent
 /bin/sh: 1: cannot create /var/lib/dkms/librealsense2-dkms/1.2.0/build/4.15.0/drivers/media/usb/uvc/.cache.mk: Directory nonexistent
 /bin/sh: 1: cannot create /var/lib/dkms/librealsense2-dkms/1.2.0/build/4.15.0/drivers/media/usb/uvc/.cache.mk: Directory nonexistent
 /bin/sh: 1: cannot create /var/lib/dkms/librealsense2-dkms/1.2.0/build/4.15.0/drivers/media/usb/uvc/.cache.mk: Directory nonexistent
 /bin/sh: 1: cannot create /var/lib/dkms/librealsense2-dkms/1.2.0/build/4.15.0/drivers/media/usb/uvc/.cache.mk: Directory nonexistent
 /bin/sh: 1: cannot create /var/lib/dkms/librealsense2-dkms/1.2.0/build/4.15.0/drivers/media/usb/uvc/.cache.mk: Directory nonexistent
 /bin/sh: 1: cannot create /var/lib/dkms/librealsense2-dkms/1.2.0/build/4.15.0/drivers/media/usb/uvc/.cache.mk: Directory nonexistent
 /bin/sh: 1: cannot create /var/lib/dkms/librealsense2-dkms/1.2.0/build/4.15.0/drivers/media/usb/uvc/.cache.mk: Directory nonexistent
 /bin/sh: 1: cannot create /var/lib/dkms/librealsense2-dkms/1.2.0/build/4.15.0/drivers/media/usb/uvc/.cache.mk: Directory nonexistent
 /bin/sh: 1: cannot create /var/lib/dkms/librealsense2-dkms/1.2.0/build/4.15.0/drivers/media/usb/uvc/.cache.mk: Directory nonexistent
 /bin/sh: 1: cannot create /var/lib/dkms/librealsense2-dkms/1.2.0/build/4.15.0/drivers/media/usb/uvc/.cache.mk: Directory nonexistent
 /bin/sh: 1: cannot create /var/lib/dkms/librealsense2-dkms/1.2.0/build/4.15.0/drivers/media/usb/uvc/.cache.mk: Directory nonexistent
 /bin/sh: 1: cannot create /var/lib/dkms/librealsense2-dkms/1.2.0/build/4.15.0/drivers/media/usb/uvc/.cache.mk: Directory nonexistent
 /bin/sh: 1: cannot create /var/lib/dkms/librealsense2-dkms/1.2.0/build/4.15.0/drivers/media/usb/uvc/.cache.mk: Directory nonexistent
 /bin/sh: 1: cannot create /var/lib/dkms/librealsense2-dkms/1.2.0/build/4.15.0/drivers/media/usb/uvc/.cache.mk: Directory nonexistent
 /bin/sh: 1: cannot create /var/lib/dkms/librealsense2-dkms/1.2.0/build/4.15.0/drivers/media/usb/uvc/.cache.mk: Directory nonexistent
 arch/x86/Makefile:156: CONFIG_X86_X32 enabled but no binutils support
 /bin/sh: 1: cannot create /var/lib/dkms/librealsense2-dkms/1.2.0/build/4.15.0/drivers/media/usb/uvc/.cache.mk: Directory nonexistent
 /bin/sh: 1: cannot create /var/lib/dkms/librealsense2-dkms/1.2.0/build/4.15.0/drivers/media/usb/uvc/.cache.mk: Directory nonexistent
 /bin/sh: 1: cannot create /var/lib/dkms/librealsense2-dkms/1.2.0/build/4.15.0/drivers/media/usb/uvc/.cache.mk: Directory nonexistent
 /bin/sh: 1: cannot create /var/lib/dkms/librealsense2-dkms/1.2.0/build/4.15.0/drivers/media/usb/uvc/.cache.mk: Directory nonexistent
 /bin/sh: 1: cannot create /var/lib/dkms/librealsense2-dkms/1.2.0/build/4.15.0/drivers/media/usb/uvc/.cache.mk: Directory nonexistent
 /bin/sh: 1: cannot create /var/lib/dkms/librealsense2-dkms/1.2.0/build/4.15.0/drivers/media/usb/uvc/.cache.mk: Directory nonexistent
 /bin/sh: 1: cannot create /var/lib/dkms/librealsense2-dkms/1.2.0/build/4.15.0/drivers/media/usb/uvc/.cache.mk: Directory nonexistent
 /bin/sh: 1: cannot create /var/lib/dkms/librealsense2-dkms/1.2.0/build/4.15.0/drivers/media/usb/uvc/.cache.mk: Directory nonexistent
 /bin/sh: 1: cannot create /var/lib/dkms/librealsense2-dkms/1.2.0/build/4.15.0/drivers/media/usb/uvc/.cache.mk: Directory nonexistent
 /bin/sh: 1: cannot create /var/lib/dkms/librealsense2-dkms/1.2.0/build/4.15.0/drivers/media/usb/uvc/.cache.mk: Directory nonexistent
 /bin/sh: 1: cannot create /var/lib/dkms/librealsense2-dkms/1.2.0/build/4.15.0/drivers/media/usb/uvc/.cache.mk: Directory nonexistent
 /bin/sh: 1: cannot create /var/lib/dkms/librealsense2-dkms/1.2.0/build/4.15.0/drivers/media/usb/uvc/.cache.mk: Directory nonexistent
 /bin/sh: 1: cannot create /var/lib/dkms/librealsense2-dkms/1.2.0/build/4.15.0/drivers/media/usb/uvc/.cache.mk: Directory nonexistent
 /bin/sh: 1: cannot create /var/lib/dkms/librealsense2-dkms/1.2.0/build/4.15.0/drivers/media/usb/uvc/.cache.mk: Directory nonexistent
 /bin/sh: 1: cannot create /var/lib/dkms/librealsense2-dkms/1.2.0/build/4.15.0/drivers/media/usb/uvc/.cache.mk: Directory nonexistent
 /bin/sh: 1: cannot create /var/lib/dkms/librealsense2-dkms/1.2.0/build/4.15.0/drivers/media/usb/uvc/.cache.mk: Directory nonexistent
 /bin/sh: 1: cannot create /var/lib/dkms/librealsense2-dkms/1.2.0/build/4.15.0/drivers/media/usb/uvc/.cache.mk: Directory nonexistent
[... cut ...]
 make[1]: *** No rule to make target '/var/lib/dkms/librealsense2-dkms/1.2.0/build/4.15.0/drivers/media/usb/uvc/Makefile'. Stop.
 Makefile:1728: recipe for target 'uvcvideo.ko' failed
 make: *** [uvcvideo.ko] Error 2
 make: Leaving directory '/usr/src/linux-headers-4.15.0-24-generic'
DKMSKernelVersion: 4.15.0-24-generic
Date: Mon Jul  2 15:11:32 2018
Package: librealsense2-dkms 1.2.0-0ubuntu5
PackageVersion: 1.2.0-0ubuntu5
SourcePackage: librealsense2
Title: librealsense2-dkms 1.2.0-0ubuntu5: librealsense2-dkms kernel module failed to build
```

pedrombmachado, 2018-07-02:
I forgot to mention that the camera was working with the previous kernels
Found linux image: /boot/vmlinuz-4.13.0-45-generic
Found initrd image: /boot/initrd.img-4.13.0-45-generic
Found linux image: /boot/vmlinuz-4.4.0-130-generic
Found initrd image: /boot/initrd.img-4.4.0-130-generic
The new version has been installed via sudo apt dist-upgrade

ev-mp, 2018-07-02:
Hello @pedrombmachado , 
Librealsense DKMS support for 4.15 is [not yet available](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#linux-distribution). Currently it supports 4.4, 4.10 and 4.13 LTS kernels only.
It is not as the original issue with deploying Ubuntu on MacBook and building patches manually.

pedrombmachado, 2018-07-02:
Hello @ev-mp, 
  Will the be supported 4.15 as it is LTS release? Or should I roll-back to version 4.10?

ev-mp, 2018-07-02:
DKMS support for 4.15 LTS is in work and will be available within two-week frame.
In the meanwhile. you have the option to switch back to 4.10/4.13 to get DKMS-based installation.
Alternatively, you can follow the instructions from the link in the previous post and install the patch for 4.15 manually .

pedrombmachado, 2018-07-02:
Hi @ev-mp,
  I am trying to apply the patch but I am stuck in step3
Install Intel Realsense permission scripts located in librealsense source directory:
```
sudo cp config/99-realsense-libusb.rules /etc/udev/rules.d/
sudo udevadm control --reload-rules && udevadm trigger
```
My question is where can I find the `99-realsense-libusb.rules` from ?
I have done
```
sudo updatedb
sudo locate 99-realsense-libusb.rules
```
and there was no result.

ev-mp, 2018-07-03:
The scripts shall be executed with `pwd` being the librealsense's root, as the manual [instructs](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#prerequisites)
>Navigate to librealsense root directory to run the following scripts...

The specific file can be found in  librealsense source tree - 
https://github.com/IntelRealSense/librealsense/blob/master/config/99-realsense-libusb.rules.

You should have obtained it as part of downloading/cloning the source repository.

pedrombmachado, 2018-07-04:
Hi @ev-mp ,
  Problem solved. Many thanks. I added some changes to the instructions to make them more user-friendly and created a pull request. 
My problem is now solved.

mjog, 2018-07-05:
So... getting back to the original problem, is the issue that I'm using the 4.15 kernel?

RealSense-Customer-Engineering, 2018-07-19:
[Realsense Customer Engineering Team Comment]
hi @mjog,

i don't think it's due to 4.15 kernel, would like to know which version of source you are using? 4.15 kernel is supported since libRealsense v2.11.1,

RealSense-Customer-Engineering, 2018-07-23:
[Realsense Customer Engineering Team Comment]
hi @mjog,

do you still having this issue?
your case seems kernel patch failed, do you see any error when applying the patch?

closed 2018-08-09 by RealSense-Customer-Engineering