realsenseai/librealsense #1986: Unable to upgrade D435 firmware from Linux

mjog, 2018-07-02:
After following the instructions here: https://www.intel.com/content/www/us/en/support/articles/000028171/emerging-technologies/intel-realsense-technology.html

I am unable to update the firmware on an up-to-date Ubuntu 18.04 install:

```
mjg@payens:~$ lsusb | grep Intel
Bus 001 Device 004: ID 8086:0ad6 Intel Corp. 
mjg@payens:~$ intel-realsense-dfu -i ../Signed_Image_UVC_5_9_13_0.bin -b 001 -d 004 -f
D400 busnum = 1
D400 devnum = 4
D400 FW file path = ../Signed_Image_UVC_5_9_13_0.bin
Could not find device in opmode: Device found, but incorrect (vid, pid) combo. Found (8086, ad6)
device pid = 2774
Valid pid not found
```

ev-mp, 2018-07-02:
Hello @mjog ,
The device is enumerated in USB2 mode and not USB3 as required.
Please check the cable/connectors, then rerun "realsense-viewer" to verify that the device is recognized properly before using FW upgrade
[image]

weiwuhuhu, 2018-10-12:
> That resolved it, thanks. I keep on forgetting only one of my ports is USB3 and it didn't occur to me to check.

hello,
      I have the same problem with you,and i used usb3 ,when i run the "realsense-viewer", it' ok. But i can't update the firmware either.

radfordi, 2019-04-24:
I had this issue, but wasn't related to USB2.0 vs USB3.0.  Instead it was because I had both Debian packages, `rs-tool-intel` and `intel-realsense-dfu`, installed at the same time.  These two packages are currently both available from same repository and they both provide an executable with the same name, but with different paths.
```
    /bin/intel-realsense-dfu
    /usr/bin/intel-realsense-dfu
```
With the default Ubuntu path, the later takes precedence meaning I think I had the latest package installed, but when programming I got an error because I was really running the obsolete tool.

>    Could not find device in opmode: Device found, but incorrect (vid, pid) combo. Found (8086, b3a)

Having both of these packages available from the same repository is a source of confusion.