realsenseai/librealsense #1989: Is it safe to downgrade the FW on a D415?

xtrawurst, 2018-07-02:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 5.9.13 | 
| Operating System & Version |  Win10 | 
| SDK Version                            | 2.13.0 | 

### Issue Description
Since the new 5.9.13 is causing some problems (see #1988), can I use the RS DFU tool to downgrade my FW to 5.9.11 or would that cause any problems?

RealSense-Customer-Engineering, 2018-07-03:
[Realsense Customer Engineering Team Comment]
hi @xtrawurst,

please refer to FW  Specification Update document:
https://realsense.intel.com/intel-realsense-downloads/#firmware
to check if issue fixed in FW v5.9.13 have no impact yo you.

dorodnic, 2018-07-03:
@xtrawurst it is safe to downgrade if you choose to. Please share the full error you got with the new firmware, and as much information about your system as you can, and we will try to reproduce this problem

xtrawurst, 2018-07-03:
@dorodnic Please find the description of the error here:
https://github.com/IntelRealSense/librealsense/issues/1988

dorodnic, 2018-07-11:
Let's follow-up at #1988

closed 2018-07-11 by dorodnic