realsenseai/librealsense #1990: JSON: aux-param-autoexposure-setpoint vs. param-autoexposure-setpoint

xtrawurst, 2018-07-02:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Operating System & Version |  Win10 | 

### Issue Description
What's the difference? When I export the "Default" preset I see them both in the resulting JSON.

dorodnic, 2018-07-03:
Hi @xtrawurst 
No difference. JSON structure is based on **several** internal Intel tools developer prior to this SDK. One of these tools expects `aux-param-colorcorrection` to be the name of the JSON field while the other calls the same thing `param-colorcorrection`. When we export, we write both, to make sure we are compatible with everyone. When you import, we will only take one of the fields.

xtrawurst, 2018-07-03:
Ok, so when I write a JSON settings file myself I should always write both
too?

Rafael Spring
CTO, Co-Founder
DotProduct LLC / DotProduct GmbH
rafael@dotproduct3d.com
www.dotproduct3d.com
m: +49 1525 3822738
@DotProduct3D

dorodnic, 2018-07-03:
I believe writing just one would be sufficient, unless you want to also be compatible with internal / NDA tools (such as IPDev)

dorodnic, 2018-07-11:
Let us know if you have more questions on this subject

closed 2018-07-11 by dorodnic