realsenseai/librealsense #1991: D435 - how to use complete sensor?

LeMajors, 2018-07-02:
When using the D435 with the RealSense Viewer I switched off "Depth" and switched on the "Infrared 1". With 1280 x 800 (16:10) I got the biggest angle of view for x and y. With "Depth" on I can't find a working resolution that doesn't clip the image in one direction.

At this link: https://software.intel.com/en-us/realsense/d400 the FOV is 85.2° x 58° (+/- 3°) which is also more 16:10.

I there a reasen for not getting the whole resolution for depth or did I do something wrong?

RealSense-Customer-Engineering, 2018-07-05:
[Realsense Customer Engineering Team Comment]
hi @LeMajors,

do you mean why IR camera resolution is 1280x800 but depth is 1280x720 as maximum?
if so it's due to stereo camera technology we are using, depth can only be generated on the overlap area of two camera, so it can't reach same resolution of cameras.

LeMajors, 2018-07-05:
Ok, thanks for this info!
Marcus

closed 2018-07-05 by LeMajors