realsenseai/librealsense #1994: Cannot import pyrealsense2 even realsense2.dll and pyrealsense2.pyd is next to my script

kidserge, 2018-07-03:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (05.09.11.00) | 
| Operating System & Version |  Win 10  | 
| Platform                                 | PC Pycharm 2018.1.4 on Conda 4.5.4 |
| SDK Version                            |  2  | 
| Language                            |  python 3.5  | 

This is my first time post any issues, if I do some mistake please feel free to point it out.

I try to use realsense D435 image and analysis using tensorflow to detect hand movement.
The problem is I cannot import pyrealsense2 to my program.

I build the realsense2.dll and pyrealsense2.pyd according to https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/readme.md via Python 3.5.5 and also make conda environment using python 3.5.

However, I try to import pyrealsense2 in Pycharm which do not appear on the package list.
As a result, I place realsense2.dll and pyrealsense2.pyd next to my script as instructed and try import pyrealsense2 result in ImportError: DLL load failed: %1 is not a valid Win32 application.

Anyone have any idea what should I try next?

Best

RealSense-Customer-Engineering, 2018-07-05:
[Realsense Customer Engineering Team Comment]
@kidserge
(1) please check if the python (32 bit or 64 bit) you using  match the library type you build  x86 (32 bit) or x64

(2) please copy you dll and pyd file you generate  to the realsense sample code ...\realsense\wrapper\python\example\
and run the sample code python-tutorial-t-depth.py

RealSense-Customer-Engineering, 2018-07-23:
[Realsense Customer Engineering Team Comment]
Hi @kidserge
is this still an issue..
you can also refer to issue #2021 which has similar issue as yours

RealSense-Customer-Engineering, 2018-08-10:
[Realsense Customer Engineering Team Comment]
Hi @kidserge
is this still an issue for you?
the GitHub has newer version of pre-compiled releasense.dll and pytrealsense2.pyd
that does not need 2015/2017  VC++ redistributable installed

closed 2018-08-21 by dorodnic