realsenseai/librealsense #1996: Realsense viewer keeps requiring firmware upgrade

danielw256, 2018-07-03:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.09.13.00 | 
| Operating System & Version |   Win 10 1803 build 17134.112  |             
| Platform                                 | PC |
| SDK Version                           |  2.13.0                         | 
| Language                               |  C                         | 
 
I am developing a C++ program on visual studio 2017 with a D415. On a Dell laptop, I recently upgraded the librealsense driver from 2.9.0 to 2.13.0. The RS viewer reported the firmware was out of date. So, I upgraded the firmware. The error in RS viewer went away. A few days later, the RS viewer reported the firmware was out of date. I repeated the procedure and the error went away.

I put the D415on a PC with a Gigabyte motherboard. The RS viewer reported the firmware was out of date. I repeated the procedure and the error went away.

Am I doing something wrong that resets the firmware and requires upgrade to the current version? I have had to do so 2 extra times. Does the firmware reset when I put the camera on a new computer?

RealSense-Customer-Engineering, 2018-07-06:
[Realsense Customer Engineering Team Comment]
hi @danielw256,

are you using same unit?
seems your FW is v5.8.15, want to make sure it's update failed or somehow it rollback to old version.
do you installed UWP driver for D400?
can you please how driver info of D400 on Device manager?
thanks.

danielw256, 2018-07-06:
All info in this issue is about the same D415 unit.
The update succeeded. RS error went away. Then, somehow rollbacked to old version (3 times). What are the ways this can occur? I did nothing specific to roll it back.
I let Windows automatically download drivers. How would I know if UWP driver is installed?

[image]

[image]

RealSense-Customer-Engineering, 2018-07-09:
[Realsense Customer Engineering Team Comment]
hi @danielw256,

yes, from the driver properties, it's UWP driver installed.
so please go "search windows" search "services", open the services and search for "RS400Service", right click on it to open it's properties page.
change the start up type to disabled, and change the service status to stop.
reboot the system, then the FW version should not change again.

closed 2018-07-30 by RealSense-Customer-Engineering