realsenseai/librealsense #1998: Y16 "Save snapshot" Crashes RealSense Viewer

apoorva2398, 2018-07-03:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 
| Firmware Version                   | 05.09.13.00 | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC  |
| SDK Version                            |  2.12.0                          | 
| Language                            |  C++                        | 
| Segment			|  Computer vision in healthcare                         | 

### Issue Description
"Save snapshot" in the RealSense Viewer crashes when saving Y16. It outputs files that are incomplete.

timprepscius, 2018-07-03:
I am also interested in this.  I think I recall in the specs that onboard the d435, the infrared is 10bit.

Besides asking the d435 SDK for Y16 (which also failed for me) I also asked for RS2_FORMAT_RAW10 and that failed as well.

I would be interested in getting the stereo camera output in the highest resolution and contrast possible.

RealSense-Customer-Engineering, 2018-07-03:
[Realsense Customer Engineering Team Comment]
Hello,

The Y16 format is reserved for custom calibration. Y16 provides unrectified images. For applications, only use the Y8 format. The supported depth formats are listed in the datasheet, page 52, table 4-2. The Y8 format provides rectified images.

Do not use Y16 in your applications. Use only the Y8 format in your applications.

We will take a look at why the RealSense Viewer crashes.

Regards,
Jesus G.
Intel Customer Support

timprepscius, 2018-07-03:
A little bit more information btw, may relate to the viewer:

the rs-save-to-disk.cpp line:

```
            // Write images to disk
            stbi_write_png(png_file.str().c_str(), vf.get_width(), vf.get_height(),
                           vf.get_bytes_per_pixel(), vf.get_data(), vf.get_stride_in_bytes());

```
doesn't seem to work properly with Y16
or every other program I have doesn't read the output correctly - could be them though - looks like an endian issue perhaps.

If I save the frames data to a raw file, I can read it back and view it.  So the data is correct before transformation to png.

dorodnic, 2018-07-12:
@apoorva2398 @timprepscius 
We are using [stb](https://github.com/nothings/stb) inside the Viewer, and not something like OpenCV, to simplify life for users who are building the SDK from source. It should be easy enough to modify the code to export raw 16-bit data, or integrate other 3rd party. 
If you find lightweight enough solution we would be happy to integrate it via a pull-request.

apoorva2398, 2018-07-24:
As @timprepscius mentioned, I think it could be an issue with the endianness of depth raw data. Reading it as little endian gives the correct depth values for each pixel but doesn't display correctly, and vice versa for big endian. 

Little endian gives the correct depth value (499 mm) but displays the image incorrectly (with colormap jet):
[image]

Big endian displays the image correctly but gives the wrong depth value (6221 mm >> 499 mm!):
[image]

RealSense Viewer snapshot:
[image]

RealSense-Customer-Engineering, 2018-08-08:
[Realsense Customer Engineering Team Comment]
hi @apoorva2398,

pull request 2201 for this, please have a try.

dorodnic, 2018-08-21:
I believe this issue has been **partially** resolved. 
Trying to save Y16 no longer **crashes** the Viewer but we still do not support saving 16-bit greyscale data. Other libraries like OpenCV can come to rescue if this must be done, however, since Y16 format for RealSense devices is only intended to be used in calibration flows, I'm Ok with not supporting this out of the box. 
Feel free to open a separate issue if you must have this capability in the SDK and we will handle it as an enhancement request.

closed 2018-08-21 by dorodnic