realsenseai/librealsense #1999: spectrum of 435 stereo imagers

timprepscius, 2018-07-03:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 5.9.13 | 
| Operating System & Version |  MacOS  | 
| Platform                                 | PC  |
| SDK Version                            |  2.13                           | 
| Language                            |  C++                           | 
| Segment			|  others                          | 

### Issue Description

It seems like the stereo imagers of the 435 is not just IR.
Is this correct?  

What is the spectrum of measured light?

Is there anyway to do some sort of filter so that it is just ~850nm IR, or so infrared1 & infrared2 send back just the ~850nm IR.

agrunnet, 2018-07-03:
It is “all pass”. All visible and IR wavelengths. 
For the D435 you cannot select which wavelength it sees. For the d415 it has Bayer Pattern filters so it does have some spectral selectivity. Check out the Remove IR setting that allows you to remove the IR pattern from the left imager even when the projector is turned on.

timprepscius, 2018-07-03:
Thank you for this information.

I'm going to try to apply a physical camera IR pass filter, see what effect that has.

agrunnet, 2018-07-04:
Above 940nm but the sensitivity does fall quite fast.

RealSense-Customer-Engineering, 2018-07-17:
[Realsense Customer Engineering Team Comment]
hi @timprepscius,

do you need more support on this topic?

closed 2018-07-31 by RealSense-Customer-Engineering