realsenseai/librealsense #2005: Can i run two REALSENSE D415 in multithreading？

linzihao1999, 2018-07-05:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415| 
| Firmware Version                   |05.08.15.00 | 
| Operating System & Version |   Ubuntu 16  | 
| Kernel Version (Linux Only)    | 4.4.55                                        | 
| Platform                                 | arm board |
| SDK Version                            |  2.11.1                         | 
| Language                            |  c++                        | 
| Segment			|                 |

dorodnic, 2018-07-05:
Hi @linzihao1999 
Yes.
Different instances of `pipeline` or `device` objects can co-exist on different threads, and you can send frames from thread to thread using `frame_queue` primitive.

dorodnic, 2018-07-12:
@linzihao1999 do you have any follow-up questions on this?

closed 2018-07-23 by RealSense-Customer-Engineering