realsenseai/librealsense #2007: Strange when no uplink/ethernet is connected get OpCodes do not match! Sent 16 but received -1!

vikaskr85, 2018-07-05:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | 05.09.11.00 | 
| Operating System & Version |   Raspbian GNU/Linux 9.3  | 
| Kernel Version (Linux Only)    |  (4.9.59-v7+, 4.14.39-v7+,4.14.52-v7+)                                         | 
| Platform                                 | Raspberry Pi  |
| SDK Version                            |  { 2.10.0 }                          | 
| Language                            |  {C }                          | 
| Segment			|  {others }                          | 

### Issue Description
Hi ,

We are using raspberrypi and testing.

When we connect ethernet/uplink and use rs-enumerate-devices all works but its acting strange when no uplink/ethernet is connected and call rs-enumerate-devices shows

Realsense error calling rs2_create_device(info_list:0x131bf98, index:0): 
OpCodes do not match! Sent 16 but received -1!

Please help.

RealSense-Customer-Engineering, 2018-07-08:
[Realsense Customer Engineering Team Comment]
Hi @vikaskr85,

It's weird that the issue is related to the Ethernet connection. How about other apps like rs-capture, realsense-viewer, etc?
Do you have another platform to try? Only see this on your raspberrypi platform?

RealSense-Customer-Engineering, 2018-07-10:
[Realsense Customer Engineering Team Comment]
hi @vikaskr85,

I see you file Zendesk ticket for this issue, so we will support this on Zendesk.

vikaskr85, 2018-07-10:
We have made observation and progress.

The issue is not present after we update sensor firmware to 05.09.13.

closed 2018-07-10 by vikaskr85

hlc111, 2018-08-02:
Hi vikaskr85,
As i know RealSense only support USB3.0, could you kindly share some information about how raspberry work with RealSense.

I'm also very Interested with raspberry&RealSense.

Thanks!

vikaskr85, 2018-08-02:
update sensor firmware to 05.09.13 and we followed the link
https://github.com/IntelRealSense/librealsense/blob/master/doc/RaspberryPi3.md

hlc111, 2018-08-03:
Hi vikaskr85,

Got it.

Thanks!