realsenseai/librealsense #2008: Unfortunately, realsense_app has stopped: Android App

fotsing365, 2018-07-05:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   |  | 
| Operating System & Version |   Android 6.0.1  | 
| Kernel Version (Linux Only)    |                             | 
| Platform                                 | Rooted Lenovo YT3-850F  |
| SDK Version                            |  2.12.0                         | 
| Language                            |       C++                  | 
| Segment			|                          | 

### Issue Description
<After following instructions from this [link>](https://github.com/IntelRealSense/librealsense/blob/master/doc/android/AndroidJavaApp.md) the android apps crashes by displaying this message **Unfortunately, realsense_app has stopped**. But I noticed that my D435 seems to be set on despite this crash because the IR led is on. 
 During the android studio  installation the process seems to be right but the **logcat consol** display this.07-05 06:02:20.970 11520-11520/? I/art: Late-enabling -Xcheck:jni
07-05 06:02:21.039 11520-11520/? I/Zygote: ZygoteInit::mMethod=public static void android.app.ActivityThread.main(java.lang.String[])
07-05 06:02:21.224 11520-11520/com.example.realsense_app W/art: Failed execv(/system/bin/dex2oat --runtime-arg -classpath --runtime-arg  --instruction-set=arm --instruction-set-features=smp,div,-atomic_ldrd_strd --runtime-arg -Xrelocate --boot-image=/system/framework/boot.art --runtime-arg -Xms64m --runtime-arg -Xmx512m --instruction-set-variant=cortex-a7 --instruction-set-features=default --dex-file=/data/app/com.example.realsense_app-2/split_lib_dependencies_apk.apk --oat-file=/data/dalvik-cache/arm/data@app@com.example.realsense_app-2@split_lib_dependencies_apk.apk@classes.dex) because non-0 exit status
07-05 06:02:21.570 11520-11520/com.example.realsense_app W/art: Failed execv(/system/bin/dex2oat --runtime-arg -classpath --runtime-arg  --instruction-set=arm --instruction-set-features=smp,div,-atomic_ldrd_strd --runtime-arg -Xrelocate --boot-image=/system/framework/boot.art --runtime-arg -Xms64m --runtime-arg -Xmx512m --instruction-set-variant=cortex-a7 --instruction-set-features=default --dex-file=/data/app/com.example.realsense_app-2/split_lib_slice_0_apk.apk --oat-file=/data/dalvik-cache/arm/data@app@com.example.realsense_app-2@split_lib_slice_0_apk.apk@classes.dex) because non-0 exit status
07-05 06:02:21.671 11520-11520/com.example.realsense_app W/art: Failed execv(/system/bin/dex2oat --runtime-arg -classpath --runtime-arg  --instruction-set=arm --instruction-set-features=smp,div,-atomic_ldrd_strd --runtime-arg -Xrelocate --boot-image=/system/framework/boot.art --runtime-arg -Xms64m --runtime-arg -Xmx512m --instruction-set-variant=cortex-a7 --instruction-set-features=default --dex-file=/data/app/com.example.realsense_app-2/split_lib_slice_1_apk.apk --oat-file=/data/dalvik-cache/arm/data@app@com.example.realsense_app-2@split_lib_slice_1_apk.apk@classes.dex) because non-0 exit status
07-05 06:02:21.762 11520-11520/com.example.realsense_app W/art: Failed execv(/system/bin/dex2oat --runtime-arg -classpath --runtime-arg  --instruction-set=arm --instruction-set-features=smp,div,-atomic_ldrd_strd --runtime-arg -Xrelocate --boot-image=/system/framework/boot.art --runtime-arg -Xms64m --runtime-arg -Xmx512m --instruction-set-variant=cortex-a7 --instruction-set-features=default --dex-file=/data/app/com.example.realsense_app-2/split_lib_slice_2_apk.apk --oat-file=/data/dalvik-cache/arm/data@app@com.example.realsense_app-2@split_lib_slice_2_apk.apk@classes.dex) because non-0 exit status
07-05 06:02:21.854 11520-11520/com.example.realsense_app W/art: Failed execv(/system/bin/dex2oat --runtime-arg -classpath --runtime-arg  --instruction-set=arm --instruction-set-features=smp,div,-atomic_ldrd_strd --runtime-arg -Xrelocate --boot-image=/system/framework/boot.art --runtime-arg -Xms64m --runtime-arg -Xmx512m --instruction-set-variant=cortex-a7 --instruction-set-features=default --dex-file=/data/app/com.example.realsense_app-2/split_lib_slice_3_apk.apk --oat-file=/data/dalvik-cache/arm/data@app@com.example.realsense_app-2@split_lib_slice_3_apk.apk@classes.dex) because non-0 exit status
07-05 06:02:21.959 11520-11520/com.example.realsense_app W/art: Failed execv(/system/bin/dex2oat --runtime-arg -classpath --runtime-arg  --instruction-set=arm --instruction-set-features=smp,div,-atomic_ldrd_strd --runtime-arg -Xrelocate --boot-image=/system/framework/boot.art --runtime-arg -Xms64m --runtime-arg -Xmx512m --instruction-set-variant=cortex-a7 --instruction-set-features=default --dex-file=/data/app/com.example.realsense_app-2/split_lib_slice_4_apk.apk --oat-file=/data/dalvik-cache/arm/data@app@com.example.realsense_app-2@split_lib_slice_4_apk.apk@classes.dex) because non-0 exit status
07-05 06:02:22.051 11520-11520/com.example.realsense_app W/art: Failed execv(/system/bin/dex2oat --runtime-arg -classpath --runtime-arg  --instruction-set=arm --instruction-set-features=smp,div,-atomic_ldrd_strd --runtime-arg -Xrelocate --boot-image=/system/framework/boot.art --runtime-arg -Xms64m --runtime-arg -Xmx512m --instruction-set-variant=cortex-a7 --instruction-set-features=default --dex-file=/data/app/com.example.realsense_app-2/split_lib_slice_5_apk.apk --oat-file=/data/dalvik-cache/arm/data@app@com.example.realsense_app-2@split_lib_slice_5_apk.apk@classes.dex) because non-0 exit status
07-05 06:02:22.149 11520-11520/com.example.realsense_app W/art: Failed execv(/system/bin/dex2oat --runtime-arg -classpath --runtime-arg  --instruction-set=arm --instruction-set-features=smp,div,-atomic_ldrd_strd --runtime-arg -Xrelocate --boot-image=/system/framework/boot.art --runtime-arg -Xms64m --runtime-arg -Xmx512m --instruction-set-variant=cortex-a7 --instruction-set-features=default --dex-file=/data/app/com.example.realsense_app-2/split_lib_slice_6_apk.apk --oat-file=/data/dalvik-cache/arm/data@app@com.example.realsense_app-2@split_lib_slice_6_apk.apk@classes.dex) because non-0 exit status
07-05 06:02:22.243 11520-11520/com.example.realsense_app W/art: Failed execv(/system/bin/dex2oat --runtime-arg -classpath --runtime-arg  --instruction-set=arm --instruction-set-features=smp,div,-atomic_ldrd_strd --runtime-arg -Xrelocate --boot-image=/system/framework/boot.art --runtime-arg -Xms64m --runtime-arg -Xmx512m --instruction-set-variant=cortex-a7 --instruction-set-features=default --dex-file=/data/app/com.example.realsense_app-2/split_lib_slice_7_apk.apk --oat-file=/data/dalvik-cache/arm/data@app@com.example.realsense_app-2@split_lib_slice_7_apk.apk@classes.dex) because non-0 exit status
07-05 06:02:22.330 11520-11520/com.example.realsense_app W/art: Failed execv(/system/bin/dex2oat --runtime-arg -classpath --runtime-arg  --instruction-set=arm --instruction-set-features=smp,div,-atomic_ldrd_strd --runtime-arg -Xrelocate --boot-image=/system/framework/boot.art --runtime-arg -Xms64m --runtime-arg -Xmx512m --instruction-set-variant=cortex-a7 --instruction-set-features=default --dex-file=/data/app/com.example.realsense_app-2/split_lib_slice_8_apk.apk --oat-file=/data/dalvik-cache/arm/data@app@com.example.realsense_app-2@split_lib_slice_8_apk.apk@classes.dex) because non-0 exit status
07-05 06:02:22.430 11520-11520/com.example.realsense_app W/art: Failed execv(/system/bin/dex2oat --runtime-arg -classpath --runtime-arg  --instruction-set=arm --instruction-set-features=smp,div,-atomic_ldrd_strd --runtime-arg -Xrelocate --boot-image=/system/framework/boot.art --runtime-arg -Xms64m --runtime-arg -Xmx512m --instruction-set-variant=cortex-a7 --instruction-set-features=default --dex-file=/data/app/com.example.realsense_app-2/split_lib_slice_9_apk.apk --oat-file=/data/dalvik-cache/arm/data@app@com.example.realsense_app-2@split_lib_slice_9_apk.apk@classes.dex) because non-0 exit status
07-05 06:02:22.442 11520-11520/com.example.realsense_app I/InstantRun: starting instant run server: is main process
07-05 06:02:23.020 11520-11520/com.example.realsense_app W/art: Before Android 4.1, method android.graphics.PorterDuffColorFilter android.support.graphics.drawable.VectorDrawableCompat.updateTintFilter(android.graphics.PorterDuffColorFilter, android.content.res.ColorStateList, android.graphics.PorterDuff$Mode) would have incorrectly overridden the package-private method in android.graphics.drawable.Drawabl [... cut ...]
    Build Date: 06/07/16 Tue
    Local Branch: 
    Remote Branch: quic/LA.BR.1.2.7_rb1.21
    Local Patches: NONE
    Reconstruct Branch: NOTHING
07-05 06:02:23.446 11520-11722/com.example.realsense_app I/OpenGLRenderer: Initialized EGL, version 1.4
07-05 06:02:23.953 11520-11520/com.example.realsense_app W/art: Before Android 4.1, method int android.support.v7.widget.DropDownListView.lookForSelectablePosition(int, boolean) would have incorrectly overridden the package-private method in android.widget.ListView

RealSense-Customer-Engineering, 2018-07-09:
[Realsense Customer Engineering Team Comment]
@fotsing365
Please try below commands before start the application. Need execute the commands for each phone reboot.

    adb shell
    su
    chmod 0777 /dev/bus/usb/*
    chmod 0777 /dev/bus/usb/*/*
    chmod 0777 /dev/video*
    setenforce 0

fotsing365, 2018-07-09:
HI @RealSense-Customer-Engineering I just repeated it more time but result don't change. I still face to this crash. I would like to mention that I used NDK 16b to build project. And you @aaru2211 have succed to run the apps?

bilalbinrais, 2018-07-10:
exact same problem with me. im on android 7.0. Rooted device. getenforce returns permissive.  is there any way to see what crash log contains?

fotsing365, 2018-07-11:
Hi everyone.  
After configuring my tablet using `adb tcpip`commands I am now able to see logcat message through wifi. While realsense_app crashing  **logcat** consol displays this message:

OpenGL ES Shader Compiler Version: E031.29.00.00
    Build Date: 06/07/16 Tue
    Local Branch: 
    Remote Branch: quic/LA.BR.1.2.7_rb1.21
    Local Patches: NONE
    Reconstruct Branch: NOTHING
07-11 06:19:57.793 28664-28775/com.example.realsense_app I/OpenGLRenderer: Initialized EGL, version 1.4
07-11 06:19:57.965 28664-28664/com.example.realsense_app W/art: Before Android 4.1, method int android.support.v7.widget.DropDownListView.lookForSelectablePosition(int, boolean) would have incorrectly overridden the package-private method in android.widget.ListView
07-11 06:19:58.102 28664-28768/com.example.realsense_app E/art: JNI ERROR (app bug): accessed stale local reference 0x200019 (index 6 in a table of size 1)
07-11 06:19:58.132 28664-28768/com.example.realsense_app A/art: art/runtime/java_vm_ext.cc:410] JNI DETECTED ERROR IN APPLICATION: use of deleted local reference 0x200019
    art/runtime/java_vm_ext.cc:410] "Thread-1884" prio=5 tid=14 Runnable
    art/runtime/java_vm_ext.cc:410]   | group="main" sCount=0 dsCount=0 obj=0x12e7c0a0 self=0xb85ade48
    art/runtime/java_vm_ext.cc:410]   | sysTid=28768 nice=0 cgrp=default sched=0/0 handle=0xa326f930
    art/runtime/java_vm_ext.cc:410]   | state=R schedstat=( 141754065 12984482 175 ) utm=12 stm=2 core=2 HZ=100
    art/runtime/java_vm_ext.cc:410]   | stack=0xa3173000-0xa3175000 stackSize=1014KB
    art/runtime/java_vm_ext.cc:410]   | held mutexes= "mutator lock"(shared held)
    art/runtime/java_vm_ext.cc:410]   native: #00 pc 0035a447  /system/lib/libart.so (_ZN3art15DumpNativeStackERNSt3__113basic_ostreamIcNS0_11char_traitsIcEEEEiPKcPNS_9ArtMethodEPv+126)
    art/runtime/java_vm_ext.cc:410]   native: #01 pc 0033afcf  /system/lib/libart.so (_ZNK3art6Thread4DumpERNSt3__113basic_ostreamIcNS1_11char_traitsIcEEEE+138)
    art/runtime/java_vm_ext.cc:410]   native: #02 pc 0024de43  /system/lib/libart.so (_ZN3art9JavaVMExt8JniAbortEPKcS2_+750)
    art/runtime/java_vm_ext.cc:410]   native: #03 pc 0024e54b  /system/lib/libart.so (_ZN3art9JavaVMExt9JniAbortFEPKcS2_z+62)
    art/runtime/java_vm_ext.cc:410]   native: #04 pc 0033bd1b  /system/lib/libart.so (_ZNK3art6Thread13DecodeJObjectEP8_jobject+678)
    art/runtime/java_vm_ext.cc:410]   native: #05 pc 000ffa3f  /system/lib/libart.so (_ZN3art11ScopedCheck5CheckERNS_18ScopedObjectAccessEbPKcPNS_12JniValueTypeE.constprop.95+666)
    art/runtime/java_vm_ext.cc:410]   native: #06 pc 0010e9c9  /system/lib/libart.so (_ZN3art8CheckJNI11CallMethodVEPKcP7_JNIEnvP8_jobjectP7_jclassP10_jmethodIDSt9__va_listNS_9Primitive4TypeENS_10InvokeTypeE+636)
    art/runtime/java_vm_ext.cc:410]   native: #07 pc 0010f2c7  /system/lib/libart.so (_ZN3art8CheckJNI21CallStaticVoidMethodVEP7_JNIEnvP7_jclassP10_jmethodIDSt9__va_list+26)
    art/runtime/java_vm_ext.cc:410]   native: #08 pc 00018e6b  /data/app/com.example.realsense_app-2/lib/arm/libnative-lib.so (_ZN7_JNIEnv20CallStaticVoidMethodEP7_jclassP10_jmethodIDz+82)
    art/runtime/java_vm_ext.cc:410]   native: #09 pc 000184a1  /data/app/com.example.realsense_app-2/lib/arm/libnative-lib.so (???)
    art/runtime/java_vm_ext.cc:410]   native: #10 pc 00018255  /data/app/com.example.realsense_app-2/lib/arm/libnative-lib.so (???)
    art/runtime/java_vm_ext.cc:410]   native: #11 pc 00018237  /data/app/com.example.realsense_app-2/lib/arm/libnative-lib.so (???)
    art/runtime/java_vm_ext.cc:410]   native: #12 pc 000180cb  /data/app/com.example.realsense_app-2/lib/arm/libnative-lib.so (???)
    art/runtime/java_vm_ext.cc:410]   native: #13 pc 00022107  /data/app/com.example.realsense_app-2/lib/arm/libnative-lib.so (???)
    art/runtime/java_vm_ext.cc:410]   native: #14 pc 00041693  /system/lib/libc.so (_ZL15__pthread_startPv+30)
    art/runtime/java_vm_ext.cc:410]   native: #15 pc 00019265  /system/lib/libc.so (__start_thread+6)
    art/runtime/java_vm_ext.cc:410]   (no managed stack frames)
    art/runtime/java_vm_ext.cc:410] 
07-11 06:19:58.302 28664-28786/com.example.realsense_app A/libc: Fatal signal 11 (SIGSEGV), code 1, fault addr 0x9f11b000 in tid 28786 (Thread-1884)
`
`

fotsing365, 2018-07-11:
Please take  a look on this [link](https://stackoverflow.com/questions/14765776/jni-error-app-bug-accessed-stale-local-reference-0xbc00021-index-8-in-a-tabl). They could have solution on this issues.

bilalbinrais, 2018-07-25:
Fixed in this thread. https://github.com/IntelRealSense/librealsense/issues/2075

RealSense-Customer-Engineering, 2018-07-26:
[Realsense Customer Engineering Team Comment]
@fotsing365 Can the resolution in https://github.com/IntelRealSense/librealsense/issues/2075 work for you?

RealSense-Customer-Engineering, 2018-08-15:
[Realsense Customer Engineering Team Comment]
@fotsing365 Is the issue resolved? Please update. Thanks!

RealSense-Customer-Engineering, 2018-08-23:
[Realsense Customer Engineering Team Comment]
@fotsing365 Any good news for this issue? Could you please update? Thanks!

RealSense-Customer-Engineering, 2018-09-03:
[Realsense Customer Engineering Team Comment]
@fotsing365 Can we follow up the remaining issue in #2143 and close this ticket accordingly? Thanks!

PR #2285 (2018-08-22): Android run fix
- Added PIE linker flags (when compiled in terminal as in the tutorial, file couldn't run because  of "Only position independent executables (PIE) are supported" error);
- Fixed local reference issue in the example code for the tutorial (as suggested by @aarun2211 and @bilalbinrais );
- In the .md files:  -DBUILD_SHARED_LIBS=false solves "librealsense2.so not found" error.

Fixes: #2008 #2075
closed 2018-09-11 by fotsing365