realsenseai/librealsense #2009: Extremely Poor .ply file quality

jasonjsong, 2018-07-05:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.09.02.00 | 
| Operating System & Version |   Win10  | 
| Platform                                 | PC  |
| SDK Version                            | 2.0                  | 
| Language                            |  {Python }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
Hello all! I have been working on a program that requires me to take snapshots of the camera and save them in a cloud point format, similar to the way a cloud point is displayed and saved in the "3D" section of the Intel Realsense Viewer. However, when opened in meshlab, the .ply files are seemingly darkand seem to hold no significant data. I have tried different lighting and distances to no avail. 
Am I missing something obvious here, or is there maybe something that I need to configure in the "settings" of my code? Any help is appreciated!

[image]

jasonjsong, 2018-07-05:
To clarify, this is just captured using the basic "example" code "export_ply_example" in the example folder of the SDK

RealSense-Customer-Engineering, 2018-07-06:
[Realsense Customer Engineering Team Comment]
hi @jasonjsong,

can you show us what's your scene?
maybe you capture with realsense viewer and export_ply_example together and show us why you feel the quality is poor.
thanks.

RealSense-Customer-Engineering, 2018-07-06:
[Realsense Customer Engineering Team Comment]
@jasonJsong,
another possibility is the ply viewer's scale/unit is different
the object you capture may just a very small object inside the ply viewer

jasonjsong, 2018-07-06:
I have taken a few screenshots of what appears.
[image]
This is the scene as seen from the viewer in 2D.
[image]
This is the scene as seen in 3D from the viewer, this is the kind of quality that I am looking for.
[image]
This is the ply file I get when I export the 3D scene as a PLY, still acceptable quality, and easily viewable. The subject in question is the plant, which is still clear in this file.
[image]
However this is what I get from the export.ply. It is missing the plant entirely. You can see that the orientation of the doors on the right hand side is correct, although they are fairly difficult to make out. Is this just the nature of the simplicity of the example program? Or am I doing something incorrectly? Thank you.

AntoineWefit, 2018-07-09:
I think that the first thing you want to do is to adjust the parameters of the camera.

If you're only interested in capturing the plant, then you should adjust the **Depth Clamp Min/Max** under _Stereo Module/Advanced Controls/Depth Table_ in the Intel RealSense Viewer. You'll capture only a small part of your scene, plus that may help you "get rid" of the light in the ceiling (which is never helping btw).
I don't know if it's going to work for you, but I suggest you then set the **Depth Units** to 100 (under _Advanced Controls/Depth Table_), and the other **Depth Units** to 0.0001 (under _Controls_).
That being done, try adjusting the **Disparity Shift** (under _Advanced Controls/Depth Table_) and maybe the **Exposure** and the **Laser Power** (under _Controls_) depending on the lighting conditions. 
Finally, enable the **Post-Processing** for the max achievable quality. (Obviously, there is a lot more parameters, but these ones are a good start).
Just export the ".json" if you don't want to use the Viewer in your workflow next time.

One thing being weird though, is that you seem to lose some data when you export the ".ply", I don't know if it's going to correct that.

Keep us updated

bruceweir, 2018-09-25:
This same issue occurred for me after running a firmware update. The solution was to turn the camera off, then on again....

closed 2018-12-12 by RealSense-Customer-Engineering