realsenseai/librealsense #2010: calculate depth if point known in two stereo images

timprepscius, 2018-07-05:
D435, all operating systems - slightly off topic perhaps

---

I am wondering if there is code in any of the examples, or tools- or anywhere else, which given a known point in two stereo images (and as well all extrinsics,intrinsics), derives the depth and/or 3d coordinates.

I can obviously figure this out via wikipedia, but, maybe this already exists in this code base.

Thanks

-tim

dorodnic, 2018-07-08:
Hi @timprepscius 
Something like:
```python
depth[disparity > 0] = (fx * baseline) / (units * disparity[disparity > 0])

fx          # lense focal length
baseline    # distance in mm between the two cameras
units       # depth units
```
(From [Depth from Stereo](https://github.com/IntelRealSense/librealsense/blob/master/doc/depth-from-stereo.md))
Of course the secret sauce is finding the correct match with sub-pixel accuracy (disparity matrix can and should include floating point values, otherwise your depth will alway be quantised to set of discreet values)

timprepscius, 2018-07-08:
Ah, this is interesting.
Thank you for this.

timprepscius, 2018-07-08:
Do you know what unit disparity is in?

Is it in pixels?  Or percentage (0->1.0) of width?  

I've been reading every doc on the web for building the camera matrixes for triangulating points - so my brain is a bit mush right now.  I'm trying to get cv::triangulatePoints to correspond (in a distance sort of way) to the formula from Depth from Stereo, but as of now no luck.

dorodnic, 2018-07-09:
Units of disparity are pixels (`fx` is the factor responsible for converting it to mm)
Would you like to share what you are trying to do, perhaps people here could help?

timprepscius, 2018-07-09:
I am trying to do this:

I have a point P in stereo-image 1 and P' in stereo-image 2. (using _Y8)
Both P and P' correspond to a single point in 3d space.

I want to calculate the 3d position of point P in relation to stereo-image 1.

--

I believe I should:
1. Disparity = (P' - P).x 
2. Depth = (fx * baseline) / (units * disparity)
3. rs2_deproject_pixel using P, Depth and the Intrisics for stereo-image-1-camera

At the moment I'm getting nonsensical results.
My best guess is that it is because the intrisics' coeffs are 0.

timprepscius, 2018-07-10:
It works, next is verifying with opencv triangulation, but will wait till later

RealSense-Customer-Engineering, 2018-07-23:
[Realsense Customer Engineering Team Comment]
@timprepscius
I close this ticket first since the code works, if there is question after your validation, please raise another question. Thank you.

closed 2018-07-23 by RealSense-Customer-Engineering