realsenseai/librealsense #2012: R200 on Visual Studio 2017

DMG22, 2018-07-06:
### Issue Description
I've recently borrowed an R200 camera for a university project and am struggling to get it to work. I have the right DCM installed and can successfully build the librealsense files with CMake. When I run realsense-viewer.cpp it works for a short time and then opens a white window and produces the following:

**Exception thrown at 0x744508B2 in rs-capture.exe: Microsoft C++ exception: std::runtime_error at memory location 0x00D6ECCC.
The thread 0x3530 has exited with code 0 (0x0).
Exception thrown at 0x744508B2 in rs-capture.exe: Microsoft C++ exception: std::runtime_error at memory location 0x00D6ECCC.
The thread 0x1fd0 has exited with code 0 (0x0).
Exception thrown at 0x744508B2 in rs-capture.exe: Microsoft C++ exception: std::runtime_error at memory location 0x00D6ECCC.
Exception thrown at 0x744508B2 in rs-capture.exe: Microsoft C++ exception: [rethrow] at memory location 0x00000000.
Exception thrown at 0x744508B2 in rs-capture.exe: Microsoft C++ exception: [rethrow] at memory location 0x00000000.
Exception thrown at 0x744508B2 in rs-capture.exe: Microsoft C++ exception: rs2::error at memory location 0x00D6F1EC.
Exception thrown at 0x744508B2 in rs-capture.exe: Microsoft C++ exception: std::runtime_error at memory location 0x00D6E3B0.
Exception thrown at 0x744508B2 in rs-capture.exe: Microsoft C++ exception: [rethrow] at memory location 0x00000000.
The thread 0x2548 has exited with code 0 (0x0).
The thread 0x1038 has exited with code 0 (0x0).
The thread 0x31c has exited with code 0 (0x0).
The thread 0x518 has exited with code 0 (0x0).
The thread 0x3f08 has exited with code 0 (0x0).
The thread 0x1f34 has exited with code 0 (0x0).
The thread 0x3ff4 has exited with code 0 (0x0).
'rs-capture.exe' (Win32): Unloaded 'C:\Windows\SysWOW64\ntasn1.dll'
'rs-capture.exe' (Win32): Unloaded 'C:\Windows\SysWOW64\ncrypt.dll'
'rs-capture.exe' (Win32): Unloaded 'C:\Windows\SysWOW64\igd10iumd32.dll'
The thread 0x468 has exited with code 0 (0x0).
'rs-capture.exe' (Win32): Unloaded 'C:\Windows\SysWOW64\PortableDeviceTypes.dll'
'rs-capture.exe' (Win32): Unloaded 'C:\Windows\SysWOW64\deviceaccess.dll'
'rs-capture.exe' (Win32): Unloaded 'C:\Windows\SysWOW64\mfsensorgroup.dll'
'rs-capture.exe' (Win32): Unloaded 'C:\Windows\SysWOW64\Windows.Media.dll'
'rs-capture.exe' (Win32): Unloaded 'C:\Windows\SysWOW64\SensorsUtilsV2.dll'
'rs-capture.exe' (Win32): Unloaded 'C:\Windows\SysWOW64\SensorsNativeApi.V2.dll'
'rs-capture.exe' (Win32): Unloaded 'C:\Windows\SysWOW64\SensorsApi.dll'
'rs-capture.exe' (Win32): Unloaded 'C:\Windows\SysWOW64\winmmbase.dll'
'rs-capture.exe' (Win32): Unloaded 'C:\Windows\SysWOW64\winmm.dll'
The thread 0x3368 has exited with code 1 (0x1).
The thread 0x316c has exited with code 1 (0x1).
The thread 0x3ac0 has exited with code 1 (0x1).
The thread 0x3dcc has exited with code 1 (0x1).
The thread 0x1fe0 has exited with code 1 (0x1).
The thread 0x1d84 has exited with code 1 (0x1).
The thread 0x3c1c has exited with code 1 (0x1).
The thread 0x3ed8 has exited with code 1 (0x1).
The thread 0x2c08 has exited with code 1 (0x1).
The thread 0x2e0 has exited with code 1 (0x1).
The thread 0x2a34 has exited with code 1 (0x1).
The thread 0x2328 has exited with code 1 (0x1).
The thread 0x4d0 has exited with code 1 (0x1).
The thread 0x3c9c has exited with code 1 (0x1).
The thread 0xd50 has exited with code 1 (0x1).
The program '[12452] rs-capture.exe' has exited with code 1 (0x1).**

My feeling is that the R200 itself is not configured correctly for two reasons.
1. The debug output is the same even if the R200 is unplugged.
2. In the device manager, all of the R200 depth, left-right, and rgb items are listed under 'Cameras', not 'Imaging Devices'.

I have tried reinstalling the DCM twice with no change. I'm not a compsci student and this is my first time with C++ so any help is appreciated.

RealSense-Customer-Engineering, 2018-07-06:
[Realsense Customer Engineering Team Comment]
hi @DMG22,

for running realsense viewer and libRealsense, you don't need DCM, so would suggest to remove it in case of any interference.
and would like to know if you can run pre-built realsense viewer with R200 you have? so we can make sure your system is good.

DMG22, 2018-07-09:
Hi @RealSense-Customer-Engineering ,

I tried uninstalling DCM, and I still have the same result when I try to use the librealsense viewer. I also downloaded the Intel Realsense Viewer v2.12.0, and the program is asking me to connect a realsense camera or add source. I can still see the Depth, Left-Right and RGB items under Cameras in the device manager, so I am pretty sure the USB connection is working to some degree.

RealSense-Customer-Engineering, 2018-07-09:
[Realsense Customer Engineering Team Comment]
hi @DMG22,

the LibRealsense v2 is not support R200, so you need to use the legacy version of librealsense which are versions 1.12.0 and earlier.

DMG22, 2018-07-10:
Hi @RealSense-Customer-Engineering,

Thanks for pointing that out. I have downloaded the legacy version of the SDK, but I have some new problems. 

When I run CMake I get the following error:

**G3D193B2D: #error At least Visual Studio 2013 Update 4 is required to compile this backend**

This is strange because I am using the latest version of Visual Studio 2017.

My other problem is that Visual Studio fails to open any of the source files specified in the C and C++ headers. Below are two examples:

**E1696	cannot open source file "cstdint"	
E1696	cannot open source file "cstddef"**	

I have searched for these files in the visual studio folder and they are definitely there. This leads me to believe it is an issue with the file path. I attempted to access Visual C++ directories in the settings but MS has apparently changed VC++ directories so you can only specify them for individual projects. I can't figure out how to open librealsense as a project, and I can't edit VC++ directories for a CMake file or non project file. I even tried copying the MSVC folder with all the C++ source files into the librealsense directory, but VS still cant find the files.

RealSense-Customer-Engineering, 2018-07-17:
[Realsense Customer Engineering Team Comment]
hi @DMG22,

per installation guide:
https://github.com/IntelRealSense/librealsense/blob/v1.12.1/doc/installation_windows.md
suggest to use Visual Studio 2013 Release 4, both Professional and Community editions.
your following issue seems also due to visual studio version not match so you don't have those files.

closed 2018-07-31 by RealSense-Customer-Engineering