realsenseai/librealsense #2014: DistortionRGB = 0 ( d415)

omerbrandis, 2018-07-08:
Dear support,

I have used "CustomRW -r " to get an xml of a d415 camera, 
the file generated includes the following section 
   <param name = "HasRGB">
        <value>1</value>
    </param>
    <param name = "ResolutionRGB">
        <value>1920</value>
        <value>1080</value>
    </param>
    <param name = "FocalLengthRGB">
        <value>1381.06</value>
        <value>1380.51</value>
    </param>
    <param name = "PrincipalPointRGB">
        <value>960.317</value>
        <value>500.743</value>
    </param>
    <param name = "DistortionRGB">
        **<value>0</value>
        <value>0</value>
        <value>0</value>
        <value>0</value>
        <value>0</value>**
    </param>

the values for DistortionRGB seem wrong ...
can someone please explain them ?

( running ubuntu 16.04, Firm ware  : 05.09.11.00, 
librealsense2.so.2.9.1 ( otherwise  CustomRW will not run) )

thanks,
Omer.

dorodnic, 2018-07-08:
Please see #1430

omerbrandis, 2018-07-09:
I have reviewed #1430, and accept the answer written there.

can you please also shed some light on 
   <param name = "PrincipalPointRGB">
        <value>960.317</value>
        <value>500.743</value>

a. the value  500.743 is a bit surprising since its very far from half of the total number of pixel on axis y (1080)

b. should the values for  PrincipalPoint change for different resolutions? 
and if so, can i use CustomRW to generate them ? ( i did not see a that CustomRW has a resolution parameter).

Thanks,
Omer.

omerbrandis, 2018-07-09:
i see that github has altered my last post.

the values i received for PrincipalPointRGB are : 960.317,500.743.

Omer.

RealSense-Customer-Engineering, 2018-07-23:
[Realsense Customer Engineering Team Comment]
hi @omerbrandis,

for your question a) it seems a bit far from 540, but calibration handled this.
b), i don't think the value change for different resolutions.

RealSense-Customer-Engineering, 2018-08-09:
[Realsense Customer Engineering Team Comment]
hi @omerbrandis,
do you need more support on this?

closed 2018-08-09 by omerbrandis