realsenseai/librealsense #2015: How to change Disparity Shift in Python

Litvak1, 2018-07-08:
Required Info |  
-- | --
Camera Model | D415
Firmware Version | 05.08.15.00
Operating System & Version | Linux (Ubuntu 16.04)
Kernel Version (Linux Only) | 4.13.0-45-generic
Platform | PC
SDK Version | ?
Language | Python

I want to take images of a table with objects, from different heights. As I have seen in realsense-viewer, if I change the value of Disparity Shift from 0 to 128, I can get images from half the MinZ range.

I need to change the Disparity Shift value during a script in python. How can I do that?

I found `pyrealsense2.STDepthTableControl.disparityShift` but didn't understand how to use it.

Thanks

dorodnic, 2018-07-09:
Hi @Litvak1 
Please see [python-rs400-advanced-mode-example.py](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/python-rs400-advanced-mode-example.py)

Litvak1, 2018-07-09:
Hi @dorodnic ,

I actually looked at the example you've mentioned before and didn't understand it. Trying again I realized what I had to do:

> device = rs.context().query_devices()[0]
> advnc_mode = rs.rs400_advanced_mode(device)
> depth_table_control_group = advnc_mode.get_depth_table()
> depth_table_control_group.disparityShift = 128
> advnc_mode.set_depth_table(depth_table_control_group)

Thanks!

closed 2018-07-09 by Litvak1

dorodnic, 2018-07-11:
Hi @Litvak1 
Sorry, we certainly can add more comments to make it easier to understand.
Glad you figured it out, always feel free open more issues if you encounter any problems!