realsenseai/librealsense #2016: Bounding box

DannySauval, 2018-07-09:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.09.13.00 | 
| Operating System & Version |  Win 10 + Visual Studio 15 (2017)  | 
| Platform                                 | PC |
| SDK Version                            |  2.0                          | 
| Language                            |  C++                          | 

### Issue Description

Hello, 

I looked a lot to the source code of the librealsense and checked out a lot of examples, but I couldn't find anything about adding a bounding box to what the camera outputs. I actually just want a scan of my face, I don't want far away info. I would like to do it through intel realsense SDK since when I use PCL I don't get a lot of framerates (because of the function pointsToPCL I think).

Do you know how I can easily do that through the SDK ?

Thank you !

Danny,

AntoineWefit, 2018-07-09:
You can easily tune the "depth of acquisition" (which would be the depth of the bounding box you are talking about) by adjusting the **Depth Clamp Min/Max** in the Intel RealSense Viewer, and export your ".json".

However, I don't believe there is a way to "crop" the FOV of the camera within the SDK (let me know if i'm wrong because I want to know too).
You can set a ROI, which is used for calculating the exposure I believe, but It doesn't act like a bounding box for the points cloud.

If you actually want to scan a face, I would recommend that you simply put the camera as close as possible to the target by making it fitting tightly in the FOV (use the Viewer to find the best parameters). You won't really need to define a bounding box when being close to the edges of the FOV, and you'll have the max achievable quality (as the error depends of the distance). I believe you can set the camera at around 250-300 mm away from your face to start tuning. The FOV of the camera in the "normal" position ("landscape") is obviously a horizontal rectangle, so you can also try to rotate it by 90° to fit the face better ("portrait" mode ?), and thus focusing almost exclusively on the acquisition of your face.

Keep us updated

DannySauval, 2018-07-11:
@AntoineWefit thank you for your answer ! Your solution did the trick. I set up depth clamp Min/Max so that I can only see the face, and when the camera is close enough I indeed don't need to crop the rest ! It would have been better to have a real way to do it, but for now it works so it is okay. I hope developers will add something to efficiently add a bounding box. The SDK seems very powerful but too young yet.

@LeoSarog thank you for your answer. But I don't want to use PCL, I already added a bounding box through it but as I said the frame drop due to converting is too much for me. That is why I want to only use the RealSense SDK.

closed 2018-07-11 by RealSense-Customer-Engineering