realsenseai/librealsense #2017: Error when try to attach D415 to  a  virtual machine

tarekmuallim, 2018-07-09:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Firmware Version                   |     | 
| Operating System & Version |   {Win (10)   | 
| Kernel Version (Linux Only)    |                                            | 
| Platform                                 | PC   |
| SDK Version                            |                             | 
| Language                            |  {C++ }                          | 
| Segment			|  {VR/ } 

Hi
I am trying to test D415 on Linux ubuntu.
I am using Oracle VM VirtualBox.
But the VM failed to  attach the realsense camera.
[image]

I get message says “Failed to attach the USB device to the virtual machine”.
Did any one have the same problem?

Did any one try to use D415 through a  virtual machine?

argosvr, 2018-07-09:
Yes... but it does not work with Oracle VM Virtual Box. Virtual Box does not have a full support for USB devices. We solved the problem using VMWare which has a good USB support and it works fine.

tarekmuallim, 2018-07-09:
Thanks
I will try VMware

dorodnic, 2018-07-12:
Hi @tarekmuallim 
Did it work with VMWare?

tarekmuallim, 2018-07-12:
Hi @dorodnic 
 
Actually I have just test it now.

VMWare transfer the camera from the real system to the VM without any problem.
But when I run realsense viewer I got this error:

> Backend in rs2_create_device(info_list:0xcb28e0, index:0):
> libusb_bulk_transfer(...) returned LIBUSB_ERROR_TIMEOUT Last Error: Resource temporarily unavailable

It looks something else not related to the driver.
Any way I will check what the problem.

Thanks

RealSense-Customer-Engineering, 2018-09-17:
[Realsense Customer Engineering Team Comment]
Hi @tarekmuallim,

Still need any support for this topic?
Using VMware workstation player should work w/ D415 camera.

closed 2018-09-25 by dorodnic