realsenseai/librealsense #2019: GetIntrinsics() in VB.Net

Maanbalo, 2018-07-09:
| Required Info                                                                                        

| Camera Model                        D435
| Firmware Version                    05.09.13.00
| Operating System & Version  Win 10 Pro for OEM
| Platform                                  PC
| SDK Version                           .NET 4.6                 
| Language                                VB.NET                     
| Segment			         Others            

I am developing an app for my D435 and I am doing it in VB.NET, I could already add the reference Intel.Realsense.dll, but with the realsense2.dll it tells me that it can not add the reference, I have to make sure that I have access to the file and that it is a valid COM component or assembly.

For now the D435 works well for me and I can work with it, but I need to get the intrinsic data to do  aprojection and projection and I do not get it.

it tell me "System.NullReferenceException" for sensor1.

"""
            frame = pipe.WaitForFrames
            depth = frame.DepthFrame
            width3D = depth.Width
            height3D = depth.Height

         
            Dim strea As VideoStreamProfile
            strea = sensor1.VideoStreamProfiles
            intrinseco = strea.GetIntrinsics()

""""

dorodnic, 2018-07-09:
Hi @Maanbalo 
Where is `sensor1` defined? You can also try something like (forgive my VB):
```
frame = pipe.WaitForFrames
depth = frame.DepthFrame
profile = TryCast(depth.Profile, VideoStreamProfile)
profile.GetIntrinsics()
```
I believe it should work

Maanbalo, 2018-07-09:
Hi, Thank you.

I used this one but:

"profile" is: Nothing

System.NullReferenceExeption

Could I used?

 ctx = New Context
 device1list = ctx.QueryDevices()
 device1 = device1list.Item(1)
 sensor1 = device1.QuerySensors.Item(0)
 sensor1 = device1.Sensors.Item(0)
 Dim strea As VideoStreamProfile
 strea = sensor1.VideoStreamProfiles(68)
 intrinseco = strea.GetIntrinsics()

But with this, I need select one number of VideoStreamProfile and perhaps , The VideoStreamProfile is not the same that the D435 used.

RealSense-Customer-Engineering, 2018-07-11:
[Realsense Customer Engineering Team Comment]
@Maanbalo
Please try the code below, assuming you only have one camera on your system, you should get the device with index 0 not 1. The code I tested with my D435 works, let me know if it also works at your side.

`Imports Intel.RealSense`

`Module Module1`

`    Sub Main()`
`        Dim ctx = New Context`
`        Dim devicelist = ctx.QueryDevices()`
`        Dim device = devicelist.Item(0)`
`        Dim sensor = device.QuerySensors.Item(0)`

`        Dim profile = sensor.VideoStreamProfiles(68)`
`        Dim Intrinsics = profile.GetIntrinsics()`
`        Console.WriteLine("Data {0}, {1}", Intrinsics.fx, Intrinsics.fy)`
`        Console.WriteLine()`
`    End Sub`

`End Module`
``

Maanbalo, 2018-07-11:
Hi, Thank you, Now run fine.
one Question...

What is the number of the VideoStreamProfiles that I have to put, by default it is 68?

My code is:

            pipe = New Pipeline()
            pipe.Start()
            frame = pipe.WaitForFrames
            depth = frame.DepthFrame
            Dim ctx = New Context
            Dim devicelist = ctx.QueryDevices()
            Dim device = devicelist.Item(0)
            Dim sensor = device.QuerySensors.Item(0)
            Dim profile = sensor.VideoStreamProfiles(68)
            intrinseco = profile.GetIntrinsics()

RealSense-Customer-Engineering, 2018-07-12:
[Realsense Customer Engineering Team Comment]
@Maanbalo
No, the 68 is not default, I follow your number in your question. You can get back the stream profile list using StreamProfiles property from sensor instance, and enumerate through the list to check which index with the format, resolution, FSP etc... you need.

Maanbalo, 2018-07-17:
Thank you! XD

closed 2018-07-17 by Maanbalo