realsenseai/librealsense #2021: Problem accessing the pyrealsense2 on python with windows 10

pierref33, 2018-07-09:
Hello,
I work with the camera RealSense D435 on windows 10. After having downloaded the SDK (librarylibrealsense-2.12.0) and test the "Intel RealSense Viewer" (that works fine !! ), I was hoping to access to the depth value for a frame of the camera with a python code. I have installed Python 3.6 and  VisualStudio 2017. I then test the code given in github (for instance: access_depth_data.py). However, when I run the code, the 'import pyrealsense2 as rs' line return the following error : **ImportError: DLL load failed: The specified module could not be found.**

I searched on forum for similar problems and installed the pyrealsense2(2.9.1) package via visual studio, I also try to launch the program via the powershell (with python 3.6.1 version), I add to the same directory the pyrealsense2.py and realsense2.dll files. However the errors persist.
I try to use python 2.7 but I was getting the error : **importERROR; importerror: module use of python27.dll conflicts with this version of python.** and was enable to solve this.

Any thought to help me resolve this problem will be more that welcome. 
Thank you in advance !!

RealSense-Customer-Engineering, 2018-07-10:
[Realsense Customer Engineering Team Comment]
@pierref33
When run Python 2.x, you will need to have use pyrealsense build with python 2.x,  same for python 3.x
you can use the prebuilt library (built with python 2.7
32bit (x86) /64 bit (x64) type compatibility also matter

for detail you can refer to below link
https://github.com/IntelRealSense/librealsense/tree/development/wrappers/python

The precompiled binaries shipped with the installer assume **python2.7**. The error `ImportError: DLL load failed: The specified module could not be found` might indicate versions mismatch or architecture (x86 vs x64) mismatch.

1. Open `librealsense2.sln` that was created in the previous step, and build the `pyrealsense2` project
2. Open the output folder of the project (e.g `build\x64-Release\Release\`) and copy `pyrealsense2.pyd` into your python's `site-packages` (e.g `C:\Python27\Lib\site-packages`)
3. Alternatively, copy the build output (`realsense2.dll` and `pyrealsense2.pyd`) next to your script.

and please check if the microsoft 2015 (or 2017)  vc++ redistributable  (x64)  is also installed

you can use python example code to ensure your tools chain is correct

ex : C:\xxxx\librealsense\wrappers\python\examples\python-tutorial-1-depth.py

pierref33, 2018-07-18:
Thank you for your quick response !!!

Your answer helps me a lot, indeed my problem was a problem of an architecture mismatch. By installing everything from the architecture X86, the programs given in exemple in the wrapper/python directory works fine with python 2.7.

However, when I am using the camera D435 and that I get lots of light the depth precision decrease too much. I think the problems come from the exposure that doesn't adapt to the environment.
**Is the Auto-exposure is set by Default for any code running for the camera?
Is there any way to set the auto-exposure, using python and set the exposure manually (using python) if this isn't enough? **

I see some examples for C++:
m_rsDevice->EnableAutoExposure(1.0f);
if (m_cameraInfo.isWide)
	m_rsDevice->SetAeControl(800);
else
	m_rsDevice->SetAeControl(500);

However, I find nothing for python.

Because It's possible to set the depth unit in python using:
profile = pipeline.start(config)
depth_sensor = profile.get_device().first_depth_sensor()	
depth_sensor.set_option(rs.option.depth_units, 0.0001);	

I was wondering if it is possible to do the same for the exposure option, knowing that a Json file give the following options to be set:
"aux-param-autoexposure-setpoint": "400", 
"controls-autoexposure-auto": "False",
"controls-autoexposure-manual": "12220",
"controls-color-autoexposure-auto": "False",
"controls-color-autoexposure-manual": "166", 
"param-autoexposure-setpoint": "400", 

Any other ways to improve the depth precision (At present I do some face detection for up to 5 meters)? 
I have already changed the depth_unit and update the resolution to high quality 848X480 and 15 frames (30 is too much for my computer).

PS_1: If this can help you when I run the code from the file python-rs400-advanced-mode-example.py given in the wrapper/python/example directory and I get the following features for the exposure:
 ('Auto Exposure Control: , Mean Intensity Set Points: 1536)

PS_2: Is the internal processing hardware of the camera (D4 ASIC) exclusively devoted to provide the best images and depth precision?  Is there any ways to do some processing in the camera as an embedded system or all the high-level processing must be performed inevitably on the host computer ? 

PS_3: I don't see any post processing code in python for the camera like in C++. Is there any way to implement an existing one, or should I try to create it from scratch?

I am aware that I have many questions and some are quite far from the original subject if you prefer I can create a post for each of them.

I hope that you can help me again.  Thank you in advance !!

RealSense-Customer-Engineering, 2018-07-19:
[Realsense Customer Engineering Team Comment]
@Pierref33
Please create other ticket that will help everyone can easier input your thought and refer to  for different subject

but as to your first question, yes , the default AE setting is ON,
you can use rs-enumerate-devices.exe -o (use option -o) which include inside the SDK sample
to list all provides options that you can configure and their default value as below
as you can see below, the default configuration for AE is ON (1)
best

Options for RGB Camera
 Supported options:                                    min        max       step  default
    Backlight Compensation                             : 0    ... 1           1     0
    Brightness                                         : -64  ... 64          1     0
    Contrast                                           : 0    ... 64          1     32
    Exposure                                           : 39   ... 10000       39    312
    Gamma                                              : 72   ... 500         1     100
    Hue                                                : -40  ... 40          1     0
    Saturation                                         : 0    ... 128         1     64
    Sharpness                                          : 0    ... 5           1     0
    White Balance                                      : 2800 ... 6500        1     4000
    Enable Auto Exposure                               : 0    ... 1           1     1
    Enable Auto White Balance                          : 0    ... 1           1     1
    Frames Queue Size                                  : 0    ... 32          1     16

RealSense-Customer-Engineering, 2018-07-19:
[Realsense Customer Engineering Team Comment]

Hi @pierref33,
you will need to build the source code to have the executable. but this is just for your reference that you can use that to check any default setting you are interested when needed, ex. auto exposure setting.
it will be helpful that you can create another ticket for other question that other can share their experience on doing Face detection

as to how to  turn on /off AE using python script
you can similar way as in issue #1258 which use python script to tune the options interested.
in your case, you can also see if adjust the laser power can help

all options are listed in rs_options.h
but one thing need to pay attention to is you need to change the option from upper case to lower case for python
ex: RS2_OPTION_LASER_POWER    -> rs.option.laser_power
ex :RS2_OPTION_ENABLE_AUTO_EXPOSURE ->rs.option.enable_auto_exposure

example code quote from #1258

pipeline_profile = pipeline.start(config)
device = pipeline_profile.get_device()
depth_sensor = device.query_sensors()[0]
laser_pwr = depth_sensor.get_option(rs.option.laser_power)
print("laser power = ", laser_pwr)
laser_range = depth_sensor.get_option_range(rs.option.laser_power)
print("laser power range = " , laser_range.min , "~", laser_range.max)
set_laser = 0
#just simply add 10 to test set function
if laser_pwr + 10 > laser_range.max:
set_laser = laser_range.max
else:
set_laser = laser_pwr + 10
depth_sensor.set_option(rs.option.laser_power, set_laser)

pierref33, 2018-07-19:
Ok I think that I understand the mechanism to go from C++ to  python, thank you for taking the time to guide me!

closed 2018-07-19 by pierref33