realsenseai/librealsense #2024: Are the gain and gamma configurable?

carun, 2018-07-10:
| Required Info                         |                       |
|---------------------------------|----------------|
| Camera Model                       | D435              |
| Firmware Version                   | 05.09.13.00    |
| Operating System & Version |   Win 10         |
| Platform                                 | PC                  |
| SDK Version                           |  2.13.0            |
| Language                               |  C++              | 
| Segment			         |  Biometrics    | 

### Issue Description

I couldn't find examples that shows how to adjust the brightness, contrast, exposure, gain and gamma. This is important as we can expect the sensor to be used in low-light environments.

carun, 2018-07-11:
@RealSense-Customer-Engineering thanks, can you point me to the right source please? I looked at the SDK headers but couldn't find anything related to gain/gamma. https://github.com/IntelRealSense/librealsense/blob/master/examples/sensor-control/rs-sensor-control.cpp doesn't provide any references either.

RealSense-Customer-Engineering, 2018-07-11:
[Realsense Customer Engineering Team Comment]
Hi @carun,

Try below example code for depth controls, and the sensor control options shows in rs_option.h.

<Code>
 rs2::device selected_device = selection.get_device();
 auto depth_sensor = selected_device.first<rs2::depth_sensor>();
 if (depth_sensor.supports(RS2_OPTION_EXPOSURE))
 {
  auto range = depth_sensor.get_option_range(RS2_OPTION_EXPOSURE);
  depth_sensor.set_option(RS2_OPTION_EXPOSURE, 40000.f);
 }
 if (depth_sensor.supports(RS2_OPTION_GAIN))
 {
  // Query min and max values:
  auto range = depth_sensor.get_option_range(RS2_OPTION_GAIN);
  depth_sensor.set_option(RS2_OPTION_GAIN, 20.f);
 }
</Code>

You can also refer to below to get more info:
https://github.com/IntelRealSense/librealsense/wiki/API-How-To

carun, 2018-07-11:
Hi @RealSense-Customer-Engineering 

Thanks for the help. Also please note that the users may not have full knowledge of the APIs already. So the elaborate the examples are, the easier for everyone to understand.

I'm just formatting the code here for GitHub community.

```cpp
rs2::device selected_device = selection.get_device();
auto depth_sensor = selected_device.first();
if (depth_sensor.supports(RS2_OPTION_EXPOSURE))
{
    auto range = depth_sensor.get_option_range(RS2_OPTION_EXPOSURE);
    depth_sensor.set_option(RS2_OPTION_EXPOSURE, 40000.f);
}
if (depth_sensor.supports(RS2_OPTION_GAIN))
{
    // Query min and max values:
    auto range = depth_sensor.get_option_range(RS2_OPTION_GAIN);
    depth_sensor.set_option(RS2_OPTION_GAIN, 20.f);
}
```

RealSense-Customer-Engineering, 2018-07-11:
[Realsense Customer Engineering Team Comment]
Thanks, @carun. Will keep improve this.

carun, 2018-07-13:
@RealSense-Customer-Engineering 

Apparently this doesn't work for RGB camera. How do we do this for RGB camera?

closed 2018-07-23 by carun