realsenseai/librealsense #2029: `rs_camera.launch` fails for D435

ChanWaiYang, 2018-07-10:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   | 05.09.14.00 | 
| Operating System & Version |   Ubuntu 16 | 
| Kernel Version (Linux Only)    |  4.15.0-24-generic | 
| Platform                                 | PC |

### Issue Description
I followed the installation steps stated at (https://github.com/intel-ros/realsense/#installation-instructions).
When i run the command 'roslaunch realsense2_camera rs_camera.launch', I encountered a few warnings and the program is stop running.

The warning I have encountered:

- WARNING [140288580376320] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to installation chapter for details.
- [ WARN] [1531210273.286442270]: Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)
- [ WARN] [1531210273.286565515]: Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)
-  10/07 16:11:13,298 WARNING [140288548550400] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to installation chapter for details.
-  10/07 16:11:13,524 ERROR [140289341515648] (types.h:179) xioctl(VIDIOC_S_CTRL) failed Last Error: Numerical result out of range
- [ WARN] [1531210273.524649930]: Reconfigure callback failed with exception xioctl(VIDIOC_S_CTRL) failed Last Error: Numerical result out of range: 

This is the full print out from my terminal.

> PARAMETERS
>  * /camera/realsense2_camera/accel_fps: 1000
>  * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
>  * /camera/realsense2_camera/align_depth: False
>  * /camera/realsense2_camera/camera_aligned_depth_to_color_frame_id: camera_aligned_de...
>  * /camera/realsense2_camera/camera_aligned_depth_to_fisheye_frame_id: camera_aligned_de...
>  * /camera/realsense2_camera/camera_aligned_depth_to_infra1_frame_id: camera_aligned_de...
>  * /camera/realsense2_camera/camera_aligned_depth_to_infra2_frame_id: camera_aligned_de...
>  * /camera/realsense2_camera/color_fps: 30
>  * /camera/realsense2_camera/color_height: 480
>  * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
>  * /camera/realsense2_camera/color_width: 640
>  * /camera/realsense2_camera/depth_fps: 30
>  * /camera/realsense2_camera/depth_height: 480
>  * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
>  * /camera/realsense2_camera/depth_width: 640
>  * /camera/realsense2_camera/enable_color: True
>  * /camera/realsense2_camera/enable_depth: True
>  * /camera/realsense2_camera/enable_fisheye: True
>  * /camera/realsense2_camera/enable_imu: True
>  * /camera/realsense2_camera/enable_infra1: True
>  * /camera/realsense2_camera/enable_infra2: True
>  * /camera/realsense2_camera/enable_pointcloud: False
>  * /camera/realsense2_camera/enable_sync: False
>  * /camera/realsense2_camera/fisheye_fps: 30
>  * /camera/realsense2_camera/fisheye_height: 480
>  * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
>  * /camera/realsense2_camera/fisheye_width: 640
>  * /camera/realsense2_camera/gyro_fps: 1000
>  * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
>  * /camera/realsense2_camera/infra1_fps: 30
>  * /camera/realsense2_camera/infra1_height: 480
>  * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
>  * /camera/realsense2_camera/infra1_width: 640
>  * /camera/realsense2_camera/infra2_fps: 30
>  * /camera/realsense2_camera/infra2_height: 480
>  * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
>  * /camera/realsense2_camera/infra2_width: 640
>  * /camera/realsense2_camera/json_file_path: 
>  * /camera/realsense2_camera/serial_no: 
>  * /rosdistro: kinetic
>  * /rosversion: 1.12.13
> 
> NODES
>   /camera/
>     realsense2_camera (nodelet/nodelet)
>     realsense2_camera_manager (nodelet/nodelet)
[... 42 lines cut ...]
> [ INFO] [1531210273.097725962]: color stream is enabled - width: 640, height: 480, fps: 30
>  10/07 16:11:13,108 WARNING [140288559417088] (backend-v4l2.cpp:866) Empty frame has arrived.
>  10/07 16:11:13,108 WARNING [140288559417088] (backend-v4l2.cpp:866) Empty frame has arrived.
>  10/07 16:11:13,108 WARNING [140288559417088] (backend-v4l2.cpp:866) Empty frame has arrived.
>  10/07 16:11:13,108 WARNING [140288559417088] (backend-v4l2.cpp:866) Empty frame has arrived.
> [ INFO] [1531210273.110181879]: publishStaticTransforms...
> [ INFO] [1531210273.110332622]: RealSense Node Is Up!
>  10/07 16:11:13,113 WARNING [140288548550400] (backend-v4l2.cpp:866) Empty frame has arrived.
>  10/07 16:11:13,113 WARNING [140288548550400] (backend-v4l2.cpp:866) Empty frame has arrived.
>  10/07 16:11:13,113 WARNING [140288548550400] (backend-v4l2.cpp:866) Empty frame has arrived.
>  10/07 16:11:13,113 WARNING [140288548550400] (backend-v4l2.cpp:866) Empty frame has arrived.
>  10/07 16:11:13,285 WARNING [140288580376320] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to installation chapter for details.
> [ WARN] [1531210273.286442270]: Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)
> [ WARN] [1531210273.286565515]: Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)
>  10/07 16:11:13,298 WARNING [140288548550400] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to installation chapter for details.
>  10/07 16:11:13,524 ERROR [140289341515648] (types.h:179) xioctl(VIDIOC_S_CTRL) failed Last Error: Numerical result out of range
> [ WARN] [1531210273.524649930]: Reconfigure callback failed with exception xioctl(VIDIOC_S_CTRL) failed Last Error: Numerical result out of range: 
> ^C[camera/realsense2_camera-3] killing on exit
> [camera/realsense2_camera_manager-2] killing on exit
> [ INFO] [1531210274.925539195]: Interrupt Signal is received! Terminating RealSense Node...
> [rosout-1] killing on exit
> [master] killing on exit
> shutting down processing monitor...
> ... shutting down processing monitor complete
> done

philip-huang, 2018-07-10:
I am experiencing the exact issue here. 
My setup includes a D435 camera running on Ubuntu 16.04 with kernel version 4.15.0-24-generic. I am using a ros wrapper 2.0.3 with a SDK version 2.10.2. My firmware version is v5.9.11

Here is a full print
SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 1000
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/camera_aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/camera_aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/camera_aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/camera_aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 848
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 848
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_imu: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 848
 * /camera/realsense2_camera/gyro_fps: 1000
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/infra1_fps: 30
 * /camera/realsense2_camera/infra1_height: 480
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra1_width: 848
 * /camera/realsense2_camera/infra2_fps: 30
 * /camera/realsense2_camera/infra2_height: 480
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra2_width: 848
 * /camera/realsense2_camera/json_file_path: 
 * /camera/realsense2_camera/serial_no: 
 * /rosdistro: kinetic
 * /rosversion: 1.12.13

NODES
  /camera/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

ROS_MASTER_URI=http://localhost:11311

process[camera/realsense2_camera_manager-1]: started with pid [26146]
process[camera/realsense2_camera-2]: started with pid [26147]
[ INFO] [1531222249.747344375]: Initializing nodelet with 4 worker threads.
[ INFO] [1531222249.780156306]: RealSense ROS v2.0.3
[ INFO] [1531222249.780211283]: Running with LibRealSense v2.11.1
[ INFO] [1531222249.868845999]: getParameters...
[ INFO] [1531222249.905701109]: setupDevice...
[ INFO] [1531222249.905741673]: JSON file is not provided
[ INFO] [1531222249.905777560]: ROS Node Namespace: camera
[ INFO] [1531222249.905807087]: Device Name: Intel RealSense D435
[ INFO] [1531222249.905834666]: Device Serial No: 814412070507
[ INFO] [1531222249.905860632]: Device FW version: 05.09.11.00
[ INFO] [1531222249.905885299]: Device Product ID: 0x0B07
[... 9 lines cut ...]
[ INFO] [1531222249.906177591]: setupPublishers...
[ INFO] [1531222249.909070551]: Expected frequency for depth = 30.00000
[ INFO] [1531222249.939877227]: Expected frequency for infra1 = 30.00000
[ INFO] [1531222249.962154906]: Expected frequency for aligned_depth_to_infra1 = 30.00000
[ INFO] [1531222249.986000149]: Expected frequency for infra2 = 30.00000
[ INFO] [1531222250.009561016]: Expected frequency for aligned_depth_to_infra2 = 30.00000
[ INFO] [1531222250.039104849]: Expected frequency for color = 30.00000
[ INFO] [1531222250.063593541]: Expected frequency for aligned_depth_to_color = 30.00000
[ INFO] [1531222250.092783384]: setupStreams...
 10/07 14:30:50,093 WARNING [140470024492928] (sensor.cpp:330) Unregistered Media formats : [ UYVY ]; Supported: [ ]
[ INFO] [1531222250.094738721]: depth stream is enabled - width: 848, height: 480, fps: 30
[ INFO] [1531222250.095325018]: infra1 stream is enabled - width: 848, height: 480, fps: 30
[ INFO] [1531222250.095885201]: infra2 stream is enabled - width: 848, height: 480, fps: 30
 10/07 14:30:50,096 WARNING [140470024492928] (backend-v4l2.cpp:1156) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
 10/07 14:30:50,096 WARNING [140470024492928] (sensor.cpp:330) Unregistered Media formats : [ RW16 ]; Supported: [ ]
[ INFO] [1531222250.099641198]: color stream is enabled - width: 848, height: 480, fps: 30
[ INFO] [1531222250.108968768]: publishStaticTransforms...
[ INFO] [1531222250.109124226]: RealSense Node Is Up!
 10/07 14:30:50,209 WARNING [140469304547072] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to installation chapter for details.
[ WARN] [1531222250.211297014]: Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)
 10/07 14:30:50,428 ERROR [140470024492928] (types.h:179) xioctl(VIDIOC_S_CTRL) failed Last Error: Numerical result out of range
[ WARN] [1531222250.429313864]: Reconfigure callback failed with exception xioctl(VIDIOC_S_CTRL) failed Last Error: Numerical result out of range: 
 10/07 14:30:50,565 WARNING [140469257860864] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to installation chapter for details.

I would really appreciate any help. Thanks!

RealSense-Customer-Engineering, 2018-07-15:
[Realsense Customer Engineering Team Comment]
Hi @ChanWaiYang, @Philip-huang,

If you have no concern, suggest to downgrade to kernel 4.13 on Ubuntu 16.04 since DKMS package (librealsense2-dkms) is supported for 4.13, not 4.15. This should resolve the metadata warning there and 'rs_camera.launch' should work normally.

RealSense-Customer-Engineering, 2018-07-16:
[Realsense Customer Engineering Team Comment]
Hi @ChanWaiYang,

Although the warning message there, does 'rs_camera.launch' run normally?
The warning should be due to the request of sensor controls in rs435_node.cpp.

RealSense-Customer-Engineering, 2018-07-23:
[Realsense Customer Engineering Team Comment]
Hi @ChanWaiYang,

Any update?

ChanWaiYang, 2018-07-24:
Hi @RealSense-Customer-Engineering Thanks for the follow-up.
With kernel 4.13 I can get the image and point cloud data on Rviz most of the time. However, the input output error still appear occasionally.

>[ INFO] [1532415741.879316930]: RealSense Node Is Up!
 24/07 15:02:26,950 WARNING [140689240405888] (types.cpp:57) get_xu(id=11) failed! Last Error: Input/output error
[ WARN] [1532415746.978018721]: Reconfigure callback failed with exception get_xu(id=11) failed! Last Error: Input/output error:

RealSense-Customer-Engineering, 2018-07-24:
[Realsense Customer Engineering Team Comment]
Hi @ChanWaiYang,

Can you also have kernel log to me using 'dmesg' when you saw this warning?
So far, it's not easy to reproduce it on my platform. And, like to double check your current test configuration: Ubuntu 16.04 + Kernel 4.13 + D435 FW 05.09.14.00 + ROS wrapper 2.0.3 + librs 2.12.0, right?

philip-huang, 2018-07-25:
I am experiencing the same issue as @ChanWaiYang . When I try to use a HighDensity (or any) presets, the input output error still appears sometimes, so I have to replug the camera every time I see this error.

Here is a section of the dmesg when I saw this warning
```
[  106.531627] media: Linux media interface: v0.10
[  106.532983] videodev: loading out-of-tree module taints kernel.
[  106.533077] videodev: module verification failed: signature and/or required key missing - tainting kernel
[  106.537995] Linux video capture interface: v2.00
[  106.548971] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) 435 (8086:0b07)
[  106.550454] uvcvideo: Unable to create debugfs 3-2 directory.
[  106.550493] uvcvideo 3-4:1.0: Entity type for entity Intel(R) RealSense(TM) 435 with was not initialized!
[  106.550495] uvcvideo 3-4:1.0: Entity type for entity Processing 2 was not initialized!
[  106.550497] uvcvideo 3-4:1.0: Entity type for entity Intel(R) RealSense(TM) 435 with was not initialized!
[  106.550498] uvcvideo 3-4:1.0: Entity type for entity Camera 1 was not initialized!
[  106.550583] input: Intel(R) RealSense(TM) 435: Int as /devices/pci0000:00/0000:00:14.0/usb3/3-4/3-4:1.0/input/input16
[  106.550756] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) 435 (8086:0b07)
[  106.551471] uvcvideo: Unable to create debugfs 3-2 directory.
[  106.551510] uvcvideo 3-4:1.3: Entity type for entity Processing 7 was not initialized!
[  106.551512] uvcvideo 3-4:1.3: Entity type for entity Extension 8 was not initialized!
[  106.551513] uvcvideo 3-4:1.3: Entity type for entity Camera 6 was not initialized!
[  106.551583] usbcore: registered new interface driver uvcvideo
[  106.551584] USB Video Class driver (1.1.2.realsense-1.2.0)
[  114.745411] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[  115.097887] uvcvideo: Failed to query (SET_CUR) UVC control 11 on unit 3: -32 (exp. 1).

```

RealSense-Customer-Engineering, 2018-07-26:
[Realsense Customer Engineering Team Comment]
Hi @philip-huang,

You mean you run the dynamic control using "rqt_reconfigure" and see the same warning issue, right?
Per your log below, segfault error also happens. For each time, you replug the camera, then the same segfault appears?

[155.597542] rqt[4356]: segfault at 18 ip 00007f1014ca73c6 sp 00007f0fd5b5adb0 error 4 in libc-2.23.so[7f1014c29000+1c0000]

@ChanWaiYang, @Philip-huang,
Please try to upgrade the firmware to 5.9.14 to see if below warning issue resolved.

 16/07 09:25:14,420 WARNING [140217749550976] (types.cpp:57) get_xu(id=11) failed! Last Error: Input/output error
[WARN] [1531704314.528621388]: Reconfigure callback failed with exception get_xu(id=11) failed! Last Error: Input/output error:

philip-huang, 2018-07-30:
Hi @RealSense-Customer-Engineering 
I have updated the firmware to 5.8.14 and the warning issue is still not resolved.

RealSense-Customer-Engineering, 2018-07-31:
[Realsense Customer Engineering Team Comment]
Hi @philip-huang,

Will further check internally to see if any solution there.

RealSense-Customer-Engineering, 2018-08-21:
[Realsense Customer Engineering Team Comment]
Hi @ChanWaiYang, @Philip-huang,

A possible workaround is to have an simple app to reset D415 device once you have the stream error.  Then, you can re-run the launch file.
The code like below. You can have a try.

rs2::pipeline pipe;
rs2::pipeline_profile profile = pipe.start();
auto dev = profile.get_device();
dev.hardware_reset();

RealSense-Customer-Engineering, 2018-08-28:
[Realsense Customer Engineering Team Comment]
Please also try the firmware to 5.10.3. It should resolve the warning of input/output error.

closed 2018-12-17 by RealSense-Customer-Engineering