realsenseai/librealsense #2031: Unity instructions only generate Win32 projects in Visual Studio

jaredgaul, 2018-07-10:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.09.14.00 | 
| Operating System & Version |   Win 10  | 
| Kernel Version (Linux Only)    |                                           | 
| Platform                                 | PC |
| SDK Version                            | 2                         | 
| Language                            |  unity                          | 
| Segment			|  others                          | 

### Issue Description
Following the instructions to generate the librealsense2.sln Visual Studio project (using version 2017 community) only produced projects for 32bit. Trying to open/run the unity samples after a successful compilation doesn't work because it expects the plugins to have been built in x64 not Win32. Adding an additional option of -DCMAKE_GENERATOR_PLATFORM=x64 to the instructions for cmake fixed the issue though.

RealSense-Customer-Engineering, 2018-07-11:
[Realsense Customer Engineering Team Comment]
Hi @jaredgaul,

To use CMAKE-GUI tool as an alternative way instead of CMAKE command-line, you can configure and generate VS2017 x64 librealsense project easily.

RealSense-Customer-Engineering, 2018-07-17:
[Realsense Customer Engineering Team Comment]
Hi @jaredgaul,

Suggest you can also file a PR to update the instructions to help us improve.

RealSense-Customer-Engineering, 2018-07-26:
[Realsense Customer Engineering Team Comment]
Hi @jaredgaul,

Will close this first. If any more question, you can still re-open or create another ticket. Thanks.

closed 2018-07-26 by RealSense-Customer-Engineering