realsenseai/librealsense #2032: Unity Align Images functionality only works with identical texture resolutions

jaredgaul, 2018-07-10:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.09.14.00 | 
| Operating System & Version |   Win 10  | 
| Kernel Version (Linux Only)    |                                           | 
| Platform                                 | PC |
| SDK Version                            | 2                         | 
| Language                            |  unity                          | 
| Segment			|  others                          | 

### Issue Description
By modifying the Unity sample for the "RealSenseTexturesDepthAndColor" scene I found that the alignment script doesn't work unless the textures for depth and rgb are the same resolution. This doesn't allow us to maximize our rgb resolution to 1920x1080 as doing so ends up with a corrupted depth texture. Having them both at 1280x720 worked fine though.

dorodnic, 2018-07-11:
Sounds like a bug on Unity wrapper level, `align` processing block should be able to work fine with any pair of resolutions. We will look into it.

matkatz1, 2018-07-31:
This issue is solved on release 2.14.1:
[https://github.com/IntelRealSense/librealsense/tree/v2.14.1](https://github.com/IntelRealSense/librealsense/tree/v2.14.1)

RealSense-Customer-Engineering, 2018-08-02:
[Realsense Customer Engineering Team Comment]
@jaredgaul
Please check if your problem is solved. Thank you.

closed 2018-08-14 by RealSense-Customer-Engineering