realsenseai/librealsense #2033: D435 External Connector

timprepscius, 2018-07-10:
D435, all operating systems

----

This information is probably obvious to many, however I am having a difficult time finding exactly what I need.

For the D435.  There is an external connector located on the top of the device.
From the reference specifications I believe it is a:

9 Positions Header, Shrouded Connector
SM09B-SRSS- TB(LF)(SN)
www.jst-mfg.com

Can someone post the exact connector model number which will fit into the D435?

Is it this one?
http://www.jst-mfg.com/product/detail_e.php?series=244

What would even be better is a link to DigiKey or some other reseller for the item.  Looking up the above connector on DigiKey doesn't yield results.  Or it seems to not yield correct results.

Maybe it is this one?
https://www.digikey.com/product-detail/en/jst-sales-america-inc/09JQ-BT/455-2971-ND/4941249

Thanks a lot.

ankarako, 2018-07-12:
@timprepscius  Hello,

I got [these](http://export.farnell.com/jst-japan-solderless-terminals/shr-09v-s-b/connector-housing-9way/dp/1679115), and they fit just fine on D415 & D435 also.
You will also need its corresponding [pin connectors](http://export.farnell.com/jst-japan-solderless-terminals/ssh-003t-p0-2/crimp-terminal-sh-series/dp/1679142?MER=sy-me-pd-mi-acce)

To assemble the connector you will also need the specified crimp tool  for those pins. But it is too expensive, so maybe you want to improvise a little.

In addition (depending on your needs) you will need either just two cables (GND, external-sync signal) or a 9-way ribbon cable (if you want to access all of the pins).

Hope that helps

timprepscius, 2018-07-12:
JST - Some links for other who might be having this problem:
https://www.digikey.com/product-detail/en/jst-sales-america-inc/SM09B-SRSS-TB-LF-SN/455-1809-1-ND/926880
https://www.digikey.com/product-detail/en/jst-sales-america-inc/SHR-09V-S-B/455-1384-ND/759873
https://www.digikey.com/product-detail/en/jst-sales-america-inc/SHR-09V-S/455-1399-ND/759888
https://www.digikey.com/product-detail/en/jst-sales-america-inc/SSH-003T-P0.2/455-1561-1-ND/720818
https://www.digikey.com/product-detail/en/jst-sales-america-inc/YRS-859/455-2569-ND/2298138

The crimping tools is $1000.  WTF.  Talk about a company that needs to be put out of business.

And who will put them out of business?

Aliexpress:
https://www.aliexpress.com/item/50Pcs-SH-1-0-Wire-Cable-Connector-DIY-SH1-0-JST-2-3-4-5-6/32867440519.html

You can search Aliexpress for JST 1mm pitch.

Of course, I don't know if this is exactly compatible.  I will find out in 12-22 days (minus 2)

Also:
Wurth electronik gives away free samples of all of their stuff.  I could not find this exact compatible product, but it's probably there.

RealSense-Customer-Engineering, 2018-07-12:
[Realsense Customer Engineering Team Comment]
Hi

The P/N of the 9-pin connector is "SM09B-SRSSTB(LF)(SN)" (made by www.jst-mfg.com)
It can be ordered from Digikey as an example:
https://www.digikey.com/product-detail/en/jst-sales-america-inc/SM09B-SRSS-TB-LF-SN/455-1809-1-ND/926880

Also there is a Whitepaper including the connector info and its mating plug ....
https://realsense.intel.com/wp-content/uploads/sites/63/Multiple_Camera_WhitePaper_rev10.pdf

Thanks

RealSense-Customer-Engineering, 2018-07-13:
[Realsense Customer Engineering Team Comment]
Thank you.

You're welcome to bookmark this page. There is a lot of official info published here:

https://realsense.intel.com/intel-realsense-downloads/#whitepaper

Should you have any question, please feel free to contact us.

Have a great day

timprepscius, 2018-07-13:
Hey there, thanks for this. I've read most of those papers.

I found some references for presentations here:
https://www.researchgate.net/profile/Anders_Grunnet-Jepsen/amp

Scroll to the bottom of the page for the references.

Do you have any of those presentation papers uploaded anywhere?

I'm interested in reading pretty much anything I can get a hold of.

Thanks!

RealSense-Customer-Engineering, 2018-07-15:
[Realsense Customer Engineering Team Comment]
Hi

No, I think the realsense.intel.com is the official page to public. I agree you can also get a lot of info from GitHub as well.

Thanks

closed 2018-07-21 by RealSense-Customer-Engineering

Ken-Thorburn, 2019-07-19:
I'm trying to utilize the top connector on a D415, but I do not have an Intel username or password to access the provided material.

ankarako, 2019-07-22:
Hi @Ken-Thorburn ,

You can check [here](https://github.com/VCL3D/VolumetricCapture/wiki) for some info regarding HW sync with RealSense sensors.

Best,
Antonis