realsenseai/librealsense #2036: Add polygons to ply

DannySauval, 2018-07-11:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.09.13.00 | 
| Operating System & Version |  Win 10 + Visual Studio 15 (2017)  | 
| Platform                                 | PC |
| SDK Version                            |  2.0                          | 
| Language                            |  C++                          | 

### Issue Description

Hello everybody, 

I am trying to create a mesh from a .ply file, outputed from the realsense viewer. But I just found out that in this ply there is no faces ! When I worked with Kinect, I could get the faces in the ply. How can I do it with this SDK ? I tried to look in the source code, in the function void viewer_model::render_3d_view(const rect& viewer_rect, texture_buffer* texture, rs2::points points) inside the file model-views.cpp. In this there is a part that seems to be creating quads, I think it is related to what I want, but I haven't been able to figure out how it works.

[https://pastebin.com/VJwyhCus](url)

I put a pastebin code since I don't know why the preview for the code is weird.

Theoretically, I need something like this : 

4 indice1 indice2 indice3 indice4

Which is a quad composed of the vertex corresponding to the indices. But I tried to read the a, b, d and c values, but the values are way too big. I don't understand.

Si, can someone help me to generate a ply file with faces ?

Thanks !

RealSense-Customer-Engineering, 2018-07-16:
[Realsense Customer Engineering Team Comment]
@DannySauval
Currently the PLY output contains only vertices, the feature of adding face into PLY has been requested and put into the development backlog.

DannySauval, 2018-07-17:
@RealSense-Customer-Engineering okay thank you ! I hope it will be quick ! #hope hahah

RealSense-Customer-Engineering, 2018-07-17:
[Realsense Customer Engineering Team Comment]
@DannySauval
I will keep you updated if the feature request has been put into our development sprint. But since the resource is limited, so I don't think this will be quick.

DannySauval, 2018-08-01:
Hello, I'm now okay I added them myself modifying the export_ply function to use Ascii instead of binary PLY in order to """play""" with maths and draw the triangles. It is not realtime usable (it takes about 4 seconds to create the mesh) but for my use it is enough (and I still have to optimize) ! Thank you for your help.

dorodnic, 2018-08-02:
Hi @DannySauval 
Please consider sharing your implementation on GitHub. This is likely to be useful for some else here. Maybe we can add a flag (textual vs. binary) to `export_ply`

DannySauval, 2018-08-03:
Huuum I have to ask, I don't think I am allowed to share the code since I did it in a company !

dorodnic, 2018-10-08:
@DannySauval 
We merged an implementation of this feature, thank you for the feedback

closed 2018-10-09 by RealSense-Customer-Engineering