realsenseai/librealsense #2037: High CPU usage with rs2::pipeline

carun, 2018-07-11:
| Required Info                         |                       |
|---------------------------------|----------------|
| Camera Model                       | D435              |
| Firmware Version                   | 05.09.13.00    |
| Operating System & Version |   Win 10         |
| Platform                                 | PC                  |
| SDK Version                           |  2.13.0            |
| Language                               |  C++              | 
| Segment			         |  Biometrics    | 

### Issue Description

On a 8 core box, the following code seems to be using 100% CPU across all the cores.

```cpp
int main(int argc, char * argv[]) try
{
    // Declare depth colorizer for pretty visualization of depth data
    rs2::colorizer color_map;

    //Create a configuration for configuring the pipeline with a non default profile
    rs2::config cfg;

    //Add desired streams to configuration
    cfg.enable_stream(RS2_STREAM_COLOR, 1920, 1080, RS2_FORMAT_BGR8, 30);

    // Declare RealSense pipeline, encapsulating the actual device and sensors
    rs2::pipeline pipe;
    // Start streaming with default recommended configuration
    pipe.start(cfg);

    while (waitKey(10))
    {
        auto frameset = pipe.wait_for_frames(); // Wait for next set of frames from the camera
        auto frame = color_map(frameset.get_color_frame());

        // Query frame size (width and height)
        const int w = frame.as<rs2::video_frame>().get_width();
        const int h = frame.as<rs2::video_frame>().get_height();
    }

    return EXIT_SUCCESS;
}
```

Apparently this is caused by

```cpp
    rs2::pipeline pipe;
    // Start streaming with default recommended configuration
    pipe.start();
```

When `pipe.start()` is commented out, the CPU usage falls below 20% (even that's high, but we can come to that at a later stage).

dorodnic, 2018-07-11:
Hi @carun 
Are you building the SDK from source? 
If yes, what is the value of CMake flag BUILD_WITH_OPENMP?

carun, 2018-07-11:
Hi @dorodnic 

I didn't build the SDK from source. I'm using the one from GitHub releases.

carun, 2018-07-11:
Intel Realsense Viewer seems to use less than 15% of CPU across the cores.

dorodnic, 2018-07-11:
Thank you, we will look into it.

dorodnic, 2018-07-11:
I can confirm that the pre-compiled version was (unintentionally) built with `BUILD_WITH_OPENMP=true`. Going to be fixed in the upcoming release. Thank you for feedback

carun, 2018-07-11:
Thanks, do you know which version this issue was introduced so that I can use the older version for my testing?

carun, 2018-07-11:
This issue is persistent since the initial version of the SDK 2.0.

carun, 2018-07-12:
CPU usage is less than 15% when with a custom build.

```
cmake -G "Visual Studio 15 2017 Win64" -DCMAKE_BUILD_TYPE=Release ..
cmake --build . --config Release
```

RealSense-Customer-Engineering, 2018-07-16:
[Realsense Customer Engineering Team Comment]
@carun
Yes, just like dorodnic said, librealsense.dll consumes as much CPU resource as possible if the BUILD_WITH_OPENMP is turned on while building the binary. the pre-built binary is with BUILD_WITH_OPENMP turned on so that's why you see the high CPU rate issue. But when you build the binary by yourself, the default CMake file config for BUILD_WITH_OPENMP is OFF, so with this binary, you won't see the high CPU rate issue.

dorodnic, 2018-07-16:
We are working on new installer right now with BUILD_WITH_OPENMP off.

RealSense-Customer-Engineering, 2018-07-16:
[Realsense Customer Engineering Team Comment]
@carun
Do you still have any other question? can I close the ticket?

carun, 2018-07-16:
@RealSense-Customer-Engineering Thanks for your help.

closed 2018-07-16 by carun

madanmch, 2019-04-17:
Hi @dorodnic, we are also facing the same issue with the latest SDK as well as source code built with OPENMP off scenario while polling color info with multiple D435 cameras connected. Is there any solution for this problem? Please let us know